zephyrproject-rtos/zephyr #82425: System fatal when execute 'date set' shell command compiled with Zephyr SDK's Picolibc on 3.7

QifHE, 2024-12-02:
**Describe the bug**

System fatal happened when I executed 'date set' shell command with Picolibc enabled, using the toolchain from the official Zephyr SDK.

The same result happened on QEMU-RISCV64, QEMU-X86, and my company's own ARM-R52 board.
QEMU_X86:
[image]
QEMU_RISCV64:
[image]
However, when I switched to the Minimal Libc, the shell command worked fine.
[image]
The  ARM-R52 board also showed the same results.

**To Reproduce**

QEMU_X86:
cd /home/zephyr/samples/hello_world
Open prj.conf and enable SHELL, POSIX_API, and PICOLIBC
export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
export CROSS_COMPILE=/home/zephyr-sdk-0.16.8/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-
cmake -DBOARD=qemu_x86 -B build/qemu_x86 -DBUILD_VERSION=3.7.0 -DTOOLCHAIN_HAS_PICOLIBC=1 && make -C build/qemu_x86 -j4 VERBOSE=1
make -C build/qemu_x86/ run
Type in 'date set 2024-12-02 11:11:11' or whatever date and time

QEMU_RISCV64:
just change the board and the toolchain, everything else stays the same.

**Expected behavior**

Zephyr fatal or the shell gets stuck.

**Impact**

Cannot use the RTC and Picolibc at the same time.

**Logs and console output**

See the above pictures.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Version used: Zephyr 3.7

**Additional context**

Not sure if it is an SDK or a system bug, so I'm submitting my issue here.

kartben, 2024-12-03:
this sounds *vaguely* familiar, maybe @keith-packard can comment?

keith-packard, 2024-12-03:
Hrm. This works for me.
```
$ west build -b qemu_x86 samples/hello_world/ -- -DCONFIG_SHELL=y -DCONFIG_POSIX_API=y -DCONFIG_PICOLIBC=y
$ ninja -C build run
ninja: Entering directory `build'
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v4.0.0-1420-g537b52e7fe8f ***
Hello World! qemu_x86/atom

uart:~$ date get
1970-01-01 00:00:01 UTC
uart:~$ date set 2024-12-02 11:11:11
2024-12-02 11:11:11 UTC
uart:~$ date get
2024-12-02 11:11:12 UTC
uart:~$ date get
2024-12-02 11:11:14 UTC
uart:~$ QEMU: Terminated
```

keith-packard, 2024-12-03:
Also tested with SDK version 0.16.8. Both tests using `main` 537b52e7fe8f183fc7e51f5d0948464cf71ffa38

kartben, 2024-12-03:
@QifHE  the result of your initial `date get` command is suspicious (year 2262...). Not sure you noticed, and not sure if that can help track down the root cause...  

> Hrm. This works for me. Using SDK version 0.17.0

yep, same here...

cfriedt, 2024-12-03:
@QifHE - is there a chance that you updated zephyr to a more recent commit but did not also run a `west update`? That often causes spectacular failures in very entertaining and surprising ways 😅

QifHE, 2024-12-04:
@kartben @keith-packard @cfriedt @vcgomes Thank you all for the replies! Unfortunately, I am working with an offline compilation environment, which makes it hard to keep the code up to date, and I can only use CMAKE instead of West. The code was pulled initially when v3.7.0 just got released back in #76047 and have been made some small changes.
Since it cannot be reproduced in other environment, I am closing this issue. I'll continue to investigate in my environment, and maybe try to bypass this issue if I cannot solve in the end. Finally, thank you again, everyone.

QifHE, 2024-12-04:
Close for now.

closed 2024-12-04 by QifHE