zephyrproject-rtos/zephyr #82437: drivers: flash: flash_mcux_flexspi: since adding support for W25Q512JV, FlexSPI with the W25Q128JV no longer works

jtlas19, 2024-12-02:
**Describe the bug**

Hello,

Our custom board, using an IMX RT1040, with the W25Q128JV flash chip over FlexSPI as it's main flash, fails to boot after updating our Zephyr version to latest LTS 3.7.1 release (commit 9f824289b28d7aea2eee74f62787c385a5005453). After spending time to narrow it down to a specific commit that caused our issue, the conclusion we came to was that 29b2bf89f8401d35bf3dbb2fe17922944d52edfb broke the flash initialization. After removing that added switch case inside the `flash_flexspi_nor_check_jedec` function, our board is able to boot and operate exactly as it was prior, being on version 3.7 release. As far as I am aware, the command '0xEC' for Quad page programming, does not exist on the W25Q128JV whereas it does on the W25Q512JV.

**To Reproduce**

1. Use a W25Q128JV over FlexSPI using the `nxp_imx_flexspi_nor` driver
2. Attempt to boot on latest LTS 3.7.1

**Expected behavior**

This added switch case does not seem to apply to the W25Q128JV, instead of assuming that the vendor ID means this change will work for all W25Q* flash chips, more indepth checks are needed to validate what flash chip is currently in use before setting custom LUT tables.

**Impact**

Board does not boot. We have created a fork of LTS 3.7.1 with commit 29b2bf89f8401d35bf3dbb2fe17922944d52edfb reverted and will continue to use that until this is fixed. 

**Environment (please complete the following information):**

OS: Windows
Toolchain: Zephyr SDK
Commit: 9f824289b28d7aea2eee74f62787c385a5005453

**Additional context**

I did read briefly on the original PR for adding support for the W25Q512JV that there is other correct ways to do this, is there a time frame on implementing such additions? The PR in question: https://github.com/zephyrproject-rtos/zephyr/pull/80261

danieldegrasse, 2024-12-02:
https://github.com/zephyrproject-rtos/zephyr/pull/81102 should fix this on 4.0, but I think the better way to do this is to check all 3 bytes of the JEDEC ID, rather than the two most significant. I'll create a PR to do that. Long term we shouldn't be placing custom LUT tables in this driver at all, but that's a different conversation

jtlas19, 2024-12-03:
Awesome, we moved to 4.0 and the issue isn't present anymore. Let me know if you need any testers for your future fix. Thanks again!

danieldegrasse, 2024-12-03:
I'm actually going to reopen this issue, because I'll need something to link to with the backport PR to fix this issue on 3.7 (we require backport PRs to reference a specific issue)

PR #82675 (2025-01-06): [Backport v4.0-branch] drivers: flash: flash_mcux_flexspi_nor: check all 3 bytes of JEDEC ID
Backport c12030acb9d5815af290581099ba0b4b05a8910c from #82448.

Fixes #82437
closed 2025-01-17 by decsny