zephyrproject-rtos/zephyr #82449: Bluetooth: mesh: provisionor cdb failure

WilliamGFish, 2024-12-02:
**Description **
When compiling Bluetooth mesh provisioner sample and other applications I receive an error: Failed to create CDB (err -5)

> <dbg> bt_mesh_crypto_psa: bt_mesh_key_import: PSA_IMPORT_KEY -134 
bt_mesh_cdb: Unable to import network key

Only started to happen after some recent changes around the depreciation of TinyCrypt.

**To Reproduce**
Compile and run Bluetooth Mesh Provisioner sample.

**Expected behavior**
creation of CDB with the key that has been provided.

**Impact**
showstopper

**Environment :**
 - *** Booting Zephyr OS build v4.0.0-1313-g2376310fcbf2 ***
 - Windows 10
 - Toolchain (Zephyr SDK, & GCC 13 ...)

mesh_proisioner - src main.c
```
	bt_rand(net_key, 16);

	err = bt_mesh_cdb_create(net_key);
	if (err == -EALREADY) {
		printk("Using stored CDB\n");
	} else if (err) {
		printk("Failed to create CDB (err %d)\n", err);
		return err;
	} else {
		printk("Created CDB\n");
		setup_cdb();
	}
```
subsys-bluetooth-mesh-crypto_psa.c
bt_mesh_key_import()
```
	status = psa_import_key(&key_attributes, in, 16, &out->key);
	LOG_DBG("PSA_IMPORT_KEY %d", status);
	err = status == PSA_SUCCESS ? 0 :
		status == PSA_ERROR_ALREADY_EXISTS ? -EALREADY : -EIO;
```

WilliamGFish, 2024-12-03:
This is related to depreciating Tinycrypt see #82323

alxelax, 2024-12-04:
Hi @WilliamGFish, I suspect that this issue is due to not enabled secure storage. Mesh provisioner is built with enabled settings, that adds persistent property for network, application and device keys. However, mbedtls cannot allocate place to store them during importing.
I hope my PR should fix it: https://github.com/zephyrproject-rtos/zephyr/pull/82319
I'm a bit got stuck with secure storage unit tests, but hopefully manage them soon.
As a workaround, you could locally add this string in your KConfig file: 
https://github.com/zephyrproject-rtos/zephyr/pull/82319/files#diff-007d797c63a5cbfd92a1922fc7fe3e40502988f252054ebeb6da1d77e5f25e87R1494
and remove these: 
https://github.com/zephyrproject-rtos/zephyr/pull/82319/files#diff-593e435b4fd61bbd201c8c4dd55a1ca1b705b698df5499d5f968c6499414bdf9L68
https://github.com/zephyrproject-rtos/zephyr/pull/82319/files#diff-fb5682f6af21bb48291e3d8c7e6c7a2ba99a9601dfc2e97f4deabe459424350dL14-L15

PavelVPV, 2025-02-10:
This issue doesn't reproduce on https://github.com/zephyrproject-rtos/zephyr/commit/55d3deb030bf10198b675fe6ce2effe738f9c01b:

`mesh_provisioner`:
```
*** Booting Zephyr OS build v4.0.0-4871-g55d3deb030bf ***
WARNING: Using a potentially insecure PSA ITS encryption key provider.
Initializing...
Bluetooth initialized
Mesh initialized
Loading stored settings
Created CDB
Provisioning completed
Configuring self...
```

`mesh`:
```
*** Booting Zephyr OS build v4.0.0-4871-g55d3deb030bf ***
WARNING: Using a potentially insecure PSA ITS encryption key provider.
Initializing...
Bluetooth initialized
Mesh initialized
set: on delay: 0 ms time: 0 ms
set: off delay: 0 ms time: 0 ms
```

`mesh_demo`:
```
*** Booting Zephyr OS build v4.0.0-4871-g55d3deb030bf ***
WARNING: Using a potentially insecure PSA ITS encryption key providerialized
Loading stored settings
Provisioning completed
Configuring...

```

closed 2025-02-10 by PavelVPV