zephyrproject-rtos/zephyr #82463: twister: it is not work to pass additional args to native_sim test binary

lylezhu2012, 2024-12-03:
**Describe the bug**
I created a [test suite](https://github.com/zephyrproject-rtos/zephyr/pull/82316). I am trying to run the test suite on native_sim with the command `west twister -v -p native_sim -T tests/classic/sdp_c -O sdp_c --aggressive-no-clean -X bumble:usb:0 -- --bt-dev=hci1`.

The additional args "-- --bt-dev=hci1" cannot be passed to test binary with the following errors,
```
==== Test  started at 2024-12-03 14:27:10.602727 ====
uart connected to pseudotty: /dev/pts/5
Warning: Bluetooth device missing.
Specify either a local hci interface --bt-dev=hciN
or a valid hci tcp server --bt-dev=ip_address:port
*** Booting Zephyr OS build v4.0.0-856-g4a9f819ab6b4 ***
```

**To Reproduce**
Steps to reproduce the behavior:
1. run command `west twister -v -p native_sim -T tests/classic/sdp_c -O sdp_c --aggressive-no-clean -X bumble:usb:0 -- --bt-dev=hci1`
2. See error in file `handler.log`

**Expected behavior**
Argument `--bt-dev=hci1` should be passed to bin `zephyr.exe`.

**Impact**
The test suite cannot be passed.

**Logs and console output**
logs of file `handler.log`
```
==== Test  started at 2024-12-03 14:27:10.602727 ====
uart connected to pseudotty: /dev/pts/5
Warning: Bluetooth device missing.
Specify either a local hci interface --bt-dev=hciN
or a valid hci tcp server --bt-dev=ip_address:port
*** Booting Zephyr OS build v4.0.0-856-g4a9f819ab6b4 ***
```

**Environment (please complete the following information):**

 - OS: WSL
 - Toolchain: Zephyr SDK
 - Commit SHA: [PR](https://github.com/zephyrproject-rtos/zephyr/pull/82316)

nashif, 2024-12-03:
passing arguments works as expected, I think the issue here is the fact you are using pytest, so that is a completely different behavior and you need to do adapt the test for pytest and pass arguments using pytest.

lylezhu2012, 2024-12-04:
> passing arguments works as expected, I think the issue here is the fact you are using pytest, so that is a completely different behavior and you need to do adapt the test for pytest and pass arguments using pytest.

I tested command line `west twister -v -p native_sim -T tests/classic/sdp_c -O sdp_c --aggressive-no-clean -X bumble:usb:0 --pytest-args='--bt-dev=hci1'`.
The error is ported,
`pytest: error: unrecognized arguments: --bt-dev=hci1`

The following is error log,
```
pytest --twister-harness -s -v --build-dir=/mnt/c/lylezhu/zephyrproject/zephyr/sdp_c/native_sim_native/tests/classic/sdp_c/sdp.c --junit-xml=/mnt/c/lylezhu/zephyrproject/zephyr/sdp_c/native_sim_native/tests/classic/sdp_c/sdp.c/report.xml --log-file-level=DEBUG '--log-file-format=%(asctime)s.%(msecs)d:%(levelname)s:%(name)s: %(message)s' --log-file=/mnt/c/lylezhu/zephyrproject/zephyr/sdp_c/native_sim_native/tests/classic/sdp_c/sdp.c/twister_harness.log --platform=native_sim/native /mnt/c/lylezhu/zephyrproject/zephyr/tests/classic/sdp_c/pytest --dut-scope=session --log-cli-level=DEBUG '--log-cli-format=%(levelname)s: %(message)s' --base-timeout=150 --device-type=native --twister-fixture=bumble:usb:0 --log-level=DEBUG --bt-dev=hci1 -p twister_harness.plugin

ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --bt-dev=hci1
inifile: None
rootdir: /mnt/c/lylezhu/zephyrproject/zephyr
```

nashif, 2024-12-04:
@gchwier @PerMac  can you help here?

PerMac, 2024-12-05:
https://docs.zephyrproject.org/latest/develop/test/pytest.html#passing-extra-arguments

lylezhu2012, 2024-12-05:
> https://docs.zephyrproject.org/latest/develop/test/pytest.html#passing-extra-arguments

> west twister -v -p native_sim -T tests/classic/sdp_c -O sdp_c --aggressive-no-clean -X bumble:usb:0 --pytest-args='--bt-dev=hci1'

The correct command line should be `west twister -v -p native_sim -T tests/classic/sdp_c -O sdp_c --aggressive-no-clean -X bumble:usb:0 --pytest-args='`**-k** `--bt-dev=hci1'`, rigth?

PerMac, 2024-12-05:
no, the example uses `-k` which is pytest arg to filter by the test name, you don't need it in your case. whatever is in `--pytest-args` is passed 1:1 to pytest call. You can call twister with extra `-v --log-level debug` and it will print you the full pytest command twister is calling. It will help you to figure out if it looks good and also you can copy the command and tweak with it to find the proper way to call just pytest without involving twister, and afterwards adapt the twister call or even better, the yaml file with your test, since this arg is required by the test to work

PerMac, 2024-12-05:
In your conftest.py you only define `--hci-transport` arg. I don't see `--bt-dev` being defined anywhere in your linked tests.

nashif, 2024-12-05:
@PerMac 

> no, the example uses `-k` which is pytest arg to filter by the test name, you don't need it in your case. whatever is in `--pytest-args` is passed 1:1 to pytest call. You can call twister with extra [...]

he is trying to pass the argument to zephyr.exe executable in this case via `extra_test_args`, not to pytest.

lylezhu2012, 2024-12-05:
> @PerMac
>
>
> he is trying to pass the argument to zephyr.exe executable in this case via `extra_test_args`, not to pytest.

Correct. Refer to https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-tools.html#using-the-host-system-bluetooth-controller, the args `--bt-dev=hcix` needs to be passed to DUT runner.
Just like as, run the DUT with command line
`$ sudo ./build/zephyr/zephyr.exe --bt-dev=hci0`

PerMac, 2024-12-05:
I am checking it now.

lylezhu2012, 2024-12-05:
I found in the script of twister_harness, there are no args passed to DUT runner command line,
```
class NativeSimulatorAdapter(BinaryAdapterBase):
    """Simulator adapter to run `zephyr.exe` simulation"""

    def generate_command(self) -> None:
        """Set command to run."""
        self.command = [str(self.device_config.app_build_dir / 'zephyr' / 'zephyr.exe')]
```

PerMac, 2024-12-05:
Indeed, there is no such option in pytest yet. I had a call with @gchwier and here is the summary:
- The earliest we would be able to add such feature is end of the next week or even week after
- It looks like something relatively easy, so maybe @lylezhu2012 you'd like to add such feature to pytest? If so, here are our initial thoughts:

question: in principle, can such extra args work only with native_sim or does it make sense for other device types (qemu, hardware, etc)? I guess not, since only with native_sim the binary is executed "as it is" and other types have wrappers around, like hw is governed by a runner (e.g. jlink with it's own set of recognizable args). 

If only native sim, then the `generate_command` in the place you pointed https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/pytest-twister-harness/src/twister_harness/device/binary_adapter.py#L120C9-L120C25 can be expanded with an additional arg (e.g. `extra_test_args`), correct? 

The best would be if such extra arg  can be passed from twister cli and/or test yaml. It seems there is already a handling of such case for ztest tests in twister, but for whatever reason it is implemented not as a regular arg_parser arg but with some extra internal handling. It is also currently not supported from a yaml. https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/environment.py#L951

Twister can be then extended to pass this arg to pytest harness, and pytest harness to handle passing this arg to pytest.

PR #82620 (2024-12-12): twister: Pass extra test args to pytest
Pass additional args to test binary.
Fixes #82463

to test just run command:
`./scripts/twister -vv -p native_sim -T samples/subsys/testsuite/pytest/shell  -ll debug --no-clean --pytest-args='-k test_shell_print_version' -- -stop_at=3 -testargs abc`

in the log you can find, that twister calls:
`pytest --twister-harness -s -v ... --device-type=native '--extra-test-args=-stop_at=3 -testargs abc' '-k test_shell_print_version' -p twister_harness.plugin`
and then pytest calls:
`./twister-out/native_sim_native/samples/subsys/testsuite/pytest/shell/sample.pyte [... cut]
commit c4d884cc (2024-12-12): twister: Pass extra test args to pytest

Pass additional args to test binary. Twister
passes it only for native_sim.
Fixes #82463
closed 2024-12-12 by nashif