zephyrproject-rtos/zephyr #82473: IMXRT1170_EVK: after latest zephyr update M4 runs 50% slower from FlexSPI when M7 is running from FlexSPI, when M7 is stopped M4 runs normally

RadekPolyend, 2024-12-03:
Use case:
M7 is master and runs from FlexSPI, M4 is slave and runs from FlexSPI as well. M4 runs around 50% slower than before update but when M7 is stopped by debugger M4 core runs as it should.

nordicjm, 2024-12-03:
Can you edit your post and include all the information (filled in) from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md?plain=1 ? This is mandatory

RadekPolyend, 2024-12-03:
It's really hard to point out the reproduction steps because there is no possibility to build vanilla zephyr with M4 XIP support

nordicjm, 2024-12-03:
> It's really hard to point out the reproduction steps because there is no possibility to build vanilla zephyr with M4 XIP support

Not sure why you are reporting an issue to zephyr if the issue is in a vendor or third party system that bolts on to zephyr, report the issue to the vendor directly

RadekPolyend, 2024-12-03:
Well because this is zephyr behavior, not application behavior. If the same application has been working on previous version of zephyr in the way it should and it doesn't on newer version, which is the only thing that has changed, than this is zephyr issue, isn't it ?

nordicjm, 2024-12-03:
If you can't provide any details or any information to reproduce this, there is nothing for anyone to investigate

DerekSnell, 2024-12-04:
Hi @RadekPolyend ,

This sounds like an application optimization topic, instead of a Zephyr bug.  Typically, multicore applications will configure the cores to execute from different memories.  Especially when performance is a concern, this will reduce contention in the SOC and external memory interfaces.  This is why the default Zephyr config for the MIMXRT1170-EVK places the [CM4 code in internal SRAM](https://github.com/zephyrproject-rtos/zephyr/blob/595d81a941c5bdcb67d7d33c9125d0aba2194a6a/boards/nxp/mimxrt1170_evk/mimxrt1170_evk_mimxrt1176_cm4.dts#L28), while the CM7 core eXecutes In Place (XIP) from external flash.  It seems for your board, you changed this to have the code of both cores in external flash.

The FlexSPI interface to external flash is a particular bottleneck to consider.  The raw memory bandwidth is limited, especially if using external QSPI flash.  The i.MX RT architecture leverages caches and internal SRAMs to help improve this performance, and the application note [AN12437](https://www.nxp.com/docs/en/application-note/AN12437.pdf) covers this in detail, and shows how the FlexSPI performance compares to other options.  If both cores are XIPing from the FlexSPI flash, then there is constant contention between the two masters, waiting on the read transfers from the other core.  Then their performance depends heavily on the cache and the FlexSPI prefetch buffers.

I do not know what specific change occurred in the Zephyr code base that shows this performance drop in your application.  However, I think the choice to have both cores XIP from external flash makes the application vulnerable to performance changes with subtle changes in code.  In this use-case, the performance will depend heavily on what is available in cache, and how the two cores are accessing the external flash concurrently.  Seemingly minor changes in the code may have a large impact on the performance.  If performance is a concern, a better option is to leverage the different memories to avoid bottlenecks, and minimze raw accesses to the slow external flash.  You could try keeping both XIPing from the external flash, but may need to spend some continual effort optimizing your application.  This cache appnote [AN12042](https://www.nxp.com/docs/en/application-note/AN12042.pdf) may be helpful.

Another option is to analyze your app, and only relocate the code that needs low-latency, or is executed frequently to other RAMs.  That could help reduce contention between the cores for the critical code that matters.  One example of this is the [zperf sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/zperf).  When built for the MIMXRT1170-EVK board, it mostly XIPs from QSPI flash, but relocates the networking stack and Ethernet drivers to ITCM to maximize performance.

I do not think this is a bug in the Zephyr codebase, and I think we should convert this topic to a Discussion.  You are welcome to track down the specific change that causes the performance drop.  But as I shared, I think the best solution here is to configure the application to make it less vulnerable to changes in code.

Best regards

RadekPolyend, 2024-12-05:
@DerekSnell 
Thank you for your response, I am pretty aware about the contention however what strikes is the huge difference between old and new zephyr which made me think that maybe someone can try to at least to have a look at the topic. I am aware about RAM possibilities but that's good for some showcases. In real applications there is much more code than RAM is able to hold and therefore Flash solution seems to be the only reasonable solution.Of course parts of the code can be moved using zephyr relocation feature but it still may not be enough. Nevertheless just wanna to have your ears regarding the matter.

DerekSnell, 2024-12-05:
Hi @RadekPolyend ,
In that case, if you want to treat this as a bug, then I think you need to narrow down to the specific change that causes the performance drop.  Or share an app that easily replicates the performance drop that runs on the MIMXRT1170-EVK so others can investigate.
Let us know what you find, best regards

decsny, 2024-12-13:
@RadekPolyend You can definitely give more information in this ticket, you didn't even say which version of zephyr is the old version and which is the new version. Looking at the template, maybe you can't give steps to reproduce, but the rest of the sections I don't see why you can't fill out (impact, environment, expected behavior, problem description, additional context, and following all the suggested notes on each section). Feel free to reopen when you have done due diligence, otherwise this ticket is cannot be acted upon by anyone

closed 2024-12-13 by decsny