zephyrproject-rtos/zephyr #82554: Request of Write role

benothmn-st, 2024-12-04:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Nidhal Ben Othmen
1. GitHub username: benothmn-st
1. Organization (optional): STMicroelectronics
Contributing in the maintenance of lib/* in hal_stm32 module

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.
https://github.com/zephyrproject-rtos/zephyr/pull/81292
https://github.com/zephyrproject-rtos/zephyr/pull/81346
https://github.com/zephyrproject-rtos/hal_stm32/pull/240
https://github.com/zephyrproject-rtos/hal_stm32/pull/244

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

nashif, 2024-12-04:
Thank you for your contribution to the Zephyr project. 

An invite was sent to you to join the project. You can find the invite here: https://github.com/zephyrproject-rtos/zephyr/invitations.

Note that further access and permissoins to the project will be provided following the project [guidelines](https://docs.zephyrproject.org/latest/project/project_roles.html#tsc-project-roles).

closed 2024-12-04 by nashif

benothmn-st, 2024-12-04:
@nashif thank you for the invitation, I need the rights to collaborate in https://github.com/zephyrproject-rtos/hal_stm32 repo