zephyrproject-rtos/zephyr #82560: imx95_evk m7 hello_world sample doesn't work

blemouzy, 2024-12-04:
**Describe the bug**

hello_world sample produces no serial output on MX95LPD5EVK-19CM with Zephyr v4.0.0.

**To Reproduce**

1. Build sample as indicated in the Zephyr documentation:

  ```bash
  west build -b imx95_evk/mimx9596/m7 samples/hello_world
  ```

2. Build `flash.bin`

  ```bash
  make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
           MKIMG=mkimage_imx8 \
           MCU_IMG=m33_image-mx95evk.bin \
           M7_IMG=../zephyr/build/zephyr/zephyr.bin \
           flash_all
  ```

3. Burn `flash.bin` on the board using `uuu`:

  ```bash
  uuu -b emmc flash.bin flash.bin
  ```

4. Check all 4 board serials --> no Zephyr log (SM and U-Boot correctly start on M33 and A55)

**Expected behavior**

See Zephyr logs on serial.

**Environment**

 - OS: Ubuntu 24.04
 - Zephyr: v4.0.0
 - Zephyr SDK: 0.16.9
 - Yocto: scarthgap + imx-6.6.52-2.2.0

**Additional context**

Building `flash.bin` with provided m7 examples shows output on serial:

1.

  ```bash
  make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
           MKIMG=mkimage_imx8 \
           MCU_IMG=m33_image-mx95evk.bin \
           M7_IMG=../mcore-demos/imx95-19x19-evk_m7_TCM_power_mode_switch.bin \
           flash_all
  ```
  ==>
  ```
  Start SRTM communication
  Task 1 is working now
  
  #############000000Hz
  
  Select the desired operation
  
  Press  A to enter: mode
  Press  P to suspend A55 core
  Press  W to wakeup A55 core
  ```

2.

  ```bash
  make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
           MKIMG=mkimage_imx8 \
           MCU_IMG=m33_image-mx95evk.bin \
           M7_IMG=../mcore-demos/imx95-19x19-evk_m7_TCM_rpmsg_lite_pingpong_rtos_linux_remote.bin \
           flash_all
  ```
==>
  ```
  RPMSG Ping-Pong FreeRTOS RTOS API Demo...
  ```

I can't investigate further this issue as I still don't have a compatible JTAG connector for the board :/

mmahadevan108, 2024-12-05:
@JiafeiPan , can you help take a look at this issue.

JiafeiPan, 2024-12-09:
@yangbolu1991 please help on this issue, thanks.

yangbolu1991, 2024-12-10:
@blemouzy 
For M7 boot firmware, the build should be,

```
make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
         MKIMG=mkimage_imx8 \
         MCU_IMG=m33_image-mx95alt.bin \
         M7_IMG=../zephyr/build/zephyr/zephyr.bin \
         flash_m7
```
For m33_image-mx95alt.bin, you should build with this tag and this cfg.
https://github.com/nxp-imx/imx-sm/blob/lf-6.6.52-2.2.0/configs/mx95alt.cfg

The burn should be,

```
uuu -b emmc  imx-boot-imx95-19x19-lpddr5-evk-sd.bin-flash_all flash.bin
```

The first  imx-boot-imx95-19x19-lpddr5-evk-sd.bin-flash_all is from nxp.com release which is just used to start up u-boot for flashing.
The second flash.bin is the one you generate.

There is also another method to run. See PR
https://github.com/zephyrproject-rtos/zephyr/pull/80507

Thanks.

blemouzy, 2024-12-10:
Hi @yangbolu1991,

>For m33_image-mx95alt.bin, you should build with this tag and this cfg.
>https://github.com/nxp-imx/imx-sm/blob/lf-6.6.52-2.2.0/configs/mx95alt.cfg

Expected version is used:

```bash
cd build/tmp/work/imx95evk-fslc-linux/imx-system-manager/1.0.0/git

git describe --exact-match --tags
lf-6.6.52-2.2.0
```

> For M7 boot firmware, the build should be [...] 

> The burn should be [..]

I made a change in `machine/imx95evk.conf` to build `m33_image-mx95alt.bin`:

```bash
IMXBOOT_VARIANT="alt"
```

and used the provided `uuu` command and now:

- U-Boot doesn't start anymore on A55
- Zephyr starts on M7 but only the first log line is displayed:

```
*** Booting Zephyr OS build v4.0.0 ***
```

The

```
Hello World! imx95_evk/mimx9596/m7
```

is missing.
I also tried other examples and logs are not displayed.

Is there anyway to have A55 boot + all Zephyr logs?

Thanks for your help.

yangbolu1991, 2024-12-10:
@blemouzy 
I'm not sure which core you want to boot, and how you build flash.bin. If you want to use only M7 core, please follow what I methioned.

```
make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
         MKIMG=mkimage_imx8 \
         MCU_IMG=m33_image-mx95alt.bin \
         M7_IMG=../zephyr/build/zephyr/zephyr.bin \
         flash_m7
```

If you want to boot both A55 core and M7 core together, there will be other work to do.
You need to be very clear what IP modules M7 will use and what IP modules A55 will use.
For M33 system manager configuration, neither evk.cfg nor alt.cfg should be used. You must define custom cfg file to allocate permissions for M7 and A55.
Also, you must avoid M7 app and A core Linux BSP accessing IP modules which not allocated to them in code.
Then you can make full flash.bin containing A55 and M7 firmware to run.

```
make SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
         MKIMG=mkimage_imx8 \
         MCU_IMG=<m33 image with custom cfg> \
         M7_IMG=../zephyr/build/zephyr/zephyr.bin \
         flash_all
```

Thanks.

blemouzy, 2024-12-10:
> I'm not sure which core you want to boot

I want to boot M33 (of course), A55 and M7 cores.
My goal is to use Zephyr on M7 and make an example of rpmsg based communication between Linux and Zephyr.

> how you build flash.bin

With the following command:

```bash
make -f soc.mak \
        SOC=iMX95 LPDDR_TYPE=lpddr5 REV=A0 OEI=YES \
        MKIMG=mkimage_imx8 \
        MCU_IMG=m33_image-mx95alt.bin \
        M7_IMG=../zephyr/build/zephyr/zephyr.bin \
        flash_all
```

> For M33 system manager configuration, neither evk.cfg nor alt.cfg should be used. You must define custom cfg file to allocate permissions for M7 and A55. Also, you must avoid M7 app and A core Linux BSP accessing IP modules which not allocated to them in code.

Understood: there is no existing configuration to work both with Linux and Zephyr and I have to do my own one.

But do you have any idea of why Zephyr logs are not displayed with `m33_image-mx95alt.bin` ?

blemouzy, 2024-12-11:
When disabling all board interfaces (just keeping `lpuart3` on), `hello_world` sample run correctly with `m33_image-mx95alt.bin`:

```
*** Booting Zephyr OS build v4.0.0 ***
Hello World! imx95_evk/mimx9596/m7
```

I also try to build `flash.bin` with `mx95alt` + `MSEL=2` to also start the A55, but Linux crashes at startup...

So I guess I can close this ticket even if it's quite frustrating to not have a working Linux + Zephyr helloworld.

closed 2024-12-11 by blemouzy

sivasachu, 2026-10-09:
> 15\. And I finally found the difference between `mx95evk.cfg` and `mx95alt.cfg`.
>          With the following patch, Linux and Zephyr hello_world correctly run:
>          --- a/configs/mx95evk.cfg
>          +++ b/configs/mx95evk.cfg
>          @@ -330,11 +330,11 @@ CPU_M7P             msel=2, start=1, stop=1
>          
>           SCMI_AGENT0         name="M7"
>           MAILBOX             type=mu, mu=9, test=8, priority=high
>          -CHANNEL             db=0, xport=smt, check=crc32, rpc=scmi, type=a2p, \
>          +CHANNEL             db=0, xport=smt, rpc=scmi, type=a2p, \
>                               test=default
>          -CHANNEL             db=1, xport=smt, check=crc32, rpc=scmi, \
>          +CHANNEL             db=1, xport=smt, rpc=scmi, \
>                               type=p2a_notify, notify=24
>          -CHANNEL             db=2, xport=smt, check=crc32, rpc=scmi, \
>          +CHANNEL             db=2, xport=smt, rpc=scmi, \
>                               type=p2a_priority
>              
>            
>          I don't understand why `check=crc32` prevents Zephyr to boot (and not FreeRTOS) but that definitely fix the issue.

I made these changes to my cfg file and generated a m33_image.bin and used it to create a new flash.bin.
My need is to boot M33(SM)+M7(FreeRTOS)+A55(Android Automotive OS).
Still it didn't worked showing the same earlier error.

Reset LM 2, reason=fccu, errId=19

Reset LM 2, reason=fccu, errId=19

Reset LM 2, reason=fccu, errId=19

@nashif @blemouzy @yangbolu1991 @fabiobaltieri @mmahadevan108   Can anyone provide me a quick solution.

blemouzy, 2026-10-09:
FYI: the root issue has been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/86347.