zephyrproject-rtos/zephyr #82566: ADC_dt example is not working on runtime with same54_xpro board

sidi-namou, 2024-12-04:
**Describe the bug**

**To Reproduce**
Run this command and load the binary into the target
west build -p always -b same54_xpro samples/drivers/adc/adc_dt

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

sidi-namou, 2024-12-04:
The terminal output is  below:

*** Booting Zephyr OS build v4.0.0-1006-g5a3f0b9506f8 ***
ADC controller device adc@43002000 not ready

The board I used : same54_xpro

nandojve, 2024-12-10:
Hi @attie-argentum ,

Can you check this?

sidi-namou, 2024-12-11:
Hello, I finally solved this issue.
It accured because the api pinctrl_apply_state returns negative value. So the ADC device did not initialized as expected.
To solved it I added default pinctrl as below:
 In same54_xpro-pinctrl.dtsi I added these lines
adc0_default: adc0_default {
		group1 {
			pinmux = <PA4B_ADC0_AIN4>;
		};
	};

	adc1_default: adc1_default {
		group1 {
			pinmux = <PC2B_ADC1_AIN4>;
		};
	};
And In same54_xpro.dts, I added these lines too:
&adc0 {
	status = "okay";
	pinctrl-0 = <&adc0_default>;
	pinctrl-names = "default";
};

&adc1 {
	status = "okay";
	pinctrl-0 = <&adc1_default>;
	pinctrl-names = "default";
};
Now the sample adc_dt is working as expected.

closed 2024-12-11 by sidi-namou

attie-argentum, 2024-12-12:
@sidi-namou Excellent, well done, and thanks for sharing your solution.