zephyrproject-rtos/zephyr #82625: can`t compile the chre demo

shenkegs, 2024-12-05:
The log :
(.venv) sk@sk-x670e:~/zephyrproject/zephyr$ west build -b native_sim samples/modules/chre
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/sk/zephyrproject/zephyr/samples/modules/chre
-- CMake version: 3.30.2
-- Cache files will be written to: /home/sk/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/sk/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: native_sim, qualifiers: native
-- Found host-tools: zephyr 0.17.0 (/home/sk/zephyr-sdk-0.17.0)
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /home/sk/zephyrproject/zephyr/boards/native/native_sim/native_sim.dts
-- Generated zephyr.dts: /home/sk/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/sk/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/sk/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/sk/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/sk/zephyrproject/zephyr/build/zephyr/dts.cmake

/home/sk/zephyrproject/zephyr/samples/modules/chre/prj.conf:1: warning: attempt to assign the value 'y' to the undefined symbol CHRE

/home/sk/zephyrproject/zephyr/samples/modules/chre/prj.conf:4: warning: attempt to assign the value 'y' to the undefined symbol CHRE_LOG_LEVEL_DBG

error: Aborting due to Kconfig warnings

Parsing /home/sk/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/sk/zephyrproject/zephyr/boards/native/native_sim/native_sim_defconfig'
Merged configuration '/home/sk/zephyrproject/zephyr/samples/modules/chre/prj.conf'
CMake Error at /home/sk/zephyrproject/zephyr/cmake/modules/kconfig.cmake:395 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/sk/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/sk/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/sk/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/sk/zephyrproject/.venv/bin/python3 -B/home/sk/zephyrproject/zephyr/build -GNinja -S/home/sk/zephyrproject/zephyr/samples/modules/chre

fabiobaltieri, 2024-12-10:
cc @kartben

yperess, 2025-02-09:
Re-opening, I'm still planning on doing this. Needed https://github.com/zephyrproject-rtos/chre/pull/8 to merge but I think I need to update that PR too by now.

closed 2025-04-25 by github-actions[bot]

avisconti, 2026-01-12:
I was looking at this part for my first time ever but seems that this issue is still there. It's a pity that CHRE is not supported...