zephyrproject-rtos/zephyr #82627: driver: esp32-i2c: add support for i2c_target API

dancollins, 2024-12-05:
**Is your feature request related to a problem? Please describe.**
`CONFIG_I2C_TARGET` provides a great way to use a board as an I2C slave device. We're using this on STM32 parts as well as Nordic's nrfx-twis driver for a similar feature. The ESP32 also supports being an I2C slave, but not using any Zephyr APIs or drivers.

**Describe the solution you'd like**
Support i2c_target API, allowing for a consistent experience across boards/SoCs.

**Describe alternatives you've considered**
Nordic has not done this and, instead, exposes their nrfx-twis driver which provides the same functionality. I will likely use the ESP32 I2C HAL and the slave API provided by Espressif.

evinlodder, 2025-10-30:
@sylvioalves are you still planning on working on this? A project I am working on would benefit from the API and I could try to tackle it myself if it is low on your priority list.

sylvioalves, 2025-11-03:
Hi, we have no plans yet to support I2C as target mode. You are welcome to enabled this and we can support you. Let me know, thanks!

evinlodder, 2025-12-12:
@sylvioalves I have a start [here](https://github.com/evinlodder/zephyr/blob/add_esp32_i2c_target/drivers/i2c/i2c_esp32.c). However, I have a question: on the DET_START interrupt, it doesn't seem like the slave_rw bit is set correctly and the main_state_last (both in sr reg field) doesn't seem to indicate either TX/RX phase (always in address shift phase for me, for some reason?). What is the best way to tell if the controller wants to read/write so that i can call the _requested callback?

Sorry if this seems like a stupid question, I have very little experience with this. If you think there would be a better place to get help lmk!

sylvioalves, 2025-12-12:
That is great! Would you mind creating a draft PR instead? Easy to analyze.