zephyrproject-rtos/zephyr #82633: Input 'wifi connect' using Wi-Fi shell, shows redundant log of 'Subcommands:'

MaochenWang1, 2024-12-06:
**Describe the bug**
Using rd_rw612_bga board (or any board supports Wi-Fi), Input 'wifi connect' using Wi-Fi shell, shows large redundant log of 'Subcommands:'.
This issue is not seen in v3.7.

**To Reproduce**
Input 'wifi connect' in console.

**Expected behavior**
Expected only shows 'wifi connect' related parameters prompt.
But actually will show all the redundant wifi cmd help log.

**Impact**
Difficult for user to use shell.

**Logs and console output**
`uart:~$ wifi connect
connect: wrong parameter count
connect - Connect to a Wi-Fi AP
          <-s --ssid "<SSID>">: SSID.
          [-c --channel]: Channel that needs to be scanned for connection. 0:any
          channel.
          [-b, --band] 0: any band (2:2.4GHz, 5:5GHz, 6:6GHz]
          [-p, --psk]: Passphrase (valid only for secure SSIDs)
          [-k, --key-mgmt]: Key Management type (valid only for secure SSIDs)
          0:None, 1:WPA2-PSK, 2:WPA2-PSK-256, 3:SAE-HNP, 4:SAE-H2E, 5:SAE-AUTO,
          6:WAPI,7:EAP-TLS, 8:WEP, 9: WPA-PSK, 10: WPA-Auto-Personal, 11: DPP
          12: EAP-PEAP-MSCHAPv2, 13: EAP-PEAP-GTC, 14: EAP-TTLS-MSCHAPv2,
          15: EAP-PEAP-TLS, 16:EAP_TLS_SHA256
          17: FT_PSK, 18: FT_SAE, 19: FT_EAP, 20: FT_EAP_SHA384
          21: WPA-Auto-Personal-H2E, 22: WPA-Auto-Personal-AUTO
          [-w, --ieee-80211w]: MFP (optional: needs security type to be
          specified)
          : 0:Disable, 1:Optional, 2:Required.
          [-m, --bssid]: MAC address of the AP (BSSID).
          [-t, --timeout]: Timeout for the connection attempt (in seconds).
          [-a, --anon-id]: Anonymous identity for enterprise mode.
          [-K, --key1-pwd for eap phase1 or --key2-pwd for eap phase2]:
          Private key passwd for enterprise mode. Default no password for
          private key.
          [-S, --suiteb-type]: 1:suiteb, 2:suiteb-192. Default 0: not suiteb
          mode.
          [-T, --TLS-cipher]: 0:TLS-NONE, 1:TLS-ECC-P384, 2:TLS-RSA-3K.
          [-V, --eap-version]: 0 or 1. Default 1: eap version 1.
          [-I, --eap-id1]: Client Identity. Default no eap identity.
          [-P, --eap-pwd1]: Client Password.
          Default no password for eap user.
          [-R, --ieee-80211r]: Use IEEE80211R fast BSS transition connect.[-d,
          --domain-match]: Server domain name constraint.
          [-h, --help]: Print out the help for the connect command.

Subcommands:
  11k                   : Configure 11k or get 11k status.
                          [enable/disable]

  11k_neighbor_request  : Send Neighbor Report Request frame.
                          [ssid <ssid>]

  11v_btm_query         : <query_reason: The reason code for a BSS transition
                          management query>.

  ap                    : Access Point mode commands.
  channel               : wifi channel setting
                          This command is used to set the channel when
                          monitor or TX-Injection mode is enabled
                          Currently 20 MHz is only supported and no BW parameter
                          is provided
                          [-i, --if-index <idx>] : Interface index
                          [-c, --channel <chan>] : Set a specific channel number
                          to the lower layer
                          [-g, --get] : Get current set channel number from the
[... 144 lines cut ...]
                          channel set has to be separated by a _. For example, a
                          valid channel specification can be 2:1,6_5:36 or
                          2:1,6-11,14_5:36,163-177,52. Care should be taken to
                          ensure that configured channels don't exceed
                          CONFIG_WIFI_MGMT_SCAN_CHAN_MAX_MANUAL
                          [-h, --help] : Print out the help for the scan
                          command.

  statistics            : Wi-Fi interface statistics.
                          [reset] : Reset Wi-Fi interface statistics
                          [help] :  Print out the help for the statistics
                          command.
  status                : Status of the Wi-Fi interface.

  twt                   : Manage TWT flows.
  version               : Print Wi-Fi Driver and Firmware versions

  wps_pbc               : Start a WPS PBC connection.

  wps_pin               : Set and get WPS pin.
                          [pin] Only applicable for set.
`

**Additional context**
This issue is introduced by https://github.com/zephyrproject-rtos/zephyr/commit/32aa1d634ed9992505ea551b567cdb2e65d418b9.

MaochenWang1, 2024-12-06:
`uart:~$ wifi connect
connect: wrong parameter count
connect - Connect to a Wi-Fi AP
          <-s --ssid "<SSID>">: SSID.
          [-c --channel]: Channel that needs to be scanned for connection. 0:any
          channel.
          [-b, --band] 0: any band (2:2.4GHz, 5:5GHz, 6:6GHz]
          [-p, --psk]: Passphrase (valid only for secure SSIDs)
          [-k, --key-mgmt]: Key Management type (valid only for secure SSIDs)
          0:None, 1:WPA2-PSK, 2:WPA2-PSK-256, 3:SAE-HNP, 4:SAE-H2E, 5:SAE-AUTO,
          6:WAPI,7:EAP-TLS, 8:WEP, 9: WPA-PSK, 10: WPA-Auto-Personal, 11: DPP
          12: EAP-PEAP-MSCHAPv2, 13: EAP-PEAP-GTC, 14: EAP-TTLS-MSCHAPv2,
          15: EAP-PEAP-TLS, 16:EAP_TLS_SHA256
          17: FT_PSK, 18: FT_SAE, 19: FT_EAP, 20: FT_EAP_SHA384
          21: WPA-Auto-Personal-H2E, 22: WPA-Auto-Personal-AUTO
          [-w, --ieee-80211w]: MFP (optional: needs security type to be
          specified)
          : 0:Disable, 1:Optional, 2:Required.
          [-m, --bssid]: MAC address of the AP (BSSID).
          [-t, --timeout]: Timeout for the connection attempt (in seconds).
          [-a, --anon-id]: Anonymous identity for enterprise mode.
          [-K, --key1-pwd for eap phase1 or --key2-pwd for eap phase2]:
          Private key passwd for enterprise mode. Default no password for
          private key.
          [-S, --suiteb-type]: 1:suiteb, 2:suiteb-192. Default 0: not suiteb
          mode.
          [-T, --TLS-cipher]: 0:TLS-NONE, 1:TLS-ECC-P384, 2:TLS-RSA-3K.
          [-V, --eap-version]: 0 or 1. Default 1: eap version 1.
          [-I, --eap-id1]: Client Identity. Default no eap identity.
          [-P, --eap-pwd1]: Client Password.
          Default no password for eap user.
          [-R, --ieee-80211r]: Use IEEE80211R fast BSS transition connect.[-d,
          --domain-match]: Server domain name constraint.
          [-h, --help]: Print out the help for the connect command.

Subcommands:
  11k                   : Configure 11k or get 11k status.
                          [enable/disable]

  11k_neighbor_request  : Send Neighbor Report Request frame.
                          [ssid <ssid>]

  11v_btm_query         : <query_reason: The reason code for a BSS transition
                          management query>.

  ap                    : Access Point mode commands.
  channel               : wifi channel setting
                          This command is used to set the channel when
                          monitor or TX-Injection mode is enabled
                          Currently 20 MHz is only supported and no BW parameter
                          is provided
                          [-i, --if-index <idx>] : Interface index
                          [-c, --channel <chan>] : Set a specific channel number
                          to the lower layer
                          [-g, --get] : Get current set channel number from the
                          lower layer
                          [-h, --help] : Help
                          Usage: Get operation example for interface index 1
                          wifi channel -g -i1
                          Set operation example for interface index 1 (setting
                          channel 5)
                          wifi -i1 -c5.

  connect               : Connect to a Wi-Fi AP
                          <-s --ssid "<SSID>">: SSID.
                          [-c --channel]: Channel that needs to be scanned for
                          connection. 0:any channel.
                          [-b, --band] 0: any band (2:2.4GHz, 5:5GHz, 6:6GHz]
                          [-p, --psk]: Passphrase (valid only for secure SSIDs)
                          [-k, --key-mgmt]: Key Management type (valid only for
[... 126 lines cut ...]
                          specified as a list of comma separated values where
                          each value is either a single channel or a channel
                          range specified as chan_start-chan_end. Each band
                          channel set has to be separated by a _. For example, a
                          valid channel specification can be 2:1,6_5:36 or
                          2:1,6-11,14_5:36,163-177,52. Care should be taken to
                          ensure that configured channels don't exceed
                          CONFIG_WIFI_MGMT_SCAN_CHAN_MAX_MANUAL
                          [-h, --help] : Print out the help for the scan
                          command.

  statistics            : Wi-Fi interface statistics.
                          [reset] : Reset Wi-Fi interface statistics
                          [help] :  Print out the help for the statistics
                          command.
  status                : Status of the Wi-Fi interface.

  twt                   : Manage TWT flows.
  version               : Print Wi-Fi Driver and Firmware versions

  wps_pbc               : Start a WPS PBC connection.

  wps_pin               : Set and get WPS pin.
                          [pin] Only applicable for set.
`

MaochenWang1, 2024-12-06:
Hi @jukkar 
Is there any reason why need to change the wifi shell cmd in https://github.com/zephyrproject-rtos/zephyr/commit/32aa1d634ed9992505ea551b567cdb2e65d418b9?
This issue that I report, is expected by your change or not?

jukkar, 2024-12-09:
The reason for that change was to allow wifi credentials shell commands to be found under wifi shell.
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/wifi_credentials/wifi_credentials_shell.c

jukkar, 2024-12-09:
The issue seems to be related to shell, it prints all the commands if there is an error when giving the command.
Not sure if there is a way around this.

jukkar, 2024-12-09:
I will investigate this more, there is something wrong with the wifi shell. For example it does not print help for each command properly, like what net-shell is doing.

PR #82744 (2024-12-11): net: wifi: shell: Remove obsolete subcmd settings from commands
If a shell command does not have a sub command, then the 3rd parameter to SHELL_SUBCMD_ADD() macro should be set to NULL so that the help will only print information specific to that command and not all wifi commands.

Fixes #82633
commit 1f2595a4 (2024-12-11): net: wifi: shell: Remove obsolete subcmd settings from commands

If a shell command does not have a sub command, then the 3rd parameter
to SHELL_SUBCMD_ADD() macro should be set to NULL so that the help
will only print information specific to that command and not all
wifi commands.

Fixes #82633
closed 2024-12-11 by kartben