zephyrproject-rtos/zephyr #82645: Bluetooth MESH qualification tests regressions due to Tinycrypt deprecation

sjanc, 2024-12-06:
**Describe the bug**
Bluetooth MESH qualification tests regressions after https://github.com/zephyrproject-rtos/zephyr/pull/79931 merge

**To Reproduce**
Execute DFUM/SR/FD/BV-01-C qualification test case with AutoPTS.

**Expected behavior**
Tests pass

**Impact**
Bluetooth Mesh qualification is affected

**Logs and console output**
....
[00:00:13.643,676] <inf> bt_mesh_main: Primary Element: 0x0002
[00:00:13.643,707] <err> bt_mesh_main: Failed to import device key
[00:00:13.643,707] <err> bt_mesh_provisionee: Failed to provision (err -5)
....
[00:00:42.123,748] <dbg> bttester: tester_event: service 0x04 opcode 0x8f
[00:00:42.130,859] <err> bt_mesh_net_keys: Unable to generate beacon key
.....

This is affecting multiple tests (mainly DFU and MBTM):

MESH/NODE/KR/BV-01-C - Key Refresh Procedure as Node via Configuration Client Messages
MESH/NODE/KR/BV-02-C - Key Refresh Procedure as Node via Secure Network Beacons
MESH/NODE/KR/BV-04-C - Key Refresh Procedure as Node via Mesh Private Beacons
MESH/NODE/KR/BV-03-C - Key Refresh Procedure Skipping Phase 2 via Secure Network Beacons
MESH/NODE/KR/BV-05-C - Key Refresh Procedure Skipping Phase 2 via Mesh Private Beacons
MESH/CFGCL/KR/BV-01-C - Key Refresh as Configuration Client via Configuration Messages
MESH/CFGCL/KR/BV-02-C - Key Refresh as Configuration Client via Secure Network Beacons
MESH/CFGCL/KR/BV-05-C - Key Refresh with Reject listing via Mesh Private Beacons
MESH/CFGCL/KR/BV-03-C - Key Refresh with Reject Listing via Configuration Messages
MESH/CFGCL/KR/BV-04-C - Key Refresh with Reject Listing via Secure Network Beacons
MESH/CFGCL/KR/BV-06-C - Key Refresh with Reject listing via Mesh Private Beacons
MESH/NODE/KR/BI-01-C - Receive Config Key Refresh Phase Set Messages with Valid and Invalid Transition Values
MESH/NODE/KR/BI-02-C - Ignore New Keys During Key Refresh Procedure
MESH/NODE/KR/BI-04-C - Receive Messages with Old Keys During Configuration Client Key Refresh
MESH/NODE/KR/BI-05-C - Receive Messages with Old Keys During Secure Network Beacon Key Refresh
MESH/NODE/KR/BI-06-C - Receive Messages with Old Keys During Mesh Private Beacon Key Refresh
MESH/NODE/FRND/FN/BV-14-C - Friend Key Refresh via Secure Network Beacons - Friend Node
MESH/NODE/FRND/FN/BV-25-C - Friend Key Refresh via Mesh Private Beacons - Friend Node
MESH/NODE/FRND/FN/BV-15-C - Friend Key Refresh via Configuration Client Messages - Friend Node
MESH/NODE/FRND/FN/BV-20-C - Adding another Friend Update upon reception of Secure Network Beacons - Friend Node
MESH/NODE/FRND/FN/BV-26-C - Adding another Friend Update upon reception of Mesh Private Beacons - Friend Node
MESH/SR/PROX/BV-23-C - Send Secure Network Beacon when Secondary NetKey Added
MESH/SR/PROX/BV-24-C - Send Mesh Private Beacon when Secondary NetKey Added
MESH/NODE/CFG/GPXY/BV-02-C - Network ID Advertising on Multiple Subnets
MESH/NODE/CFG/MP/BV-02-C - Disabling Publishing - AppKey/Config NetKey Deleted. This test case requires configuring TSPX_iut_model_id_used.
MESH/NODE/CFG/NKL/BV-01-C - Respond to Config NetKey Add and Get Messages
MESH/NODE/CFG/NKL/BV-02-C - Respond to Config NetKey Update Message
MESH/NODE/CFG/NKL/BV-03-C - Respond to Config NetKey Delete Message
MESH/NODE/CFG/NKL/BI-01-C - Respond to Config NetKey Update Message (Invalid NetKeyIndex)
MESH/NODE/CFG/NKL/BI-03-C - Respond to Config NetKey Add Message (Insufficient Resources)
MESH/NODE/CFG/AKL/BV-02-C - Respond to Config AppKey Update Message
MESH/NODE/CFG/AKL/BI-02-C - Respond to Config AppKey Update Message (Invalid AppKey Index and NetKey Index)
MESH/NODE/CFG/AKL/BI-04-C - Respond to Config AppKey Add Message (NetKeyIndex and AppKeyIndex Already Stored)
MESH/NODE/CFG/NID/BV-02-C - Node Identity Advertising on Multiple Subnets
MESH/NODE/CFG/NID/BV-03-C - Node Identity Config NetKey Deleted
MESH/NODE/CFG/HBP/BV-06-C - Disabling publishing of Heartbeat messages - Config NetKey Deleted
MESH/SR/RPR/LNK/BV-19-C - Link Active State with Unprovisioned Device over PB-ADV - Handle Network Key Removal
MESH/SR/PRB/PGPXY/BV-02-C - Private Network Identity Advertising on Multiple Subnets
MESH/SR/PRB/PNID/BV-02-C - Private Node Identity Advertising on Multiple Subnets
MESH/SR/PRB/PNID/BV-03-C - Private Node Identity NetKey Deleted
MESH/NODE/SBR/NET/BV-01-C - Receive Network Layer Message, DST as Unicast Address, IUT as Subnet Bridge
MESH/NODE/SBR/NET/BV-02-C - Ignore Replayed Network Layer Message
MESH/SR/BCM/BDS/BV-01-C - Bridge Directive State Add Behavior
MESH/SR/BCM/BDS/BV-02-C - Bridge Directive State Remove Behavior
MESH/SR/BCM/BDS/BV-03-C - Bridge Directive State Get Behavior
MESH/SR/BCM/BDS/BV-04-C - Bridge Directive State Subnets Get Behavior
[... 105 lines cut ...]
DFUM/SR/FU/BV-08-C - Receive Firmware Update Start - BLOB Trasfer Busy, IUT as Target Node
DFUM/SR/FU/BV-09-C - Firmware Update Apply, IUT as Target Node
DFUM/SR/FU/BV-10-C - Receive Firmware Update Cancel - Idle Phase, IUT as Target Node
DFUM/SR/FU/BV-11-C - Receive Firmware Update Cancel - Transfer Active Phase, IUT as Target Node
DFUM/SR/FU/BV-12-C - Receive Firmware Update Cancel - Transfer Error Phase, IUT as Target Node
DFUM/SR/FU/BV-13-C - Receive Firmware Update Cancel - Verification Succeeded Update Phase, IUT as Target Node
DFUM/SR/FU/BV-14-C - Receive Firmware Update Cancel - Verification Failed Update Phase, IUT as Target Node
DFUM/SR/FU/BV-15-C - Receive Firmware Update Cancel - Applying Update Phase, IUT as Target Node
DFUM/SR/FU/BV-16-C - Receive Firmware Update Start - Transfer Error Phase, IUT as Target Node
DFUM/SR/FU/BV-17-C - Receive Firmware Update Start - Verification Success Phase, IUT as Target Node
DFUM/SR/FU/BV-18-C - Receive Firmware Update Start - Verification Failed Phase, IUT as Target Node
DFUM/SR/FU/BV-19-C - Receive Firmware Update Start - Applying Update, IUT as Target Node
DFUM/SR/FU/BV-20-C - Receive Firmware Update Apply - Idle Phase, IUT as Target Node
DFUM/SR/FU/BV-21-C - Receive Firmware Update Apply - Transfer Error Phase, IUT as Target Node
DFUM/SR/FU/BV-22-C - Receive Firmware Update Apply - Transfer Active Phase, IUT as Target Node
DFUM/SR/FU/BV-23-C - Receive Firmware Update Apply - Verification Failed Phase, IUT as Target Node
DFUM/SR/FU/BV-24-C - Resume Firmware Transfer, IUT as Target Node
MBTM/CL/BT/BV-01-C - Transfer BLOB Procedure, Push BLOB Transfer Mode
MBTM/CL/BT/BV-02-C - Transfer BLOB Procedure, Pull BLOB Transfer Mode
MBTM/CL/BT/BV-03-C - Cancel Transfer Procedure
MBTM/CL/BT/BV-04-C - Transfer BLOB Procedure - Failure
MBTM/CL/BT/BV-05-C - Transfer BLOB Procedure - Transfer Timeout
MBTM/CL/BT/BV-06-C - Transfer BLOB Procedure - Partial Block Report Timeout
MBTM/CL/BT/BV-07-C - Transfer BLOB Procedure - Disable Receiver
MBTM/CL/BT/BV-08-C - Get Transfer Progress Procedure

valeriosetti, 2024-12-06:
Based an offline discussion with @alxelax he thinks that https://github.com/zephyrproject-rtos/zephyr/pull/82319 can help in fixing this problem. Therefore I propose for that PR to be merged and then re-run these test to check if it really did resolve these issues.

sjanc, 2024-12-06:
OK,  I'll trigger PR run on that with some smoke testing

sjanc, 2024-12-19:
So with that PR it is stil failing,  I also tried to enable following in prj.conf but that doesn't help either
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y

[00:00:47.309,295] <err> bt_mesh_net_keys: Unable to generate beacon key

I also get this:
[00:00:23.146,118] <inf> bt_mesh_main: Primary Element: 0x0002
[00:00:23.146,148] <err> bt_mesh_main: Failed to import device key
[00:00:23.146,148] <err> bt_mesh_provisionee: Failed to provision (err -5)
[00:00:23.177,856] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:23.764,465] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:24.259,765] <err> os: ***** BUS FAULT *****
[00:00:24.259,796] <err> os:   Precise data bus error
[00:00:24.259,796] <err> os:   BFAR Address: 0x8
[00:00:24.259,826] <err> os: r0/a1:  0x00005ec3  r1/a2:  0x00000000  r2/a3:  0x000003e8
[00:00:24.259,857] <err> os: r3/a4:  0x2000a0ac r12/ip:  0x00000000 r14/lr:  0x00047755
[00:00:24.259,857] <err> os:  xpsr:  0x21000000
[00:00:24.259,857] <err> os: Faulting instruction address (r15/pc): 0x00022946
[00:00:24.259,887] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:24.259,918] <err> os: Current thread: 0x20007560 (BT MESH WQ)
[00:00:30.374,481] <err> os: Halting system

so what should be enabled in application to get this working as prior to https://github.com/zephyrproject-rtos/zephyr/pull/79931 ?

m-alperen-sener, 2025-01-08:
I see that CONFIG_SETTINGS=y is not enough for SECURE_STORAGE to be selected, BT_SETTINGS needs to be enabled [here](https://github.com/zephyrproject-rtos/zephyr/blob/08377ea9775aa1028ff3fc73cb0dfd50002429e2/subsys/bluetooth/mesh/Kconfig#L1494) in #82319 .

And besides for this particular test case and for all the other test cases, in addition to prj.conf and overlay-mesh.conf, required kconfigs are added by AutoPTS bot when needed in [iut_config.py](https://github.com/auto-pts/auto-pts/blob/master/autopts/bot/iut_config/zephyr.py#L168-L294).

I suggest running test either with bot or following default `iut_config.py`

sjanc, 2025-01-09:
@m-alperen-sener from quick check it looks like CONFIG_SECURE_STORAGE=y seems to fix tester, I wonder if that would be worth separate PR (with just this fix) since #82319 is covering also other things and have some CI issues

m-alperen-sener, 2025-01-09:
I completed the test run for DFUM and MBTM, both mesh layers are passing with #82319, however there are still failures on MESH layer. That needs to be checked, I ll report the ones I can figures out why.

I would suggest enabling SECURE storage in `iut_config.py` it won't hurt anything, just an innocent duplication, and I think it would be beneficial to get some regression from mesh layer until #82319 merges. However it is OK to wait I guess.

sjanc, 2025-01-09:
OK, I think we can enable this in autopts for now,   PR sent    https://github.com/auto-pts/auto-pts/pull/1335

sjanc, 2025-01-30:
this is fixed in main

closed 2025-01-30 by sjanc