zephyrproject-rtos/zephyr #82652: sockets_service: dns: Sockets service calling closed dispatch socket callback causes fatal

emil-lindq, 2024-12-06:
**Describe the bug**
We have a problem with initing & closing the resolve subsystem. We have two network interfaces, and when switching between the two the resolve needs to be closed and inited to create a socket on the new network interface.

The following calls will cause a crash
```
dns_resolve_init_default(dns_resolve_get_default())
dns_resolve_close(dns_resolve_get_default())
/* Crash directly after returning from function */
```
The `sockets_service` thread is crashing. I've been digging a bit and it appears that sockets_service.c runs a poll on the socket created by `dns_resolve_init_default()`. When `dns_resolve_close()` is called it closes the socket. This causes socket_service thread's poll call to return and get back a revent 0x20 (POLLNVAL) as expected on the dns socket.
https://github.com/zephyrproject-rtos/zephyr/blob/a5389df5b9d4c8c73f8d8397654836aea3dae149/subsys/net/lib/sockets/sockets_service.c#L236

It then goes ahead and runs a callback on the closed socket, even if it's closed, and eventually ends up with a `struct dns_socket_dispatcher` context being NULL in dispatcher.c (which it was set to in `dns_resolve_close()`, causing a crash
https://github.com/zephyrproject-rtos/zephyr/blob/a5389df5b9d4c8c73f8d8397654836aea3dae149/subsys/net/lib/dns/dispatcher.c#L132

I don't know if this is a regression or what's the expected run of action. It seems wrong of sockets_service.c to run a callback on a closed socket, however there's no way to unregister a callback either.

**Expected behavior**
Initing and closing resolve context should work.

**Impact**
Not able to switch network interfaces on the dns.

**Logs and console output**

```
>>>>>>>>Calling init
[00:00:18.657,000] <dbg> net_dns_resolve: dns_resolve_init_locked: (net_workq): [0] 8.8.8.8                                                                                                                                                                                                                                                                     
[00:00:18.657,000] <dbg> net_sock: zsock_socket_internal: (net_workq): socket: ctx=0x2403c2ac, fd=5                                                                                                                                                                                                                                                             
[00:00:18.657,000] <dbg> net_sock_svc: socket_service_thread: (net_socket_service): Received restart event.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             
[00:00:23.657,000] <err> net_mgr_eth_lte_wifi: i am here!!!!                                                               
>>>>>>>>Calling close                                                                                                                                                                                                                                     
[00:00:23.657,000] <dbg> net_sock: zsock_close_ctx: (net_workq): close: ctx=0x2403c2ac, fd=5                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             
[00:00:23.657,000] <err> os: ***** BUS FAULT *****                                                                                                                                                                                                                                                                                                              
[00:00:23.657,000] <err> os:   Precise data bus error                                                                                                                                                                                                                                                                                                           
[00:00:23.657,000] <err> os:   BFAR Address: 0x857aba7b                                                                                                                                                                                                                                                                                                         
[00:00:23.657,000] <err> os: r0/a1:  0x857aba6d  r1/a2:  0x0000000e  r2/a3:  0xffffff81                                                                                                                                                                                                                                                                         
[00:00:23.657,000] <err> os: r3/a4:  0xffffffff r12/ip:  0x00000005 r14/lr:  0x08097963                                                                                                                                                                                                                                                                         
[00:00:23.657,000] <err> os:  xpsr:  0xb1000000                                                                                                                                                                                                                                                                                                                 
[00:00:23.657,000] <err> os: s[ 0]:  0x00000030  s[ 1]:  0x00000005  s[ 2]:  0x00000000  s[ 3]:  0x00000014                                                                                                                                                                                                                                                     
[00:00:23.657,000] <err> os: s[ 4]:  0x00000030  s[ 5]:  0x24053b14  s[ 6]:  0x00000004  s[ 7]:  0x080650df                                                                                                                                                                                                                                                     
[00:00:23.657,000] <err> os: s[ 8]:  0x24053ae8  s[ 9]:  0x080c9bec  s[10]:  0x240067e8  s[11]:  0x00000004                                                                                                                                                                                                                                                     
[00:00:23.657,000] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000124  s[15]:  0x240164d8                                                                                                                                                                                                                                                     
[00:00:23.657,000] <err> os: fpscr:  0x00000000                                                                                                                                                                                                                                                                                                                 
[00:00:23.657,000] <err> os: Faulting instruction address (r15/pc): 0x080979d2                                                                                                                                                                                                                                                                                  
[00:00:23.657,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0                                                                                                                                               [... cut ...]
[00:00:23.657,000] <err> os: Current thread: 0x240175a8 (net_socket_service)                                                                                                                                                                                                                                                                                    
[00:00:23.658,000] <err> os: Halting system                                                                    
```
Debug output showing context as NULL after sockets_service calls the callback
[image]

**Environment (please complete the following information):**
Using Zephyr commit c0be51df2fb83f2e772d74b73e10350662af12b8

emil-lindq, 2024-12-06:
For instance, changing
https://github.com/zephyrproject-rtos/zephyr/blob/e12096ee4a592a351e8ef825b3692187484bc956/subsys/net/lib/sockets/sockets_service.c#L259
to 
`            if (ctx.events[i].revents > 0 && (ctx.events[i].revents & 0x20) == 0) {
`
the resolve doesn't switch but it doesn't crash at the very least.

jukkar, 2024-12-09:
@emil-lindq I have a fix proposal at #82731, please try it if possible and report back.

emil-lindq, 2024-12-09:
@jukkar it works great, thank you!!

PR #82731 (2024-12-11): net: dns: Close socket service properly from dispatcher
We need to close the socket service context when dispatcher is unregistered.

Fixes #82652
commit da148ab3 (2024-12-11): net: dns: Close socket service properly from dispatcher

We need to close the socket service context when dispatcher is
unregistered.

Fixes #82652
commit f4946e04 (2024-12-12): net: dns: Close socket service properly from dispatcher

We need to close the socket service context when dispatcher is
unregistered.

Fixes #82652

(cherry picked from commit da148ab3bca534a624c3fd06d7e2aa5981ffb976)
PR #82882 (2024-12-14): [Backport v3.7-branch] net: dns: Close socket service properly from dispatcher
Backport d98fe736848a0c2157085959975035354181a68e~4..d98fe736848a0c2157085959975035354181a68e from #82731.

Fixes #82652
closed 2024-12-11 by kartben