zephyrproject-rtos/zephyr #82653: spi: implement half-duplex tests

erwango, 2024-12-06:
**Is your enhancement proposal related to a problem? Please describe.**

https://github.com/zephyrproject-rtos/zephyr/pull/75983 is adding half-duplex support to STM32 devices.
I don't know if this is the first driver in tree supporting this feature, but for sure we miss a test in tree to ensure non regression of the feature.

**Describe the solution you'd like**
Implement a half-duplex test.
This could be done as a variant of `tests/drivers/spi/spi_controller_peripheral` or be a separate `tests/drivers/spi/half_duplex`

malto101, 2024-12-18:
Hey @erwango , Is it okay if I pick this up if this isn't already picked up.

senerferhat, 2025-01-15:
Hi @erwango , I noticed that someone had expressed interest in working on this issue, but it seems there hasn't been any visible progress for a while. If it's okay, I’d like to work on implementing the half-duplex SPI test. Please let me know if I can take this up.
I’ll refer to #75983  as suggested for details and ensure the tests align with existing patterns. Let me know if there are any additional requirements or guidance. Thank you!

erwango, 2025-01-15:
@senerferhat Sure, go ahead. You can also try to contact @malto101 to get heads up on his progress on this task.

yawningwinner, 2025-03-28:
Hi @erwango, can I take up this issue?

erwango, 2025-03-28:
@yawningwinner Sure, maybe you want to synchronize with @senerferhat and @malto101 to check if they have some progress to share.

leonardo-munoz-pragmafw, 2025-04-24:
Hello @yawningwinner , I’m interested in contributing to this as my first issue. Do you have any updates on it?

leonardo-munoz-pragmafw, 2025-05-13:
Hello everyone!
I’ve been working on this issue and have reached a dead end. I created two threads, one for handling the SPI master and another for handling the SPI slave. The flow is relatively straightforward: the thread assigned to the SPI slave prepares for a reception, then the SPI master writes to the bus; afterwards the SPI slave prepares for a transmission and then the master reads from the bus.

Below are the relevant code snippets from `main.c`, `prj.conf`, and the `.overlay` file.

```c

#include <stdbool.h> 
#include <zephyr/kernel.h>

#include <zephyr/drivers/spi.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/devicetree/spi.h>

#include <zephyr/logging/log.h>

#include <zephyr/ztest.h>

LOG_MODULE_REGISTER(spi_tx, LOG_LEVEL_INF);

const struct device *const spi_master_dt = DEVICE_DT_GET(DT_NODELABEL(spi1));
const struct device *const spi_slave_dt = DEVICE_DT_GET(DT_NODELABEL(spi3));

const struct spi_config spi_master_cfg = {
    .frequency = DT_PROP(DT_NODELABEL(spi1), clock_frequency),
    .operation = SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | SPI_HALF_DUPLEX | SPI_WORD_SET(8),
    .cs = SPI_CS_CONTROL_INIT(DT_NODELABEL(spi1),10),
};

const struct spi_config spi_slave_cfg = {
    .frequency = DT_PROP(DT_NODELABEL(spi3), clock_frequency),
    .operation = SPI_OP_MODE_SLAVE | SPI_TRANSFER_MSB | SPI_HALF_DUPLEX | SPI_WORD_SET(8),
    .cs =SPI_CS_CONTROL_INIT(DT_NODELABEL(spi3),10),
};

static struct k_sem begin_spi_cycle;

void master_thread(void)
{
    uint8_t master_data = 0x0;
    int ret_val;
    struct spi_buf       buf_master = { .buf = &master_data, .len = 1 };
    struct spi_buf_set   master_set_buf  = { .buffers = &buf_master, .count = 1 }; 

    while (1) {

        master_data = 0x33;
        k_sem_take(&begin_spi_cycle,K_FOREVER);
        ret_val = spi_write(spi_master_dt, &spi_master_cfg,&master_set_buf);

        if (ret_val) {
            LOG_ERR("Error in master transmit: %d", ret_val);
        } else {
            LOG_ERR("SPI master transmit success");
        }

        k_sem_take(&begin_spi_cycle,K_FOREVER);
        ret_val = spi_read(spi_master_dt, &spi_master_cfg, &master_set_buf);

        if (ret_val>0) {
            LOG_INF("Read on Master: %d", master_data);
        } else if (ret_val == 0){
            LOG_INF("No bytes read on master");
        }else{
            LOG_INF("SPI Master read error: %d", ret_val);
        }
        k_sleep(K_MSEC(2000));
    }
}
 
void slave_thread(void)
{
    uint8_t slave_data = 0x0;
    int ret_val;
    struct spi_buf       buf_slave = { .buf = &slave_data, .len = 1 };
    struct spi_buf_set   slave_set_buf  = { .buffers = &buf_slave, .count = 1 }; 

    while (1) {

        k_sem_give(&begin_spi_cycle);
        ret_val = spi_read(spi_slave_dt, &spi_slave_cfg, &slave_set_buf);
        k_yield();
        if (ret_val>0) {
            LOG_INF("Read on slave: %d", slave_data);
        } else if (ret_val == 0){
            LOG_INF("No bytes read on slave");
        }else{
            LOG_INF("SPI Slave read error: %d", ret_val);
        }

        k_sleep(K_MSEC(1000));

        slave_data = 0xAA;
        k_sem_give(&begin_spi_cycle);
        ret_val = spi_write(spi_slave_dt, &spi_slave_cfg,&slave_set_buf);
        k_yield();
        if (ret_val) {
            LOG_ERR("Error in slave transmit: %d", ret_val);
        } else {
            LOG_ERR("SPI slave transmit success");
        }
    }
}
 
K_THREAD_DEFINE(master_tid,  1024, master_thread, NULL, NULL, NULL, 8, 0, 0);
K_THREAD_DEFINE(slave_tid,   1024, slave_thread,  NULL, NULL, NULL, 7, 0, 0);

void main(void)
{
    k_sem_init(&begin_spi_cycle,0,1);
}

```

```
&spi1 {
    status = "okay";
	clock-frequency = <2000000>;
	pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pa7 &spi1_nss_pa4>;
	pinctrl-names = "default";
};

&spi3 {
	status = "okay";
	pinctrl-0 = <&spi3_sck_pb3 &spi3_miso_pb4 &spi3_mosi_pb5 &spi3_nss_pa15>;
	pinctrl-names = "default";
	clock-frequency = <2000000>;
};

```

```
CONFIG_SPI=y
CONFIG_GPIO=y

CONFIG_SPI_STM32=y
CONFIG_SPI_SLAVE=y
CONFIG_SPI_STM32_INTERRUPT=y
CONFIG_LOG=y
```

While debugging with my logic analyzer, I found the following problems—perhaps someone can help me with these:

1. When transmitting from the master to the slave in isolation, everything works fine. However, when transmitting from the slave to the master, although the data appears on the bus, the master does not read it.

[image]

2. When I try to perform a master transmission followed by a master reception, the SPI clock ends up in a state I can’t understand, generating more clock pulses than it should.

[image]

I’m not sure if I made a mistake in my code or if this is a library error. Any help is welcome—thanks in advance!

nimisht, 2025-05-23:
Is something off in the hardware setup? Have you tied the slave select lines together for the master (SPI1 - PA4) and the slave (SPI2 - PA15) with a jumper wire? Or do they get wired internally by the software somehow? 

It may be worth monitoring the state of the slave select line to see if it's toggling correctly and framing the data transmissions. 

In the slave thread, does the line `LOG_INF("Read on slave: %d", slave_data);` display the output you expected?

closed 2026-09-07 by kartben