zephyrproject-rtos/zephyr #82678: Sysbuild: Add config variable to specify main image name

chdh, 2024-12-07:
**Is your enhancement proposal related to a problem? Please describe.**

Currently, sysbuild uses the directory name of the application as the name of the main image.

The relevant code is in [sysbuild_images.cmake](https://github.com/zephyrproject-rtos/zephyr/blob/bd1db48d256707809f725a5b11263060ad1e1299/share/sysbuild/cmake/modules/sysbuild_images.cmake#L9):

```
get_filename_component(app_name ${APP_DIR} NAME)
set(DEFAULT_IMAGE "${app_name}")
```

This is an undocumented side effect. There is currently no way to specify a fixed name, independent of the application directory name.

The problem is that if the application is stored in a directory with a different name, the image name will be different and the configuration using that name will no longer be valid (e.g. "sysbuild/&lt;image&gt;/prj.conf").

**Describe the solution you'd like**

Allow the name of the application image to be specified using a config variable, e.g. SB_CONFIG_APP_IMAGE_NAME. Use the app directory name only as a fallback.

**Describe alternatives you've considered**

- Instruct the developers of your app to always use a certain directory name when exporting the project from Git.
- Move the app main directory to a subdirectory in our Git repository.

nordicjm, 2024-12-09:
There should be one application per folder, not multiple. There are no plans to make this name configurable. Plus the part about using `sysbuld/<image/*` doesn't really make sense because you don't use that from the application itself, that only applies to an application including *other* images and not itself, all application (that is not being added by another sysbuild file) configuration should be in the application (non-sysbuild) folders and files

chdh, 2024-12-09:
OK, I understand that `sysbuild/<image>/*` is not used for the main application. But how about other configuration dependencies like `-D<image>_FILE_SUFFIX=...` (for setting the file suffix only for the main application and not globally to all images)?
And the main application is built within `build/<image>`. So references to generated files for the main application have this variable path name.

nordicjm, 2024-12-09:
> OK, I understand that `sysbuild/<image>/*` is not used for the main application. But how about other configuration dependencies like `-D<image>_FILE_SUFFIX=...` (for setting the file suffix only for [...]

This is the expected bahaviour and by design, yes you need to use the name of the application (or image if applying to an image) to use to only apply some configuration to a specific image

nordicjm, 2024-12-10:
> > yes you need to use the name of the application
> 
> OK, but when the image name of the application is not constant and depends on the directory name of the application (and this mechanism is undocumented), we have a problem.

It is consistent though, it's the name of the folder, the folder doesn't change unless you rename it

chdh, 2024-12-14:
> It is consistent though, it's the name of the folder, the folder doesn't change unless you rename it.
> The name of the **default image** is the name of the folder it resides is, there is only ever one default image,

When we export our project from Git, we need to choose a name for the project folder.
When testing different versions of the same projects, we use folders with different names.

So because the name of the main image is the name of the project folder, the main image name is not constant. It is an undocumented side effect, that the name of the main image is taken from the name of the project folder.

I don't have a problem with the names of the sub-images and I understand that I can specify their names. My problem is that I can't explicitly specify the name of the main image.

nordicjm, 2024-12-16:
> When we export our project from Git, we need to choose a name for the project folder.

Projects should be git based and remain git based

> When testing different versions of the same projects, we use folders with different names.

You can use one folder with all the files and selectively enable/disable the ones to compile in based on what is selected, or have overlays/conf files and use things like file suffix to control the version you are building

boaks, 2025-05-11:
I'm struggling with this naming pattern as well.

> When we export our project from Git, we need to choose a name for the project folder.

I consider "export" means checkout the project into a specific folder, the project is still git based and remains git based.

@nordicjm

Please check the documentation for `west init`. As long as that supports to provide different "topdir", the sysbuild domain name of the application should not depend on the name of the directory without mention that in the "west init" docu. 
It's also very unclear, which characters are then allowed for the topdir, e.g. topdir "my-app", works as directory name, does it work as sysbuild domain name? 

```
west build ... -- -Dmy-app_MY_MACRO=y
```

Is that OK? For NCS 2.9.1 / zephyr 3.7.99 I don't see, that "MY_MACRO" gets defined.

chdh, 2025-05-11:
> > When we export our project from Git, we need to choose a name for the project folder.
> 
> I consider "export" means checkout the project into a specific folder, the project is still git based and remains git based.

Yes, exactly. It's just the name of the project directory in the local file system.

`git clone <url> <directory>`

The name chosen for `<directory>` is then used by `Sysbuild` as the name for the main image.

boaks, 2025-05-12:
... and it's unclear, which additional rules (allowed characters) for that name needs **now** to be obeyed.

nordicjm, 2025-05-13:
> Is that OK? For NCS 2.9.1 / zephyr 3.7.99 I don't see, that "MY_MACRO" gets defined.

Because you are not setting a config, you need to use e.g. `-Dmy-app_CONFIG_ZCBOR=y`

nordicjm, 2025-05-13:
> ... and it's unclear, which additional rules (allowed characters) for that name needs **now** to be obeyed.

It's the name of the folder, using special characters (excluding _ and -) is highly discouraged and going to break if you use spaces or semicolons

boaks, 2025-05-13:
> you need to use e.g. -Dmy-app_CONFIG_ZCBOR=y

Thanks for the clarification about the directory name! That helps a lot. 

`CONFIG_` is related to Kconfig and for such variables it works. For the application itself it would even work with prefix, see [CMake variable namespacing](https://docs.zephyrproject.org/latest/build/sysbuild/index.html#cmake-variable-namespacing). 

But for non `CONFIG_` macros it doesn't work on "my machine". 

CMakeLists.txt
```
...
add_custom_command(TARGET app POST_BUILD
                   COMMAND echo "Build with ${MY_MACRO}"
                   COMMAND echo "Build with ${CONFIG_MY}"
                   )
...
```

```
west build ... -- -DCONFIG_MY=2 -DMY_MACRO=1 

Build with 
Build with 2

west build ... -- -Dmy-app_CONFIG_MY=4 -Dmy-app_MY_MACRO=3 

Build with 
Build with 4
```

Could you please retest/verify the usage of passing in some non "CONFIG" stuff?
That may help also for a similar issue #88303

nordicjm, 2025-05-13:
You would need to use `zephyr_get()` with `SYSBUILD` to get the variable https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/extensions.cmake#L3183

boaks, 2025-05-13:
Again thanks a lot for your clarification and help!

But I would enjoy it very much, if it would be possible to "overwrite" the "directory name" as "domain name" in a future version.

nordicjm, 2025-05-13:
> Again thanks a lot for your clarification and help!

Anytime

> But I would enjoy it very much, if it would be possible to "overwrite" the "directory name" as "domain name" in a future version.

No plans to support such a feature at this time