zephyrproject-rtos/zephyr #82701: soc: st: stm32mp13 series support

arifbalik, 2024-12-08:
## Introduction

ST Recently released their own version for STM32MP13 series bare metal development. But Zephyr currently only has support for STM32MP15 series. If there is no one working on this I would like to try and implement it.

### Problem description

Currently applications can not be build on STM32MP13 series. There is a Kconfig `SOC_SERIES_STM32MP1X` but it select symbols only available in STM32MP15xx series and not on MP13xx series

### Proposed change

I would like to define the SOC and [STM32MP135F-DK](https://www.st.com/en/evaluation-tools/stm32mp135f-dk.html) board, which is the only board I have atm.

## Detailed RFC

The SOC directory inside `hal_stm32` repo does not contain header files for stm32mp13xx series but it exists at https://github.com/STMicroelectronics/cmsis_device_mp13.git

### Proposed change (Detailed)

After updating the `hal_stm32` I would follow the standard practice for defining the SOC, but for board I'm not sure which drivers are missing, but I do not think much. The board I'll be using has all security features inside the chip, that part may require some work.

erwango, 2024-12-09:
@arifbalik There is actually a port initiated and which should be pushed upstream soon.

arifbalik, 2024-12-09:
> @arifbalik There is actually a port initiated and which should be pushed upstream soon.

Is there a branch we can follow maybe?

arnopo, 2024-12-09:
@arifbalik ,  we have started the porting but nothing sharable for time being (at PoC level)
When do you expect to start your work on stm32mp13?

arnopo, 2024-12-10:
We will try to find time to share something with you as soon as possible (by the end of this week or next week). Even if it is not clean yet, it should help you. I will keep you informed.

arifbalik, 2024-12-10:
> We will try to find time to share something with you as soon as possible (by the end of this week or next week). Even if it is not clean yet, it should help you. I will keep you informed.

That would be very nice, I would really like to contribute whereever possible. 

I already worked with stm32mp1 series so I'm familiar with it

arnopo, 2024-12-16:
Hi @arifbalik 

As discussed, please find below two links to my GitHub that allow you to run the sample blinky on the STM32MP13 board.

This code is not yet clean enough to be upstreamed, but we expect to do it in January.

- the HAL is available here: https://github.com/arnopo/hal_stm32/tree/stm32mp13xx
- the Zephyr STM32MP13 support is available here: https://github.com/arnopo/zephyr_stm32mp1/tree/stm32mp135f_dk
- the documentation entry point is available here: https://github.com/arnopo/zephyr_stm32mp1/blob/stm32mp135f_dk/boards/st/stm32mp135f_dk/doc/stm32mp135_dk.rst

Current status:
- you need to load a stm32CUbe package on SDcard to be able to initialize the DDR. Then boot the board and you will be able to load zephyr application using `west debug`
- No UART console yet 

If you want to contribute, please tell us which part you want to work on so that we can avoid duplicating efforts. On our side, we will probably focus first on making the SoC and board parts clean, so don't hesitate if you want to enable UART or some other peripherals or to focus on some frameworks such as USB, LVGL.

Thanks!

arifbalik, 2024-12-19:
@arnopo Just a quick question, I got the blinky working on my board, but whenever I enable my newly defined uart, and the driver `uart_stm32`, I get an exception as soon as the driver tries to access the memory region of the uart4 (`0x40010000`). Spesifically on this line;

https://github.com/zephyrproject-rtos/zephyr/blob/3fa522c720da00c0330e7d14c5882c191b6c0971/drivers/serial/uart_stm32.c#L1989

Which just writes to the `CR` register. After this we drop into

https://github.com/zephyrproject-rtos/zephyr/blob/3fa522c720da00c0330e7d14c5882c191b6c0971/arch/arm/core/cortex_a_r/vector_table.S#L26

Which means this is a weird memory region we are not allowed to work on? I took a quick look at the ext loader to see if there is a configuration for enabling the core for a region of memory (like in optee) but I think not.

I don't have an svd file so I can not snapshot the registers for RCC and UART4 but it seems like we have to enable access to this peripheral as far as I understand it. Am I missing something here?

arnopo, 2024-12-20:
There is probably an issue in the MMU configuration. I have updated the MMU table to declare all the peripheral buses and also moved the code to DDR, as it was running in SYSRAM.
please see my Zephyr branch: https://github.com/arnopo/zephyr_stm32mp1/tree/stm32mp135f_dk
I hope this will solve your issue

arifbalik, 2024-12-21:
> There is probably an issue in the MMU configuration. I have updated the MMU table to declare all the peripheral buses and also moved the code to DDR, as it was running in SYSRAM. please see my Zephyr branch: [arnopo/zephyr_stm32mp1@`stm32mp135f_dk`](https://github.com/arnopo/zephyr_stm32mp1/tree/stm32mp135f_dk) I hope this will solve your issue

That was it, thanks. 

I now have a console. But my workspace is kinda messy right now. I'll refactor the RCC part and open a PR

sabourin-aa, 2025-03-24:
@arnopo My work depends on the stm32mp13 support. Do you know when support for this will be merged?

arnopo, 2025-03-25:
> [@arnopo](https://github.com/arnopo) My work depends on the stm32mp13 support. Do you know when support for this will be merged?

We are finalizing internal reviews we expect to post the PR this week or next week

arnopo, 2025-03-26:
A pull request has just been sent: https://github.com/zephyrproject-rtos/zephyr/pull/87687

I linked the PR to this issue to automatically close it when merged

closed 2025-04-04 by kartben