zephyrproject-rtos/zephyr #82711: tests: drivers.gpio: 2pin_aw9523b: the filter is not filter out not support platforms, cause many build failure

hakehuang, 2024-12-09:
**Describe the bug**

drivers.gpio.2pin_aw9523b build failure for NXP platforms

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - NXP mimxrt10xx series platforms
 - What have you tried to diagnose or workaround this issue?
 - git biect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - 

```
59a82ab3575baabb8ebb1dcae8317b81a8679624
tests: drivers: gpio: gpio_api_1pin: Add AW9523B test configuration

Add test configuration for Awinic AW9523B.

This device is a GPIO Expander, and no boards currently implement it.
Define the connection that connects it via the Arduino header and
define a fixture to enable it.

This has been confirmed with nucleo_h503rb.

```

**To Reproduce**

**Expected behavior**

**Impact**

CI clean

**Logs and console output**

```
2024-12-08 06:42:01,413 - __main__ -zephyr_ci_build_deploy_test.py : 188 - INFO - -- Found BOARD.dts: /home/jenkins/agent/workspace/zephyr_test_pipe/zephyr/boards/nxp/mimxrt1020_evk/mimxrt1020_evk.dts
2024-12-08 06:42:01,413 - __main__ -zephyr_ci_build_deploy_test.py : 188 - INFO - -- Found devicetree overlay: boards/aw9523b_on_arduino_header.overlay
2024-12-08 06:42:01,413 - __main__ -zephyr_ci_build_deploy_test.py : 188 - INFO - devicetree error: ./boards/aw9523b_on_arduino_header.overlay:1 (column 1): parse error: undefined node label 'arduino_i2c'
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-1725-gbea5fc71c495

hakehuang, 2024-12-09:
@dleach02

fabiobaltieri, 2024-12-10:
@soburi could you take a look? looks like this was due to the AW9523B PR. Thanks.

soburi, 2024-12-10:
@hakehuang 

Could you tell me what command caused this issue?
This test case is required to specify "fixture," so I expect it to not run in the usual case.

fabiobaltieri, 2024-12-10:
@hakehuang if you could provide a command to reproduce the specific build failure it'd help a lot slipstreaming the troubleshooting process

dleach02, 2024-12-10:
One failure (which is blocking CI for us):

`west build -p -b mimxrt1050_evk tests/drivers/gpio/gpio_basic_api -T drivers.gpio.2pin_aw9523b`

soburi, 2024-12-11:
I would like to proceed with reverting this for now.
The reason for this is that it is difficult to determine whether or not a shield is present. (I thought this could be determined by whether or not a fixture is present, but it seems I was wrong.)
I don't think I will have time to work on it today, so I think I will be able to get it done by tomorrow.

soburi, 2024-12-11:
@dleach02 

By defining a dependency on `arduino_i2c` in `depends_on`, it seems possible to exclude it and run it in twister.

You've just given the west build command as an example of how to run it, but can you confirm whether it is necessary to run it successfully with this instead of twister?
If there is a problem with CI, I think it will be sufficient if the execution with twister is successful.

hakehuang, 2024-12-11:
> You've just given the west build command as an example of how to run it, but can you confirm whether it is necessary to run it successfully with this instead of twister?
> If there is a problem with CI, I think it will be sufficient if the execution with twister is successful.

this case is not appliable to our platform, and according to your current filter in testcase.yaml, NXP platform can not skipped even in build, so add a depends_on 'arduino_i2c'

hakehuang, 2024-12-11:
> Could you tell me what command caused this issue?
> This test case is required to specify "fixture," so I expect it to not run in the usual case.

it is build failure, the fixture is used for run, so it will not filter NXP platform our for build in CI

soburi, 2024-12-11:
> > You've just given the west build command as an example of how to run it, but can you confirm whether it is necessary to run it successfully with this instead of twister?
> > If there is a problem with CI, I think it will be sufficient if the execution with twister is successful.
> 
> this case is not appliable to our platform, and according to your current filter in testcase.yaml, NXP platform can not skipped even in build, so add a depends_on 'arduino_i2c' is recommended

OK, I will address with adding depends_on 'arduino_i2c' way.
I'm not in a situation where I can respond immediately, but I think I can respond within 24 hours.

arifbalik, 2024-12-11:
I dont know if related but recently I got test errors about `aw9523b` as well, but with with a different test application; I patched a fix for my case; https://github.com/zephyrproject-rtos/zephyr/pull/82869

PR #82865 (2024-12-14): tests: drivers: gpio: Add a dependency to the aw9523b tests.
To avoid compilation on boards that do not have arduino_i2c defined, we will clarify that it is dependent on it.

Fix #82711
closed 2024-12-14 by kartben