zephyrproject-rtos/zephyr #82712: tests: drivers.pwm.pwm_basic: mimxrt1180_evk_cm33: build failure for pwm test

hakehuang, 2024-12-09:
**Describe the bug**

<html>
<body>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
drivers.pwm.pwm_basic.pwm_invalid_port | v4.0.0-1725-gbea5fc71c495 | CMake build failure | mimxrt1180_evk/mimxrt1189/cm33 | drivers.pwm
drivers.pwm.pwm_basic.pwm_nsec | v4.0.0-1725-gbea5fc71c495 | CMake build failure | mimxrt1180_evk/mimxrt1189/cm33 | drivers.pwm
drivers.pwm.pwm_basic.pwm_cycle | v4.0.0-1725-gbea5fc71c495 | CMake build failure | mimxrt1180_evk/mimxrt1189/cm33 | drivers.pwm
</body>
</html>

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1180_evk_cm33
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?

d2040d103a30585e426268ad3611ebb326c33e8a
```
boards： nxp: mimxrt1180_evk： add TPM support
add TPM pin setting
Record TPM supported in yml and doc

```

**To Reproduce**

**Expected behavior**

build pass

**Impact**

PWM/TMP

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-1725-gbea5fc71c495

fabiobaltieri, 2024-12-10:
@lucien-nxp hey, could you take a look?

lucien-nxp, 2024-12-11:
I had tried to resolve the issue yesterday. #82761.

fabiobaltieri, 2024-12-11:
Cool sounds like this is done then, thanks.

closed 2024-12-11 by fabiobaltieri