zephyrproject-rtos/zephyr #82717: UART isr deadlock on uart_irq_rx_enable() on stm32

xhpohanka, 2024-12-09:
**Describe the bug**
I'm not sure if it is a bug or a misuse but I have met a deadlock using the same approach I have seen in zephyr codebase.
This happend with stm32h750 and  zephyr 3.5.0

I have uart isr storing received data with following loop
```c
while (uart_irq_update(uart) &&  uart_irq_is_pending(uart)) {
	int len;
	int rd_len = 0;

	uint8_t *data;
	len = ring_buf_put_claim(&rx_ring_buf, &data, rx_ring_buf.size);

	if (len > 0) {
		rd_len = uart_fifo_read(dev, data, len);

		ring_buf_put_finish(&rx_ring_buf, rd_len);
	}
	else {
		LOG_WRN("rx buf full");

		uint8_t dummy;
		rd_len = uart_fifo_read(uart_dev, &dummy, 1);
	}
}
``` 

then there is a thread consuming the data

```c
uart_irq_rx_disable(uart_dev);
len = ring_buf_get_claim(&rx_ring_buf, (uint8_t **)&data, rx_ring_buf.size);
uart_irq_rx_enable(uart_dev);

proces(data);
...
```

The critical section around claiming the buffer is maybe superfluous but shouldn't cause a problem I think.
What I see is that sometimes I end in an infinite loop in my isr after `uart_irq_rx_enable()` call. In that state `uart_irq_is_pending()` returns false. 

On stm32 this function tests `LL_USART_IsActiveFlag_RXNE() && LL_USART_IsEnabledIT_RXNE()`. But `LL_USART_IsEnabledIT_RXNE()` is still false during the lock. Can it be some caching issue? ISR seems to be fired sooner that the corresponding bit is set...

erwango, 2024-12-09:
> This happend with stm32h750 and zephyr 3.5.0

Note that v3.5.0 support ended end July (see https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management).
Would you mind testing on a more recent release (as v3.7.0) ?

xhpohanka, 2024-12-09:
> Would you mind testing on a more recent release (as v3.7.0) ?

It's not so easy for me right now, I'll try if I have some spare time...

djiatsaf-st, 2024-12-18:
Hello, 

I have performed tests on the `stm32h750b_dk` with this code example implementing your callback function, and I have not encountered any problems.

```

#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/sys/printk.h>
#include <zephyr/drivers/uart.h>
#include <zephyr/sys/ring_buffer.h>

LOG_MODULE_REGISTER(ring_app);

#define UART_DEVICE DT_NODELABEL(usart3) 

#define RING_BUF_SIZE 500

RING_BUF_DECLARE(rx_ringbuf, RING_BUF_SIZE);

const struct device *uart_dev= DEVICE_DT_GET(UART_DEVICE);

void print_uart(char *buf, int len)
{
	for (int i = 0; i < len; i++) {
		uart_poll_out(uart_dev, buf[i]);
	}
}

static void uart_cb(const struct device *dev, void *user_data)
{
    while (uart_irq_update(dev) &&  uart_irq_is_pending(dev)) {
        int len;
        int rd_len = 0;

        uint8_t *data;
        len = ring_buf_put_claim(&rx_ringbuf, &data, rx_ringbuf.size);

        if (len > 0) {
            rd_len = uart_fifo_read(dev, data, len);
            ring_buf_put_finish(&rx_ringbuf, rd_len);
        }
        else {
            LOG_WRN("rx buf full");

            uint8_t dummy;
            rd_len = uart_fifo_read(uart_dev, &dummy, 1);
        }
    }
}

int main(void)
{
    int ret, len;

    if (!uart_dev) {
        LOG_ERR("Cannot get UART device");
    }

    ret = uart_irq_callback_user_data_set(uart_dev, uart_cb, NULL);

    if( ret < 0){
        printk("failed to set uart callback \n");
    }

    uart_irq_rx_enable(uart_dev);

    while (1) {

            uint8_t *data;
            len = ring_buf_get_claim(&rx_ringbuf, (uint8_t **)&data, rx_ringbuf.size);
            if (len > 0) {
                print_uart(data, len);
            }

    }
}
```

> `stm32h750b_dk.overlay`
> 
> 
> &usart3 {
> 	dmas = <&dmamux1 2 46 STM32_DMA_PERIPH_TX>,
> 		<&dmamux1 3 45 STM32_DMA_PERIPH_RX>;
> 	dma-names = "tx", "rx";
> 
> };
> 
> &dmamux1 {
> 	status = "okay";
> };
> 
> &dma1 {
> 	status = "okay";
> };
> 
> &dma2 {
> 	status = "okay";
> };

> `prj.conf` 
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_RING_BUFFER=y

The example echoes back the information entered in the terminal with the `print_uart()` function.

Zephyr version used: `v4.0.0`.

xhpohanka, 2024-12-19:
hi @djiatsaf-st 
I have probably described it wrongly in my first post but in your test code you are missing the critical section, which was triggering the issue for me

```c
...
	uart_irq_rx_disable(uart_dev);
	len = ring_buf_get_claim(&rx_ring_buf, (uint8_t **)&data, rx_ring_buf.size);
	uart_irq_rx_enable(uart_dev);
...
```

djiatsaf-st, 2024-12-19:
> hi @djiatsaf-st I have probably described it wrongly in my first post but in your test code you are missing the critical section, which was triggering the issue for me
>
> ```c
> ... [...]
Maybe I don't have all the context of your application, but with these two lines , the test code I provided works fine, no issue encountered.
I think you should consider testing on a more recent Zephyr release.

erwango, 2024-12-19:
> I think you should consider testing on a more recent Zephyr release.

Indeed, Zephyr v3.5.0 is not supported anymore (since 06'24).

closed 2025-03-31 by github-actions[bot]