zephyrproject-rtos/zephyr #82726: LOG_ macro redefinition

zxtxin, 2024-12-09:
When I include both zephyr/posix/syslog.h and zephyr/net/socket.h，there is a "LOG_ERR" redefined error

the include path chain is
zephyr/net/socket.h -> zephyr/net/dns_resolve.h -> zephyr/net/net_core.h -> zephyr/logging/log.h

jukkar, 2024-12-09:
The error looks like this:
```
In file included from zephyr/tests/net/lib/dns_dispatcher/src/main.c:20:
zephyr/include/zephyr/posix/syslog.h:42: warning: "LOG_ERR" redefined
   42 | #define LOG_ERR     3
      | 
In file included from zephyr/tests/net/lib/dns_dispatcher/src/main.c:7:
zephyr/include/zephyr/logging/log.h:44: note: this is the location of the previous definition
   44 | #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
      | 
```

cfriedt, 2024-12-09:
Yeah, it's a known issue that is unfortunately not an easy issue to fix.

`LOG_ERR` is a standard macro defined by POSIX. However, we have used it within Zephyr for a long time as part of the logging subsystem.

We would likely need to change the `LOG_ERR()` -> `LOG_ERROR()` macro in the logging subsystem to permanently fix the issue. 

Please see samples / tests for the conventional workaround.

rruuaanng, 2024-12-10:
Maybe we can undefine it in the environment by undefing it, then defining it again? Like this
```
#undef LOG_ERR
#define LOG_ERR /* ignore body */
```
I haven't tried whether this works.

closed 2025-02-23 by github-actions[bot]