zephyrproject-rtos/zephyr #82729: BLE pairing fails on the new NRF54L15DK

maver-BL, 2024-12-09:
**Describe the bug**

BLE pairing on the new NRF54L15DK does not succeed. Connecting and advertising works, only pairing fails. I have tried the same example on the NRF52840DK and it works. Also tried pairing using Android and iOS (Samsung and Iphone 16), same result is achieved.

**To Reproduce**

1. Flash the Bluetooth fundamentals lesson 3 - exercise 1 example or any BLE pairing example (https://academy.nordicsemi.com/courses/bluetooth-low-energy-fundamentals/lessons/lesson-3-bluetooth-le-connections/topic/blefund-lesson-3-exercise-1/)
2. Try to pair to the device
3. See that it won't pair

**Expected behavior**

Should pair with the device as it does on the nrf52840dk.

**Logs and console output**

[00:00:00.433,931] <inf> Lesson5_Exercise1: Advertising successfully started

[00:00:19.862,201] <inf> Lesson5_Exercise1: Connected

[00:00:19.862,282] <wrn> net_buf: Timeout discarded. No blocking in syswq
...
[00:00:54.337,414] <err> bt_smp: SMP Timeout
[00:00:54.337,435] <wrn> bt_smp: The in-progress pairing has been deleted!
[00:00:54.337,490] <inf> Lesson5_Exercise1: Security failed: XX:XX:XX:XX:XX:XX (random) level 1 err 9

[00:00:57.428,238] <inf> Lesson5_Exercise1: Disconnected (reason 19)

**Environment (please complete the following information):**

 - OS: Linux - Manjaro
 - Zephyr SDK
 - v4.0-branch

maver-BL, 2024-12-09:
On the newest NCS (v2.9.99) pairing works using SoftDevice...

cvinayak, 2024-12-10:
nRF54L15 does not support entropy, crypto and encryption yet in the Zephyr Project repository.

fabiobaltieri, 2024-12-10:
Seems like this has been answered, closing down.

closed 2024-12-10 by fabiobaltieri