zephyrproject-rtos/zephyr #82733: samples/subsys/tracing/ sample.tracing.gpio fails on stm32 boards with gpio_loopback fixture

FRASTM, 2024-12-09:
**Describe the bug**
Running the    samples/subsys/tracing/sample.tracing.gpio   on stm32 boards where the fixture gpio_loopback is set 
will fail.

--> the **gpio_loopback** is set for the target board in order to pass tests/drivers/i2s/i2s_api or test/drivers/gpio/gpio_basic_api/ or tests/drivers/uart/uart_async_api

When executing the samples/subsys/tracing/sample.tracing.gpio, the console gives the following output which is not declared as "PASSED" :

```
...
sys_trace_idle_user
sys_trace_idle_user
thread_a: Hello World from nucleo_g474re!
sys_trace_idle_user
sys_trace_idle_user
sys_trace_idle_user
sys_trace_idle_user
sys_trace_idle_user
sys_trace_idle_user
thread_b: Hello World from nucleo_g474re!
sys_trace_idle_user
...

```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_g474re samples/subsys/tracing  -DCONF_FILE="prj_gpio.conf" -DEXTRA_DTC_OVERLAY_FILE="gpio.overlay" 
2. west flash

**Expected behavior**
Should the testcase be executed ?

**Impact**
Test camapign fails on each target board where the gpio_loopback fixture is set

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)   ubuntu 
 - Toolchain (e.g Zephyr SDK, ...)  SDK 0.16.8
 - Commit SHA or Version used  ef6f804d8f2648935afadaab09b511913ffd9f4c

masz-nordic, 2024-12-10:
Incorrect assignment, this bug is STM specific.

katgiadla, 2024-12-10:
The same issue is visible on:
- `nrf52dk/nrf52832`
- `nrf52840dk/nrf52840`
- `nrf5340dk/nrf5340/cpuapp`
- `nrf9160dk@0.14.0/nrf9160`
- `nrf54l15dk/nrf54l15/cpuapp`
- `nrf54l15dk/nrf54l05/cpuapp`
- `nrf54l15dk/nrf54l10/cpuapp`
- `nrf54l15dk/nrf54l15/cpuflpr`

fabiobaltieri, 2024-12-10:
should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/82648, tentatively closing

katgiadla, 2024-12-16:
> The same issue is visible on:
nrf52dk/nrf52832
nrf52840dk/nrf52840
nrf5340dk/nrf5340/cpuapp
nrf9160dk@0.14.0/nrf9160
nrf54h20dk@0.9.0/nrf54h20/cpuapp
nrf54h20dk@0.9.0/nrf54h20/cpurad

The issue still occurs on the all nordic platforms.

fabiobaltieri, 2024-12-16:
@ydamigos can you take a look?

ydamigos, 2024-12-16:
@fabiobaltieri  It seems to be a different issue. The output prints continuously:
```
sys_trace_isr_enter_user: 0
sys_trace_isr_exit_user: 0
...
```
It look likes it stays in an IRQ loop.

fabiobaltieri, 2024-12-16:
thanks @ydamigos, @katgiadla can you reopen a different one? (just to be clear I'm just triaging here)

closed 2024-12-16 by fabiobaltieri