zephyrproject-rtos/zephyr #82757: subsys:mgmt:mcumgr: MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE is less than sizeof(struct sockaddr) while NET_SOCKETS_PACKET has been selected.

hongquan-prog, 2024-12-10:
**Describe the bug**
When I tested the MCUMGR functionality with the nrf7002, the compiler's static assertions prevented me from compiling.
```c
BUILD_ASSERT(sizeof(struct sockaddr) <= CONFIG_MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE,
	     "CONFIG_MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE must be >= sizeof(struct sockaddr)");
```

**To Reproduce**
Enabling MCUMGR with the NRF7002 wireless card

**Expected behavior**
Build success and running.

**Impact**
Compiling failed.

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain: zephyr-sdk-0.17.0
 - latest main branch

**Additional context**
1. NET_SOCKETS_PACKET Indirectly selected by NRF70_SYSTEM_MODE

```
config NRF70_SYSTEM_MODE
	bool "nRF70 system mode"
	depends on WIFI_NRF7002 || WIFI_NRF7001
	select WIFI_NM_WPA_SUPPLICANT
	help
	  Select this option to enable system mode of the nRF70 driver

config WIFI_NM_WPA_SUPPLICANT
	bool "WPA Suplicant from hostap project [EXPERIMENTAL]"
	select POSIX_TIMERS
	select POSIX_SIGNALS
	select POSIX_API
	select FILE_SYSTEM
	select NET_SOCKETS
	select NET_SOCKETS_PACKET
	select NET_SOCKETPAIR
	select NET_L2_WIFI_MGMT
	select WIFI_NM
	select EXPERIMENTAL
	select COMMON_LIBC_MALLOC
	help
	  WPA supplicant as a network management backend for WIFI_NM.
```

2. When NET_SOCKETS_PACKET is on, NET_SOCKADDR_MAX_SIZE will become 20 and sizeof(struct sockaddr) will also become 20.

```c
#if defined(CONFIG_NET_IPV4)
#undef NET_SOCKADDR_MAX_SIZE
#undef NET_SOCKADDR_PTR_MAX_SIZE
#define NET_SOCKADDR_MAX_SIZE (sizeof(struct sockaddr_in))
#define NET_SOCKADDR_PTR_MAX_SIZE (sizeof(struct sockaddr_in_ptr))
#endif

#if defined(CONFIG_NET_SOCKETS_PACKET)
#undef NET_SOCKADDR_MAX_SIZE
#undef NET_SOCKADDR_PTR_MAX_SIZE
#define NET_SOCKADDR_MAX_SIZE (sizeof(struct sockaddr_ll))
#define NET_SOCKADDR_PTR_MAX_SIZE (sizeof(struct sockaddr_ll_ptr))
#endif

struct sockaddr {
	sa_family_t sa_family; /**< Address family */
/** @cond INTERNAL_HIDDEN */
	char data[NET_SOCKADDR_MAX_SIZE - sizeof(sa_family_t)];
/** @endcond */
};
```

3. When MCUMGR_TRANSPORT_UDP and MCUMGR_TRANSPORT_UDP_IPV4 are enabled, the value of MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE is 8.
```
config MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE
	int
	default 24 if MCUMGR_TRANSPORT_UDP && NET_IPV6
	default 8 if MCUMGR_TRANSPORT_UDP && MCUMGR_TRANSPORT_UDP_IPV4
	default 8 if MCUMGR_TRANSPORT_BT
	default 4
	help
	  Hidden option to determine minimum user data size.

config MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE
	int "Size of mcumgr buffer user data"
	range MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE 128
	default MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE
	help
	  The size, in bytes, of user data to allocate for each mcumgr buffer.

	  Different mcumgr transports impose different requirements for this
	  setting. A value of 4 is sufficient for UART and shell, a value of 8
	  is sufficient for Bluetooth. For UDP, the userdata must be large
	  enough to hold IPv4/IPv6 addresses.
```
4. Static assertion fails and stops compilation.
5. In addition to this, the compiler also suggests that the following part of the code is out of bounds. This is due to the unequal size of the two structures.
[image]
```c
static int create_socket(enum proto_type proto, int *sock)
{
	int tmp_sock;
	int err;
	struct sockaddr *addr;

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV4
	struct sockaddr_in addr4;
#endif

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV6
	struct sockaddr_in6 addr6;
#endif

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV4
	if (proto == PROTOCOL_IPV4) {
		memset(&addr4, 0, sizeof(addr4));
		addr4.sin_family = AF_INET;
		addr4.sin_port = htons(CONFIG_MCUMGR_TRANSPORT_UDP_PORT);
		addr4.sin_addr.s_addr = htonl(INADDR_ANY);
		addr = (struct sockaddr *)&addr4;
	}
#endif

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV6
	if (proto == PROTOCOL_IPV6) {
		memset(&addr6, 0, sizeof(addr6));
		addr6.sin6_family = AF_INET6;
		addr6.sin6_port = htons(CONFIG_MCUMGR_TRANSPORT_UDP_PORT);
		addr6.sin6_addr = in6addr_any;
		addr = (struct sockaddr *)&addr6;
	}
#endif

	tmp_sock = zsock_socket(addr->sa_family, SOCK_DGRAM, IPPROTO_UDP);
	err = errno;

	if (tmp_sock < 0) {
		LOG_ERR("Could not open receive socket (%s), err: %i",
			smp_udp_proto_to_name(proto), err);

		return -err;
	}

	if (zsock_bind(tmp_sock, addr, sizeof(*addr)) < 0) {
		err = errno;
		LOG_ERR("Could not bind to receive socket (%s), err: %i",
			smp_udp_proto_to_name(proto), err);

		zsock_close(tmp_sock);

		return -err;
	}

	*sock = tmp_sock;
	return 0;
}
```
Here are my changes：
```c
static int create_socket(enum proto_type proto, int *sock)
{
	int tmp_sock;
	int err;
	struct sockaddr addr;

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV4
	struct sockaddr_in *addr4;
#endif

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV6
	struct sockaddr_in6 *addr6;
#endif

	memset(&addr, 0, sizeof(struct sockaddr));
#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV4
	if (proto == PROTOCOL_IPV4) {
		addr4 = (struct sockaddr_in *)&addr;
		addr4->sin_family = AF_INET;
		addr4->sin_port = htons(CONFIG_MCUMGR_TRANSPORT_UDP_PORT);
		addr4->sin_addr.s_addr = htonl(INADDR_ANY);
	}
#endif

#ifdef CONFIG_MCUMGR_TRANSPORT_UDP_IPV6
	if (proto == PROTOCOL_IPV6) {
		addr6 = (struct sockaddr_in6 *)&addr;
		addr6->sin6_family = AF_INET6;
		addr6->sin6_port = htons(CONFIG_MCUMGR_TRANSPORT_UDP_PORT);
		addr6->sin6_addr = in6addr_any;
	}
#endif

	tmp_sock = zsock_socket(addr.sa_family, SOCK_DGRAM, IPPROTO_UDP);
	err = errno;

	if (tmp_sock < 0) {
		LOG_ERR("Could not open receive socket (%s), err: %i",
			smp_udp_proto_to_name(proto), err);

		return -err;
	}

	if (zsock_bind(tmp_sock, &addr, sizeof(addr)) < 0) {
		err = errno;
		LOG_ERR("Could not bind to receive socket (%s), err: %i",
			smp_udp_proto_to_name(proto), err);

		zsock_close(tmp_sock);

		return -err;
	}

	*sock = tmp_sock;
	return 0;
}
```

jukkar, 2024-12-10:
@hongquan-prog you should not instantiate a variable that is type `struct sockaddr` so the original code is correct in this respect. It is the length of the sockaddr in `zsock_bind()` that is wrong, it should be set to the real length of the sockaddr (so either `sizeof(struct sockaddr_in)` or `sizeof(struct sockaddr_in6)`

hongquan-prog, 2024-12-10:
> @hongquan-prog you should not instantiate a variable that is type `struct sockaddr` so the original code is correct in this respect. It is the length of the sockaddr in `zsock_bind()` that is wrong, [...]

Okay, the compiler didn't hint at the correct location, so I didn't focus on the bind function

hongquan-prog, 2024-12-10:
> @hongquan-prog you should not instantiate a variable that is type `struct sockaddr` so the original code is correct in this respect. It is the length of the sockaddr in `zsock_bind()` that is wrong, [...]

Even though this seems fine, the compilation still has a warning. If the relevant test case exists, the CI will treat it as an error.

PR #82786 (2024-12-16): subsys:mgmt:mcumgr: Fix mcumgr assertion error
In the case of IPV6 not enabled, when NET_SOCKETS_PACKET is enabled, the sizeof(struct sockarr) will change to 20, the value of MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE is 8, which is not able to pass the compilation, so I change its default value to 20.

Fixes #82757
commit bda1e767 (2024-12-16): Subsys: mgmt: mcumgr: Fix mcumgr assertion error

In the case of IPV6 not enabled, when NET_SOCKETS_PACKET is enabled,
the sizeof(struct sockarr) will change to 20, the value of
MCUMGR_TRANSPORT_NETBUF_MIN_USER_DATA_SIZE is 8, which is not able
to pass the compilation, so I change its default value to 20.

Fixes #82757
closed 2024-12-16 by kartben