zephyrproject-rtos/zephyr #82818: Zephyr 3.7.0 spi_nor driver assumes 3-byte mode when querying for SFDP

thavelka-cornell, 2024-12-10:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/82511

<div type='discussions-op-text'>

<sup>Originally posted by **thavelka-cornell** December  3, 2024</sup>
Hello Zephyr Team,

I am working on a work related project using Zephyr 3.7.0 LTS, targeting an NXP LPC family MCU.  We recently changed an external SPI NOR flash device from a 16MB part to a 32MB part.  Now I observe that our application from an unpowered start up detects the flash the first time.  On successive warm starts, e.g. MCU resets, external flash initialization fails.

At the application level the failure occurs with the call to Zephyr's `device_is_ready()`, to which we pass our `struct device* flash_dev` pointer.

Probing with a Saleae logic analyzer on the SPI bus signals between MCU and flash part reveals that on cold start up, a query for SFDP succeeds as both MCU and flash part are operating in 3-byte mode.  The "read SFDP" call happens twice, first with offset 0x000000, then soon after with offset 0x000300.  A final single byte command is then sent to put the flash chip into 4-byte addressing modes.

On warm resets, our application now fails to detect the 32MB external SPI NOR flash, because it holds the recent config setting from Zephyr SPI NOR driver which puts it in 4-byte addressing mode.

Upon warm restart the external flash device is now in 4-byte mode, but routine spi_nor_process_sfdp()` at `https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.c#L1135 appears to assume a 3-byte addressing mode.  This results in a leading byte coming back in the SFDP data read.  The leading byte corrupts the data, and the driver deems external flash device not present / not ready.

Reviewing Zephyr's in-tree SPI NOR flash driver, I do not see any Kconfig symbols which would permit an application to disable the driver from setting flash devices into 4-byte mode when the flash part supports this.

Some information I don't know is whether JEDEC compliant SPI NOR flash devices generally show the behavior I am seeing in the particular flash device in my team's design.  We have an external flash part by Infineon.  It is a 32MB SPI NOR flash device with JEDEC ID 0x01 0x60 0x19.

In a local patch I applied to spi_nor.c I was able to force our external flash part into 3-byte addressing mode near the end of https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.c#L1615, and this solves our problem at least superficially.  We can now run our application normally. starting from both cold and warm reboots.

Three questions, if appropriate to ask:

Question (1) which SPI NOR flash parts has Zephyr's flash driver been tested against?

Question (2) where may community members review the test plan and results of such Zephyr driver tests?

Question (3) has this SPI NOR driver subsystem been tested on hardware both from cold start and warm start conditions?

- Ted</div>

thavelka-cornell, 2024-12-12:
In looking for where and under what conditions Zephyr's SPI NOR driver puts a JEDEC flash device into 4-byte mode, I found https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.h#L46.  This command is mentioned in Infineon's datasheet for its S25FL256L and larger flash devices.

In the SPI NOR header file there is a define for the command to enter 4-byte mode, but no symbol defined for the complimentary "exit 4 byte mode" command.  At least for Infineon JEDEC flash devices there is such a command.

Are the flash memory commands defined in `spi_nor.h` universal across many JEDEC compliant memories?  If there is a Common Flash Interface command to exit 4-byte mode, this header file looks like the place to define such a command.

maass-hamburg, 2025-01-10:
@thavelka-cornell  Please check the jedec standard https://www.jedec.org/standards-documents/docs/jesd216b here. You may need to create a account to access, but that is free. In section 4.2 it clearly states, that for 1S-1S-1S (that is normal spi with one miso and one mosi) only a 3-byte address is to be used

thavelka-cornell, 2025-01-10:
Hello Maass Hamburg.  Thank you for the JEDEC standards link.  I have created an account there and downloaded this document.  Perhaps though I have mis-stated my problem, or my role relative to the code in which I discovered a possible problem:  I am an end user of Zephyr's SPI NOR flash API, not a developer of this driver code base.  From what code review I have done however, it appears that there is significant code in the driver to switch to and use 4-byte mode with JEDEC flash devices, possibly without checking for the physical type of SPI connection.

Our particular flash device is SPI NOR type, and is JEDEC compliant.  I believed that these facts were sufficient to qualify us for using Zephyr's SPI NOR driver.  When using Zephyr drivers, we face some levels of indirection from the lines of Zephyr driver code itself.  We express our flash device in device tree sources, of course.  Those DTS files describe whether we are using 1S-1S-1S SPI bus in our circuit, or dual-SPI or quad-SPI.  We are using single wire, 1S-1S-1S SPI.

In Kconfig project files we enable the flags to indicate we want to pull in Zephyr's JEDEC flash support.  From there, we normally assume and get to use a driver which functions "out of the box".

Our Infineon memory chip can be made to work in 4-byte mode using Zephyr 3.7.0 SPI NOR flash driver.  The memory just needs to be in 3-byte mode for the first time Zephyr driver queries it for SFDP parameters.  This is because Zephyr's driver at a low timing / byte framing level assumes the flash device is in 3-byte mode when the driver first read SFDP parameters.  After that, 4-byte mode works with all other driver API calls and flash device accesses.

It sounds however like this 4-byte mode may go against a JEDEC stanard.

A month ago I wrote a small patch to three SPI NOR driver files.  The patch provides for the external flash device to be put into 3-byte mode just before SDFP parameters reading.  The rest of the Zephyr driver remains unchanged, and continues to use 4-byte mode.

I am reading with interest the JEDEC document you suggest.  If 1S-1S-1S connection to a SPI NOR memory implies drivers should only use 3-byte mode, that would to me represent a major rewrite of Zephyr's SPI NOR flash driver.  That does not seem a likely solution, as Zephyr's in-tree drivers go through such a rigorous test and review process.

Could there be some project context I did not know to share in my request for help?  Is there a Kconfig symbol Zephyr needs to pick up to pass along to the driver that a flash device is wired 1S-1S-1S, and not wired in some higher throughput way?

maass-hamburg, 2025-01-12:
As stated in the document reading of the sfdp area is always with a 3 bytes field. Even if that chip is in 4-byte mode, reading of that should be still in with the 3-byte address. 

If you are having problems with that, consider trying the functionality that is introduced, if your chip supports it, here: https://github.com/zephyrproject-rtos/zephyr/pull/80372 . There it won't go into 4-byte mode and instead use different opcodes for 4-byte commands.

> Originally in the discussion area I asked whether the full testing applied to SPI NOR driver is available as open source to review. I am curious whether SPI NOR flash driver tests included running the driver on a system which warm reboots, but does not power cycle nor reset the flash device?

For testing: If there aren't any in this codebase, which I didn't found, there are none. You habe to test locally for yourself and write these tests too.