zephyrproject-rtos/zephyr #82860: modem_cellular: periodic_chat_script is only running once

BlimbBoat, 2024-12-11:
**Describe the bug**

```
MODEM_CHAT_SCRIPT_CMDS_DEFINE(quectel_bg95_periodic_chat_script_cmds,
			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGREG?", ok_match));
```
chat_script above only runs once and then completes.

```
[00:00:18.402,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_periodic_chat_script: step: 0
[00:00:18.402,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGREG?
[00:00:18.428,000] <dbg> modem_chat: modem_chat_log_received_command: +CGREG:  2 5 "FFFE" "F9715" 7
[00:00:18.428,000] <dbg> modem_cellular: modem_cellular_log_event: event registered
[00:00:18.428,000] <dbg> modem_cellular: modem_cellular_log_state_changed: switch from await registered to carrier on
[00:00:18.429,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:18.429,000] <dbg> modem_chat: modem_chat_script_stop: quectel_bg95_periodic_chat_script: complete
[00:00:18.429,000] <dbg> modem_cellular: modem_cellular_log_event: event script success
[00:00:18.456,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.456,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 29)
[00:00:18.456,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.456,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 12)
[00:00:18.485,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.485,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 13)
[00:00:18.510,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.510,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 12)
[00:00:18.541,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.541,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 12)
[00:00:18.541,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:18.541,000] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 9)
```

**Expected behavior**

It used to run fine, which resulted in the script being called periodically. I could not figure out what change impacted modem_cellular, as I am pretty sure I reverted most changes to test the behaviour again.

**Environment (please complete the following information):**

Running a custom board with an RC76xx and I am doing the following:

```
#if DT_HAS_COMPAT_STATUS_OKAY(quectel_bg95)
MODEM_CHAT_SCRIPT_CMDS_DEFINE(quectel_bg95_init_chat_script_cmds,
				  MODEM_CHAT_SCRIPT_CMD_RESP("ATE0", ok_match),
				  MODEM_CHAT_SCRIPT_CMD_RESP("AT+IFC=2,2", ok_match),
				  MODEM_CHAT_SCRIPT_CMD_RESP("AT+COPS=2", ok_match),
				  MODEM_CHAT_SCRIPT_CMD_RESP("AT+COPS=0", ok_match),
				  MODEM_CHAT_SCRIPT_CMD_RESP("AT+KSRAT=5", ok_match),
			      MODEM_CHAT_SCRIPT_CMD_RESP_NONE("AT+CMUX=0,0,5,127", 300));

MODEM_CHAT_SCRIPT_DEFINE(quectel_bg95_init_chat_script, quectel_bg95_init_chat_script_cmds,
			 abort_matches, modem_cellular_chat_callback_handler, 10);

MODEM_CHAT_SCRIPT_CMDS_DEFINE(quectel_bg95_dial_chat_script_cmds,		  
			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGDCONT=1,\"IP\","
							 "\""CONFIG_MODEM_CELLULAR_APN"\"",
							 ok_match),	 			   		 
			      MODEM_CHAT_SCRIPT_CMD_RESP_NONE("ATD*99***1#", 0));

MODEM_CHAT_SCRIPT_DEFINE(quectel_bg95_dial_chat_script, quectel_bg95_dial_chat_script_cmds,
			 dial_abort_matches, modem_cellular_chat_callback_handler, 10);

MODEM_CHAT_SCRIPT_CMDS_DEFINE(quectel_bg95_periodic_chat_script_cmds,
			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGREG?", ok_match));

MODEM_CHAT_SCRIPT_DEFINE(quectel_bg95_periodic_chat_script,
			 quectel_bg95_periodic_chat_script_cmds, abort_matches,
			 modem_cellular_chat_callback_handler, 4);
```
Any insight is greatly appreciated!

bjarki-andreasen, 2024-12-13:
I really have no idea, just looking at the source code, it should work just fine. only thing I can speculate is that the timeout is missed somehow? try increasing the size of `uint8_t event_buf[8];` to 32 or something, and keep an eye on `modem_cellular_delegate_event()` see if the timeout is actually added successfully

BlimbBoat, 2024-12-13:
Thanks for the comment! I will look into it next week.

I think I was also a bit unclear in my original post. I have only shown the changes I made in modem_cellular, but my actual code is most likely the reason the periodic_script is not running. I reverted my code to a previous commit and modem_cellular was working just fine. 

I will keep an eye on `modem_cellular_delegate_event()` while also investigating the changes in my code as well. Thanks!

BlimbBoat, 2024-12-13:
Managed to narrow it down a bit.

I have a ZBus 'Module' called modem and an l4_cb. At the end of it I have a zbus_chan_pub to publish the network_status. If I were to remove it the script will run periodically again.

bjarki-andreasen, 2024-12-13:
Sounds like it could be the system work queue being blocked, could you check which context calls the l4_cb?

bjarki-andreasen, 2024-12-13:
That's likely, no blocking work may be performed from there, `zbus_chan_pub` can be blocking if there are zbus callback listeners registered which do something blocking, try following the call to zbus_chan_pub() and see if anything blocking is called, like taking a sem or mutex

BlimbBoat, 2024-12-16:
The `zbus_chan_pub` indeed takes a sem with `chan_lock`. 

But if this is the problem I will move the function out of the cb. Thanks for your help!

closed 2024-12-16 by BlimbBoat