zephyrproject-rtos/zephyr #82874: WiFi doesn't seem to work on ESP32-C6

martinjaeger, 2024-12-11:
**Describe the bug**

The WiFi Support added in #81078 doesn't seem to work. Running `wifi scan` on the WiFi shell on a XIAO ESP32-C6 never outputs anything.

**To Reproduce**

Steps to reproduce the behavior:
1. `west build -b xiao_esp32c6 samples/net/wifi/shell -p`
2. `west flash`
3. `picocom -b 115200 /dev/ttyACM0 --lower-dtr`
4. Run `wifi scan`

Nothing happens.

**Expected behavior**

Output of the WiFi scan.

**Impact**

Showstopper for our project.

**Logs and console output**

```
...
I (154) boot: Jumping to the main image...
I (178) heap_runtime: ESP heap runtime init at 0x4083bcb0 size 258 kB.

[00:00:00.022,000] <err> entropy: Error enabling TRNG clock
*** Booting Zephyr OS build v4.0.0-1883-g6dafb5f4a9cb ***
uart:~$ 
uart:~$ wifi scan
Scan requested
uart:~$ 
```

**Environment (please complete the following information):**

 - OS: Manjaro Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 6dafb5f4a9cbd7043989f986d1f5963c264d413e

martinjaeger, 2024-12-16:
Thanks for the initial investigation. I tried again with `usb_serial` disabled and `wifi scan` does indeed work. However, if I run `wifi connect`, it doesn't get assigned an IP and `wifi status` always prints `SCANNING`. I used the same commands which are working fine on an ESP32C3 board.

raintamer, 2024-12-16:
Bump CONFIG_NET_MGMT_EVENT_QUEUE_SIZE from 16 to 24. The default 16 causes problems with events getting propagated to the proper place.

martinjaeger, 2024-12-17:
Ok, got it to work with disabled `usb_serial` on latest main (commit d27e4dc463890ea2adcfcd78212006b04edaa21a).

martinjaeger, 2024-12-17:
FWIW, it's not only the `usb_serial` causing trouble. WiFi also stops working if you enable `uart1` (which we need in our application for a Modbus interface). This is really weird...

PR #83213 (2024-12-23): drivers: interrupt_controller: esp32c6: reserve IRQ 1 for WiFi
IRQ 1 is reserved for WiFi in ESP-IDF, which is used as the basis for the Espressif HAL in Zephyr.

If IRQ 1 is used by Zephyr and too many other peripherals (e.g. multiple UARTs) are enabled, WiFi stops working.

Marking IRQ 1 as "not available" seems to fix the issue.

Fixes #82874