zephyrproject-rtos/zephyr #82889: Introduce option parameter for shields

soburi, 2024-12-12:
# Introduction

Zephyr supports shields for adding expansion boards, primarily modeled after Arduino shields. These assume a single shield connector per system. However, with the growing adoption of boards featuring Grove connectors, which may include multiple connectors, the current shield definition lacks the flexibility to support these configurations effectively.

This proposal extends the shield framework by allowing options to be passed at build time, enabling users to specify connection parameters such as the target port or bus.

# Problem description

The existing shield system is designed to handle single connectors and does not accommodate boards with multiple Grove connectors. Without this flexibility, users cannot easily specify different connection ports for shields, limiting the utility and adaptability of the current framework. This issue becomes more pronounced as boards with multiple Grove connectors become popular.

# Proposed change

Introduce a mechanism to define shields with configurable options, such as the connection port, by passing parameters at build time. This mechanism will generate a header file based on the user-provided options, which will be used during the devicetree preprocessing step to enable flexible overlay definitions.

# Proposed change (Detailed)

I am creating a implementation at https://github.com/zephyrproject-rtos/zephyr/pull/82825. (Work in progress)

#### Command-Line Parameterization:
Users will specify shield options at `west build` time using a format like `--shield awesome_grove_device@grove_i2c2`. This allows users to define the target connection (e.g., a specific I2C bus).

#### Header File Generation:
Based on the provided options, a header file will be generated automatically. For instance:

```c
#define SHIELD_AWESOME_GROVE_DEVICE_OPTION0 grove_i2c2
```

The shield name ensures uniqueness, enabling conflict-free option handling.

#### Integration with Devicetree:

The generated header file will be included during devicetree preprocessing. Overlays can reference the options using macros:

```c
#ifndef SHIELD_AWESOME_GROVE_DEVICE_OPTION0
#define SHIELD_AWESOME_GROVE_DEVICE_OPTION0 grove_i2c
#endif

&SHIELD_AWESOME_GROVE_DEVICE_OPTION0 {
   ...
};
```

This approach maintains backward compatibility while introducing build-time flexibility.
By properly designing the conditions, we can define an overlay that also fits into our current system.

# Dependencies

This change is independent of other features.

# Concerns and Unresolved Questions

Overlay Limitations: The same overlay cannot be reused for multiple device instances with different configurations. Addressing this limitation may require additional design changes.

# Alternatives

Some alternatives are considerable, but they are too complex.

soburi, 2024-12-12:
@fabiobaltieri @kartben 

I know you are an expert in this domain and would love to hear your thoughts if you have the time.

fabiobaltieri, 2024-12-12:
Are we?! Cool! :-) Adding few people I think may have opinions/ideas here

cc @tejlmand @gmarull @bjarki-andreasen @henrikbrixandersen @erwango @decsny 

Yeah I think it's a cool idea, some macro and devicetree shenanigans so I think it's a good fit for the project. :-)

This supports multiple instance of the same shield on different busses right? (sorry the cmake is a bit beyond me... weird, I thought I was an expert on it :-)).

Also, should this support changing the address as well? Some of those i2c boards have solder jumpers for address and you can daisy chain them so that may be a useful second option to have  multiple of those on the same bus. Maybe even give a more rigid structure to the option? like `--shield <shield_name>@<bus>:<address> or something like that?

Anyway, good stuff.

bjarki-andreasen, 2024-12-12:
When working with the nordic WiFi shields, there are two primary issues:
* Having more than connector of the same type, identified by this PR
* Having only a subset of a connector available (busses, pins), like a specific microbBUS connector instance missing SPI or reset PIN.

Having multiple connectors, the simplest way to support it is to index them, and use the `--shield foo@0, --shield foo@1` like @fabiobaltieri suggested, and then having the connectors named `arduino_i2c0, arduino_i2c1` etc.

The second issue however, if a pin is missing from the connector, but referenced in a shield overlay:
connector:
```
	arduino_header: connector {
		compatible = "arduino-header-r3";
		#gpio-cells = <2>;
		gpio-map-mask = <0xffffffff 0xffffffc0>;
		gpio-map-pass-thru = <0 0x3f>;
		gpio-map = <0 0 &gpio0 4 0>,	/* A0 */
			   <1 0 &gpio0 5 0>,	/* A1 */
			   <6 0 &gpio1 0 0>,	/* D0 */
			   <7 0 &gpio1 1 0>,	/* D1 */
			   <8 0 &gpio1 4 0>,	/* D2 */
			   <9 0 &gpio1 5 0>,	/* D3 */
	};
```
shield:
```
	foo: foo {
		compatible = "vnd,foo";
		status = "okay";

                /* D2 */
		status0-gpios = <&arduino_header 8 GPIO_ACTIVE_HIGH>;
        };
```
currently, this will fail since when looking up `arduino_header 8`, its not found. If we try to add a snippet/overlay to remove the reference from the shield:
```
&foo {
        /delete-node/ status0-gpios;
};
```
it still fails since the lookup is performed before a snippet or overlay is applied.

We essentially need something like:
shield:
```
	foo: foo {
		compatible = "vnd,foo";
		status = "okay";

#if CONFIG_ARDUINO_HEADER_8_EXISTS
                /* D2 */
		status0-gpios = <&arduino_header 8 GPIO_ACTIVE_HIGH>;
#endif
        };
```
to exclude it.

bjarki-andreasen, 2024-12-12:
Regarding implementation, I think this would be solved better with a yaml schema applied to the board somewhere, which would be used to generate code using python. something like:

schema:
```
board:
  connectors:
    - arduino
    - microbus
```
from which we would look up any nodelabel starting with `arduino` or `microbus`:
```
arduino0_header: connector {
        ...
};
arduino0_spi: &spi4 {
        ...
};
```
and generate the header from this:
```
#define SHIELD_ARDUINO0_EXISTS 1
#define SHIELD_ARDUINO0_HAS_SPI 1
#define SHIELD_ARDUINO0_HAS_HEADER 1
```
which we could easily make support existing boards by automatically appending 0 to the node name if there is only one of them, so `arduino_header` would match `SHIELD_ARDUINO0_EXISTS 1`.

This would lastly be able to fix the individual pin issue as well if we add pin awareness to the yaml schema:
```
board:
  connectors:
    - arduino:
      header:
        gpios: 16
```
which would allow the python script to check the `gpio-map` for existance of pins 0, 1, 2, ... and generate:
```
#define SHIELD_ARDUINO0_HAS_HEADER_GPIOS_0 1
#define SHIELD_ARDUINO0_HAS_HEADER_GPIOS_1 0
#define SHIELD_ARDUINO0_HAS_HEADER_GPIOS_2 1
```

This implementation is really flexible and should be easier readable than cmake (only a few wizards know that tung :D)

soburi, 2025-01-12:
@fabiobaltieri @bjarki-andreasen 

I'm sorry for not getting back to you sooner.

I updated the PR to reflect your suggestion.
I haven't implemented any tests yet, but it seems to work well as a proof of concept.

> This supports multiple instance of the same shield on different busses right? (sorry the cmake is a bit beyond me... weird, I thought I was an expert on it :-)).

After some consideration, we were able to achieve this.
However, I had to change the build mechanism, so I want you to review this.

It is described in https://builds.zephyrproject.io/zephyr/pr/82825/docs/hardware/porting/shields.html#derived-overlay

> Also, should this support changing the address as well? Some of those i2c boards have solder jumpers for address and you can daisy chain them so that may be a useful second option to have multiple of those on the same bus.

This was solved by having the data internally in a generic key=value format. 
Options can be arbitrarily defined using a schema, providing a comprehensive solution.

https://builds.zephyrproject.io/zephyr/pr/82825/docs/hardware/porting/shields.html#specifying-parameters

> Maybe even give a more rigid structure to the option? like `--shield <shield_name>@:

> Having multiple connectors, the simplest way to support it is to index them, and use the --shield foo@0, --shield foo@1 like @fabiobaltieri suggested, and then having the connectors named arduino_i2c0, arduino_i2c1 etc.

The "@" notation is also supported.
As part of general-purpose mechanism, it is treated as just a "number." The shield developer can decide how to use this number, but it will generally converge to the "connector number."

> This implementation is really flexible and should be easier readable than cmake (only a few wizards know that tung :D)

Pythonization has been addressed.👍

> Regarding implementation, I think this would be solved better with a yaml schema applied to the board somewhere, which would be used to generate code using python. something like:

The `shield.yml` has also been supported. However, as described later, some suggestions for @... seemed difficult to implement, so it may be a little different from what you expected.

https://builds.zephyrproject.io/zephyr/pr/82825/docs/hardware/porting/shields.html#shield-yaml-format

Adaptive configuration
--------------------------

Now that we can define flexible options,
it's possible to switch between the following cases with #ifdef on the command line.
I can say this is a small step forward.

```
	foo: foo {
		compatible = "vnd,foo";
		status = "okay";

#if CONFIG_ARDUINO_HEADER_8_EXISTS
                /* D2 */
		status0-gpios = <&arduino_header 8 GPIO_ACTIVE_HIGH>;
#endif
        };
```

But,

https://github.com/zephyrproject-rtos/zephyr/issues/82889#issuecomment-2538599154

> which would allow the python script to check the gpio-map for existance of pins 0, 1, 2, ... and generate:

I think the @bjarki-andreasen proposed is a bit challenging to implement now.

The processing of dts is roughly as follows.

1. Preprocessing
2. Creating EDT Creating EDT.pickle (gen_edt.py)
3. Generating devicetree-generated.h (gen_defines.py)
4. Normal build of C source

To analyze DTS, all preprocessor conditions must be resolved, and this approach reverses the order. (Bootstrap problem!)
In other words, macro generation and resolution must be completed at point 1, and with the current mechanism, I think it is impossible to handle cross-references between device trees.

Based on the @bjarki-andreasen proposed idea,
if we extend the board schema to include the shield definition,
it would be possible to automatically generate files like `arduino_r3_connector.dtsi` from this schema.
This means that duplicate definitions can be avoided.
With this method, you can get the configuration information without going through DTS, so all the processing can be done at step 1 mentioned above.
Alternatively, it would be good to consider a mechanism to ensure that the schema and DTS do not diverge, but I think this would be difficult because DTS is a flexible mechanism.

This proposal is also large enough, so it is good to discuss a separate issue.

decsny, 2025-01-13:
This proposal seems to have matured pretty well so far, should it be discussed in architecture review ? @henrikbrixandersen @carlescufi

henrikbrixandersen, 2025-01-14:
> This proposal seems to have matured pretty well so far, should it be discussed in architecture review ? @henrikbrixandersen @carlescufi

@soburi Will you be able to present this to the Arch WG meeting, perhaps next week?

soburi, 2025-01-20:
@henrikbrixandersen 

> [@soburi](https://github.com/soburi) Will you be able to present this to the Arch WG meeting, perhaps next week?

As I replied in Discode, It is okay.
(It's midnight in my region, and unless I oversleep,...)

If we're going to do it tomorrow, I'd appreciate it if you could also give me a quick comment on this.
https://github.com/zephyrproject-rtos/zephyr/pull/83668

nordicjm, 2025-01-21:
There was this one I started work on some time ago but gave up due to lack of time which has the simpler configuration of just having shield variants/qualifiers which work using the same syntax as board variants/qualifiers
https://github.com/zephyrproject-rtos/zephyr/compare/main...nordicjm:zephyr:shieldrevisiontest

```
npm1100_ek@slot1
npm1100_ek@slot2
mikroe_accel13_click/i2c
mikroe_accel13_click/i2c/1
mikroe_accel13_click/i2c/2
mikroe_accel13_click/spi
mikroe_accel13_click/spi/1
mikroe_accel13_click/spi/2
```

Though not advocating for that solution over the one presented here