zephyrproject-rtos/zephyr #82898: drivers/wifi/espat: Add support for UDP connections with changeable addresses and IP

hongquan-prog, 2024-12-12:
**Is your enhancement proposal related to a problem? Please describe.**
Currently espat cannot be used as an udp server, for example when using udp transfer in mcumgr, the client's ip and port are unknown and mcumgr does not work.

```c
static int _sock_connect(struct esp_data *dev, struct esp_socket *sock)
{
	/* Calculate the largest possible AT command length based on both TCP and UDP variants. */
	char connect_msg[MAX(sizeof("AT+CIPSTART=000,\"TCP\",\"\",65535,7200") +
				NET_IPV4_ADDR_LEN,
			     sizeof("AT+CIPSTART=000,\"UDP\",\"\",65535,65535,0,\"\"") +
				2 * NET_IPV4_ADDR_LEN)];
	char dst_addr_str[NET_IPV4_ADDR_LEN];
	char src_addr_str[NET_IPV4_ADDR_LEN];
	struct sockaddr src;
	struct sockaddr dst;
	int ret;

	if (!esp_flags_are_set(dev, EDF_STA_CONNECTED | EDF_AP_ENABLED)) {
		return -ENETUNREACH;
	}

	k_mutex_lock(&sock->lock, K_FOREVER);
	src = sock->src;
	dst = sock->dst;
	k_mutex_unlock(&sock->lock);

	if (dst.sa_family == AF_INET) {
		net_addr_ntop(dst.sa_family,
			      &net_sin(&dst)->sin_addr,
			      dst_addr_str, sizeof(dst_addr_str));
	} else {
		strcpy(dst_addr_str, "0.0.0.0");
	}

	if (esp_socket_ip_proto(sock) == IPPROTO_TCP) {
		snprintk(connect_msg, sizeof(connect_msg),
			 "AT+CIPSTART=%d,\"TCP\",\"%s\",%d,7200",
			 sock->link_id, dst_addr_str,
			 ntohs(net_sin(&dst)->sin_port));
	} else {
		if (src.sa_family == AF_INET && net_sin(&src)->sin_port != 0) {
			net_addr_ntop(src.sa_family,
				      &net_sin(&src)->sin_addr,
				      src_addr_str, sizeof(src_addr_str));
			snprintk(connect_msg, sizeof(connect_msg),
				 "AT+CIPSTART=%d,\"UDP\",\"%s\",%d,%d,0,\"%s\"",
				 sock->link_id, dst_addr_str,
				 ntohs(net_sin(&dst)->sin_port), ntohs(net_sin(&src)->sin_port),
				 src_addr_str);
		} else {
			snprintk(connect_msg, sizeof(connect_msg),
				 "AT+CIPSTART=%d,\"UDP\",\"%s\",%d",
				 sock->link_id, dst_addr_str,
				 ntohs(net_sin(&dst)->sin_port));
		}
	}
}
```
When using mcumgr, the parameters passed by the application are combined into the following instructions，the module will return an error indicating that the parameter is invalid.
```
AT+CIPSTART=0,"0.0.0.0",0,1337,0,"0.0.0.0"
```
```
AT+CIPSTART=<link ID>,<"type">,<"remote host">,<remote port>[,<local port>,<mode>,<"local IP">][,<timeout>]
```
The mode 0 means that the client's IP and port are unchangeable.

**Describe the solution you'd like**
When the remote IP and port passed by the application are both 0, connect using mode 2
```c
static int _sock_connect(struct esp_data *dev, struct esp_socket *sock)
{
	int mode;

	if (esp_socket_ip_proto(sock) == IPPROTO_TCP) {
		snprintk(connect_msg, sizeof(connect_msg),
			 "AT+CIPSTART=%d,\"TCP\",\"%s\",%d,7200",
			 sock->link_id, dst_addr_str,
			 ntohs(net_sin(&dst)->sin_port));
	} else {
		if (src.sa_family == AF_INET && net_sin(&src)->sin_port != 0) {
			net_addr_ntop(src.sa_family,
				      &net_sin(&src)->sin_addr,
				      src_addr_str, sizeof(src_addr_str));
			
			/* <mode>: In the UDP Wi-Fi passthrough
			 *
			 * 0: After UDP data is received, the parameters <"remote host">
			 * and <remote port> will stay unchanged (default).
			 * 1: Only the first time that UDP data is received from an IP 
			 * address and port that are different from the initially set value
			 * of parameters <remote host> and <remote port>, will they be changed
			 * to the IP address and port of the device that sends the data.
			 * 2: Each time UDP data is received, the <"remote host"> and 
			 * <remote port> will be changed to the IP address and port of the
			 * device that sends the data.
			 */
			if ((net_sin(&dst)->sin_addr.s_addr == 0) && (ntohs(net_sin(&dst)->sin_port) == 0)) {
				/*  */
				mode = 2;
				/* Port 0 is reserved */
				net_sin(&dst)->sin_port = net_sin(&src)->sin_port;
			} else {
				mode = 0;
			}
			
			snprintk(connect_msg, sizeof(connect_msg),
				 "AT+CIPSTART=%d,\"UDP\",\"%s\",%d,%d,%d,\"%s\"",
				 sock->link_id, dst_addr_str,
				 ntohs(net_sin(&dst)->sin_port), ntohs(net_sin(&src)->sin_port),
				 mode, src_addr_str);
		} else {
			snprintk(connect_msg, sizeof(connect_msg),
				 "AT+CIPSTART=%d,\"UDP\",\"%s\",%d",
				 sock->link_id, dst_addr_str,
				 ntohs(net_sin(&dst)->sin_port));
		}
	}
}
```

jukkar, 2024-12-12:
There is no maintainer for `esp_at` driver atm but assigning to @mniestroj for time being, please remove if I set it incorrectly.

hongquan-prog, 2024-12-13:
> There is no maintainer for `esp_at` driver atm but assigning to @mniestroj for time being, please remove if I set it incorrectly.

I've been using this driver a lot and have found some problems, I can submit a corresponding PR, but before that I'd like to know your opinions.

mniestroj, 2024-12-15:
Thanks for reporting this issue. Connecting using other mode seems reasonable. Please continue with creating PR, so we can discuss further code changes that are required. That will be easier than just discussing them here.

hongquan-prog, 2024-12-16:
> discussing

ok

PR #83023 (2024-12-18): drivers/wifi/esp_at: Use mode 2 to connect when using as a UDP server
Mode 0 cannot establish a connection when used as a UDP server,
replace mode 0 with mode 2 to save the client's IP and port as the
communication address when a message is received.

Fixes #82898
commit 13ea5877 (2024-12-18): drivers/wifi/esp_at: Use mode 2 to connect when using as a UDP server

Mode 0 cannot establish a connection when used as a UDP server,
replace mode 0 with mode 2 to save the client's IP and port as
the communication address when a message is received.

Fixes #82898
closed 2024-12-18 by kartben