zephyrproject-rtos/zephyr #82904: lib: p4workq: libraries.p4wq fails

katgiadla, 2024-12-12:
**Describe the bug**
Test `tests/lib/p4workq/libraries.p4wq` fails.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s libraries.p4wq -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error (using e.g. minicom)

**Expected behavior**
Console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-1852-g1baf93b1d169 ***
Running TESTSUITE lib_p4wq
===================================================================
START - test_fill_queue
 PASS - test_fill_queue in 0.003 seconds
===================================================================
START - test_resubmit
 PASS - test_resubmit in 0.101 seconds
===================================================================
START - test_stress
ASSERTION FAIL [z_spin_lock_valid(l)] @ zephyr/include/zephyr/spinlock.h:136
Invalid spinlock 0x20000d54
E: r0/a1:  0x00000004  r1/a2:  0x00000088  r2/a3:  0x00000001
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x0000063f
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x00007698
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000570 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: ` v4.0.0-1852-g1baf93`

**Additional information:**
Introduced probably by 2d959d96fbb73a630ef77579e87ab25c394a11d5

valeriosetti, 2024-12-12:
Just tested on a `nrf52840dk/nrf52840` and the test is fine. Please note that #82807 (reverting changes introduced in #82497) was recently merged, so you might want to test it again on current `main` branch

katgiadla, 2024-12-12:
> Just tested on a `nrf52840dk/nrf52840` and the test is fine. Please note that #82807 (reverting changes introduced in #82497) was recently merged, so you might want to test it again on current `main` branch

It has been re-tested on main branch and it is passed. Thanks for information

closed 2024-12-12 by katgiadla