zephyrproject-rtos/zephyr #82916: kernel: sched: deadline: kernel.scheduler.deadline.* fail

katgiadla, 2024-12-12:
**Describe the bug**
Tests 
- `tests/kernel/sched/deadline/kernel.scheduler.deadline` 
- `tests/kernel/sched/deadline/kernel.scheduler.deadline.scalable`
fail.

Observed for `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s kernel.scheduler.deadline -s kernel.scheduler.deadline.scalable -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error (using e.g. minicom)

**Expected behavior**
Console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-1852-g1baf93b1d169 ***
Running TESTSUITE suite_deadline
===================================================================
START - test_deadline

    Assertion failed at zephyr/tests/kernel/sched/deadline/src/main.c:86: suite_deadline_test_deadline: (n_exec == 0 is false)
threads ran too soon
 FAIL - test_deadline in 0.016 seconds
===================================================================
START - test_unqueued
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x0
E: r0/a1:  0x20000570  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000067f5
E:  xpsr:  0x81000221
E: Faulting instruction address (r15/pc): 0x000064e2
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000a90 (test_unqueued)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: ` v4.0.0-1852-g1baf93`

**Additional information:**
Introduced probably by 2d959d96fbb73a630ef77579e87ab25c394a11d5

valeriosetti, 2024-12-12:
Just tested on a `nrf52840dk/nrf52840` and both tests are fine.
Please note that https://github.com/zephyrproject-rtos/zephyr/pull/82807 (reverting changes introduced in https://github.com/zephyrproject-rtos/zephyr/pull/82497) was recently merged, so you might want to test it again on current main branch

katgiadla, 2024-12-12:
> Just tested on a `nrf52840dk/nrf52840` and both tests are fine. Please note that #82807 (reverting changes introduced in #82497) was recently merged, so you might want to test it again on current main branch

It has been re-tested on main branch and it is passed. Thanks for information

closed 2024-12-12 by katgiadla