zephyrproject-rtos/zephyr #82934: esp32s3: large buffer allocation in External RAM

jeronimoagullo, 2024-12-12:
I am working on a ESP32 display driver for VIEWE display boards based on **ESP32S3 N16R8** (https://viewedisplay.com/iot_aiot-smart-display/)

My approach is to allocate a frame buffer of 768.000 bytes, that is 480x800x2 (RGB565 format) in the external RAM. The esp32s3 used board has 8Mb external RAM so the memory should not be a limitation.

Thus, I checked out the `samples/boards/espressif/spiram_test` to allocate these bytes into external memory. It works fine for small buffers (even the memory address of allocated buffer is correct), but fails to allocate the 768.00 bytes.

**Tested approaches**
1. Firstly, I tried to increase the SPIRAM heap with `CONFIG_ESP_SPIRAM_HEAP_SIZE`. The `ext_ram_seg` increases correctly in the memory resume when building, but the sample fails too.
2. I modify the sample code to checkout what is the largest buffer that I can allocate:
```c
#include <zephyr/kernel.h>
#include <zephyr/sys/printk.h>
#include <soc/soc_memory_layout.h>
#include <zephyr/multi_heap/shared_multi_heap.h>

#define MAX_BYTES 800000

int main(void)
{
	int *m_ext, *m_int;

	for (uint32_t bytes = 0; bytes < MAX_BYTES; bytes+=1024)
	{
		printk("\n>>Bytes: %d\n", bytes);
		
		m_ext = shared_multi_heap_aligned_alloc(SMH_REG_ATTR_EXTERNAL, 32, bytes);

		if (!esp_ptr_external_ram(m_ext)) {
			printk("SPIRAM mem test failed %p\n", (void*)m_ext);
		} else {
			printk("SPIRAM mem test pass %p\n", (void*)m_ext);
		}

		shared_multi_heap_free(m_ext);

		m_int = k_malloc(bytes);
		if (!esp_ptr_internal(m_int)) {
			printk("Internal mem test failed %p\n", (void*)m_int);
		} else {
			printk("Internal mem test pass %p\n", (void*)m_int);
		}

		k_free(m_int);
	}
	
	return 0;
}
```

The result is the following:
```
>>Bytes: 129024
SPIRAM mem test pass 0x3c020080
Internal mem test pass 0x3fc91ea0

>>Bytes: 130048
SPIRAM mem test pass 0x3c020080
Internal mem test pass 0x3fc91ea0

>>Bytes: 131072
SPIRAM mem test failed 0
Internal mem test pass 0x3fc91ea0

>>Bytes: 132096
SPIRAM mem test failed 0
Internal mem test pass 0x3fc91ea0

>>Bytes: 133120
SPIRAM mem test failed 0
Internal mem test pass 0x3fc91ea0
```

The larger buffer that can be allocated is small than 131072 bytes, which is almost the default size of `ext_ram_seg`. **It seems that the CONFIG_ESP_SPIRAM_HEAP_SIZE kconfig doesn't work**.

**Question**
I would like to know if I am missing something or if it is a bug? Thanks in advance!

sylvioalves, 2024-12-16:
@jeronimoagullo Hi, I run the same code in an ESP32-S3 N32R8 and got the expected results. Will try to find a N16R8. By the way, can you provide the full initialization log output of your board? Thanks!

```c
(...)
>>Bytes: 795648
SPIRAM mem test pass 0x3c020080
Internal mem test failed 0
>>Bytes: 796672
SPIRAM mem test pass 0x3c020080
Internal mem test failed 0
>>Bytes: 797696
SPIRAM mem test pass 0x3c020080
Internal mem test failed 0
>>Bytes: 798720
SPIRAM mem test pass 0x3c020080
Internal mem test failed 0

>>Bytes: 799744
SPIRAM mem test pass 0x3c020080
Internal mem test failed 0
```

jeronimoagullo, 2024-12-16:
Hi @sylvioalves, I managed to get it working adding the Kconfig macro `CONFIG_SYS_HEAP_BIG_ONLY=y`. However, I don't understand how it changes the behaviour and what implies exactly. With this macro the test passed for any size and allocate always the buffer correctly at an external RAM address.

In any case, I send the full initialization log without the above Kconfig macro as requested:

```
---- Opened the serial port COM14 ----
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc8e158,len:0x1a74
load:0x40374000,len:0xa138
SHA-256 comparison failed:
Calculated: b03b7d710d62aadd3e2357da580125c86cbc83e748c06b626ecf0152fc502817
Expected: 0000000010440000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40377d48
I (70) soc_init: ESP Simple boot
I (70) soc_init: compile time Dec 16 2024 14:37:28
W (70) soc_init: Unicore bootloader
I (70) spi_flash: detected chip: gd
I (73) spi_flash: flash io: dio
W (75) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (88) soc_init: chip revision: v0.2
I (91) flash_init: Boot SPI Speed : 40MHz
I (95) flash_init: SPI Mode       : DIO
I (98) flash_init: SPI Flash Size : 16MB
I (102) soc_random: Enabling RNG early entropy source
I (107) boot: DRAM: lma 0x00000020 vma 0x3fc8e158 len 0x1a74   (6772)
I (113) boot: IRAM: lma 0x00001a9c vma 0x40374000 len 0xa138   (41272)
I (119) boot: padd: lma 0x0000bbe8 vma 0x00000000 len 0x4410   (17424)
I (125) boot: IMAP: lma 0x00010000 vma 0x42000000 len 0x3d14   (15636)
I (132) boot: padd: lma 0x00013d1c vma 0x00000000 len 0xc2dc   (49884)
I (138) boot: DMAP: lma 0x00020000 vma 0x3c010000 len 0x12c8   (4808)
I (144) boot: Image with 6 segments
I (147) boot: DROM segment: paddr=00020000h, vaddr=3c010000h, size=012D0h (  4816) map
I (155) boot: IROM segment: paddr=00010000h, vaddr=42000000h, size=03D12h ( 15634) map
I (174) soc_random: Disabling RNG early entropy source
I (174) boot: Disabling glitch detection
I (174) boot: Jumping to the main image...
I (175) octal_psram: vendor id    : 0x0d (AP)
I (179) octal_psram: dev id       : 0x02 (generation 3)
I (184) octal_psram: density      : 0x03 (64 Mbit)
I (188) octal_psram: good-die     : 0x01 (Pass)
I (193) octal_psram: Latency      : 0x01 (Fixed)
I (197) octal_psram: VCC          : 0x01 (3V)
I (201) octal_psram: SRF          : 0x01 (Fast Refresh)
I (206) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (211) octal_psram: BurstLen     : 0x01 (32 Byte)
I (215) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (221) octal_psram: DriveStrength: 0x00 (1/1)
I (225) esp_psram: Found 8MB PSRAM device
I (228) esp_psram: Speed: 40MHz
I (962) esp_psram: SPI SRAM memory test OK
I (1000) heap_runtime: ESP heap runtime init at 0x3fc92e20 size 346 kB.

*** Booting Zephyr OS build 4488ed1a20a9 ***

>>Bytes: 0
SPIRAM mem test failed 0
Internal mem test pass 0x3fc91e60

(...)
```

sylvioalves, 2024-12-18:
HI, what is the `CONFIG_ESP_SPIRAM_HEAP_SIZE` value you used?
See, `CONFIG_SYS_HEAP_XXXXXX_ONLY` depends on the `CONFIG_ESP_SPIRAM_HEAP_SIZE`. We can eventually update that to AUTO mode. Will do that.

jeronimoagullo, 2024-12-18:
Hi @sylvioalves you are right, by default the value of `CONFIG_ESP_SPIRAM_HEAP_SIZE` is  **262134**, according to the default value when `CONFIG_SYS_HEAP_SMALL_ONLY` is not set.

If I set `CONFIG_SYS_HEAP_AUTO` instead of `CONFIG_SYS_HEAP_BIG_ONLY` it works too since in both cases the `CONFIG_SYS_HEAP_SMALL_ONLY` is not set.

sylvioalves, 2025-02-12:
Closing as solved.

closed 2025-02-12 by sylvioalves