zephyrproject-rtos/zephyr #82938: wifi: hostap:  Null pointer crash in Zephyr driver init

ThreeEights, 2024-12-12:
**Describe the bug**

If CONFIG_WIFI_NM_WPA_SUPPLICANT=y is selected but the WiFi device driver does not provide the supplicant API, wpa_drv_zep_init(), the API pointer dev_ops is null but not checked before calling dev_ops->init(), resulting in a hard fault;  the CPU halts before printing a crash report on the console.

Discovered using samples/net/sockets/echo_async with the wifi-ipv4 snippet, building for the Raspberry Pi Pico W.

**Note:**  While this may not be a fully supported configuration, I personally don't think it ought to crash.

**To Reproduce**

1. `west build -b rpi_pico/rp2040/w -S wifi-ipv4 samples/net/sockets/echo_async -- -DCMAKE_BUILD_TYPE=Debug`
2. Attach a Raspberry Pi Pico W with debug probe.
3. Launch openocd in a separate window.
4. `west debug`
5. On the debug console:  `c`

**Expected behavior**

Report the "No op registered for init" error and exit the function.

**Impact**

Annoyance, though it took far to long to track down which layer of indirection was causing the problem.

**Logs and console output**

Console output:
```
WLAN MAC Address : 28:CD:C1:0A:0D:F9
WLAN Firmware    : wl0: Jul 27 2022 21:10:05 version 7.95.55 (68e5ffa CY) FWID 01-9886495e
WLAN CLM         : API: 12.2 Data: 9.10.39 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2022-07-27 21:02:53 
WHD VERSION      : 2.6.0.18446 : v2.6.0 : GCC 12.2 : 2023-03-17 21:20:01 +0800
*** Booting Zephyr OS build 2ed8790e8b4c ***
```

I'd be happy to include the backtrace upon request.

**Environment (please complete the following information):**

 - OS:  Ubuntu Linux
 - Toolchain:  Zephyr SDK
 - Commit SHA or Version used:  Zephyr commit 2ed8790e8b4c5cbf0ef52124bc94b41da9f78fe3

**Additional context**

jukkar, 2024-12-13:
The board uses Infineon wifi driver which sets `CONFIG_WIFI_USE_NATIVE_NETWORKING` which implies that the driver needs hostap interfaces. But the driver is offloading driver and it should not set this value. It is a bit pointless to pull in hostap support if it is not used, so to me this looks like a driver configuration issue which should be fixed there.

ThreeEights, 2024-12-14:
@jukkar - I would humbly suggest that it still makes sense to fix the bug, if for no other reason than that the next networking neophyte will get an error instead of a crash deep inside the networking stack.

jukkar, 2024-12-16:
Sure we can add the check at runtime, it is just that it would be also good if we could catch this incompatibility (offload vs native) at built time.

ThreeEights, 2024-12-30:
@jukkar @ifyall - This issue was fixed by https://github.com/zephyrproject-rtos/hostap/pull/69, already merged.

closed 2025-03-16 by github-actions[bot]