zephyrproject-rtos/zephyr #82954: RFC: Deprecate BT_GAP_AUTO_UPDATE_CONN_PARAMS

Thalley, 2024-12-13:
## Introduction
The BT_GAP_AUTO_UPDATE_CONN_PARAMS option defaults to `y` and will automatically send a connection update shortly after connection establishment. Most applications do not set these values (or maybe aren't even aware of them), and there's not a single sample that sets the `CONFIG_BT_PERIPHERAL_PREF_*` values. 

### Problem description
Since this is enabled by default, whenever we are a peripheral, we automatically send a connection update with the default `CONFIG_BT_PERIPHERAL_PREF_` values, without the application/users knowledge, and often the central devices will accept this request. 

This can lead to instabilities (e.g. if ISO is used), slower discoveries (if the default values are higher than 7.5ms typically used during discovery), or higher power usage (if the default values are lower than what the central originally wanted to use). It has also caused confusion from users not be aware of this feature and asking why their connection parameters are changing. 

### Proposed change
Propose to deprecate the option. 
Propose to default it to `n` until it can be removed. 

## Detailed RFC
Removing the feature it pretty straight forward, once it has passed through its deprecation process. 
Removing this feature will also free up some resources in the host, and make the code overall simpler. 
Any peripheral application that actively want to update the connection parameters can do so in their own application easily, so we are not removing any functionality. 

### Proposed change (Detailed)
N/A

### Dependencies
Some applications might see a slight difference in their connection parameters, but should not be anything major. 

### Concerns and Unresolved Questions
N/A

## Alternatives
N/A

Thalley, 2024-12-13:
> I definitely use this in out of tree applications to specify values for battery operated devices and think removing this is a step in the wrong direction

It is easy for you application itself to request the update. 

Another downside of this option is that the parameters are only applied once. So if you start with initial INIT_CONN_PARAMS and then apply PREF_PARAMS after the 5 seconds, and the central requests INIT_CONN_PARAMS later, then you won't be using your PREF_PARAMS anyways. 

If the application controls the parameters, rather than the host applying them once, then you can better ensure that you are actually using the parameters you prefer. 

You can of course reject the parameter updates from the central.

cvinayak, 2024-12-16:
> Another downside of this option is that the parameters are only applied once. So if you start with initial INIT_CONN_PARAMS and then apply PREF_PARAMS after the 5 seconds, and the central requests INIT_CONN_PARAMS later, then you won't be using your PREF_PARAMS anyways.

Refer to these associated Kconfig options:
https://github.com/zephyrproject-rtos/zephyr/blob/c616a1719f2a2accf427766f596cb12e7515b371/subsys/bluetooth/host/Kconfig#L772-L804

These have been added as a spec recommendation to perform the preferred connection param update  after a 5 sec delay within this time a central could use a fast connection interval for GATT discovery. The retry was added for the Controllers (like Zephyr Controller) that support peripheral initiated LL Connection Parameter Request.

> the central requests INIT_CONN_PARAMS later, then you won't be using your PREF_PARAMS anyways.

This is a small enhancement that can be added to re-initiate the auto connection update when there is a connection update mismatching the preferred connection parameters.

I can see that an enhancement being added here:
https://github.com/zephyrproject-rtos/zephyr/blob/c616a1719f2a2accf427766f596cb12e7515b371/subsys/bluetooth/host/hci_core.c#L1934-L1971

nordicjm, 2024-12-16:
> I would argue that it is easy for you application itself to request the update.

Sure but it's a useful feature which applications can enable or disable depending upon if they need it, I personally don't see the problem with having it there. It can be disabled by default if people get confused but it then people can enable it but it is something common to a lot of applications (arguably any application wanting to lower latency, or increase latency to save power) so if it's removed it's just going to result in people having to copy and paste the code that's been removed to their applications which seems like a weird thing to do for useful functionality

Thalley, 2024-12-16:
> It can be disabled by default if people get confused by it then people can enable it but it is something common to a lot of applications

Disabling it by default would be a good start

> you're at the mercy of whatever connection parameters the central has connected with which is going to be vastly different depending on if it's an android phone, apple phone, linux device, windows pc, zephyr device, some other device, etc.)

That is the case regardless. In my experience with Android (mostly), it is quite persistent in the connection updates, and will continue to spam connection update requests until it gets what it wants. 

The central device is more often than not the device that is aware of the current usecase, and thus is often better suited to determine the most appropriate connection parameters anyway. There are definitely cases with sensors etc. where the peripheral may want to control this of course. 

> These have been added as a spec recommendation to perform the preferred connection param update after a 5 sec delay within this time a central could use a fast connection interval for GATT discovery.

My opinion abut the Peripheral Preferred Connection Parameters (and the characteristic), is that it is from a simpler time. For use cases like LE Audio where a central device may be connected to multiple peripherals, and where the ISO interval also comes into play, expecting the central to use, or even care about, the peripheral preferred connection interval is not something I see happening. Furthermore, higher layer profiles like BAP set multiple recommended connection parameters depending on the situation, so sticking with a single set of parameters (like Peripheral Preferred Connection Parameters is), is more harmful than useful in such situations IMO. 

Peripheral Preferred Connection Parameters works fine for simple 1:1 connections where the data sent is fairly consistent, but for LE Audio use cases where you have multiple devices and where you want to adjust the intervals based on a state machine, it just becomes fairly useless.

nordicjm, 2024-12-16:
> Peripheral Preferred Connection Parameters works fine for simple 1:1 connections where the data sent is fairly consistent, but for LE Audio use cases where you have multiple devices and where you want to adjust the intervals based on a state machine, it just becomes fairly useless.

Understandable but lots of applications only have/need a 1:1 connection e.g. sensors, firmware update, diagnostic information, etc.

Thalley, 2024-12-16:
> Understandable but lots of applications only have/need a 1:1 connection e.g. sensors, firmware update, diagnostic information, etc.

That is true. I am definitely also biased from LE Audio where this Kconfig has been an actual issue :) 

Alternatives to deprecation could then be
1) Disable by default (make it an active choice)
2) Disable by default if LE Audio is enable

We may also want to disable the characteristic by default as well

@nordicjm since this is somewhat similar to some other automatic features we recently removed from the host (e.g. automatic restart of adv), were you also opposed to removing those? If so, then there seem to be a disagreement in the stack direction where some people prefer keeping the stack simpler, and some prefer having the stack being more complex by providing QoL things like automatic tasks like these.

nordicjm, 2024-12-16:
Making it disabled by default is fine.

> @nordicjm since this is somewhat similar to some other automatic features we recently removed from the host (e.g. automatic restart of adv), were you also opposed to removing those? If so, then [...]

I was not aware that had been removed, just had a look and indeed seems `BT_LE_ADV_OPT_ONE_TIME` has been deprecated in 4.0 (I'm still using it in code for 3.7). I've always done one time advertising and restarted on disconnect as that allows for cleaning things up or changing LED state so removing automatic re-advertising isn't something that impacts me

Thalley, 2024-12-16:
> I was not aware that had been removed, just had a look and indeed seems `BT_LE_ADV_OPT_ONE_TIME` has been deprecated in 4.0 (I'm still using it in code for 3.7). I've always done one time [...]

We are also deprecating `bt_le_set_auto_conn`, so the current course of the host is to simplify it, which is also my initial suggestion here was to deprecate yet another automatic feature from the host, as it is subpar since it only applies the values once (which doesn't really make any difference when connected any of the major OSes that change the connection parameters fairly often), and that it complicates the host without providing any real value.

cvinayak, 2024-12-17:
The use of Peripheral Preferred Connection Parameter is a "should"  in the Bluetooth v6.0 Vol 3, Part C, Section 9.3.12 Connection interval timing parameters

```
The connection interval timing parameters are used within a connection. Initial
connection interval is used to ensure procedures such as bonding, encryption setup and
service discovery are completed quickly. The connection interval should be changed
to the value in the Peripheral Preferred Connection Parameters characteristic after
initiating procedures are complete.
```

I agree with @nordicjm here https://github.com/zephyrproject-rtos/zephyr/issues/82954#issuecomment-2544848386
and in sync with Bluetooth v6.0 Vol 3, Part C, Section 9.3.9 Connection Parameter Update procedure

```
If the requested or updated connection parameters are unacceptable to the Central
or Peripheral then it may disconnect the connection with the error code 0x3B
(Unacceptable Connection Parameters). Devices should be tolerant of connection
parameters given to them by the remote device.
```
Auto-update Connection Parameters in Zephyr Host should not be deprecated. The feature should be default `y` for peripheral role support in an application, and be a conscious decision by the application to remove the PPCP characteristics in GAP service.