zephyrproject-rtos/zephyr #82992: tests: drivers: build_all: sensor: config_i2c_rtio

FlorianWeber1018, 2024-12-14:
**Describe the bug**
sensors.generic_test is failing because of a possible bug in "st lsm6dsv16x" which does not build if CONFIG_I2C_RTIO is added to the testcase.

**To Reproduce**
add       "- CONFIG_I2C_RTIO" to  sensors.generic_test and run 
"west twister -p native_sim -s tests/drivers/build_all/sensor/sensors.generic_test"

**Environment (please complete the following information):**

 - OS: Linux in docker contain on linux host
 - Toolchain Zephyr SDK
 - Commit SHA: fcbf70f44d3c6d072288f8505bb1335faabdfb94

**Additional context**
CI on Draft PR is failing because of this bug:
https://github.com/zephyrproject-rtos/zephyr/pull/82665

teburd, 2024-12-20:
@avisconti would you mind taking a look?

FlorianWeber1018, 2025-01-08:
in more detailed the spi version of "st lsm6dsv16x" in sensors.generic_test does not build - CONFIG_I2C_RTIO was added to the testcase. The i2c version does not cause the build error.

ubieda, 2025-02-19:
May be fixed by: #85972

FlorianWeber1018, 2025-02-20:
It solves the issue. Thanks for you great work.

PR #85972 (2025-02-21): sensor: lsm6ds: fix: Do not enable Streaming for buses with no RTIO support
### Description
Ensure the corresponding bus enables RTIO (I2C, I3C or SPI) before instantiating rtio-dependent elements. Otherwise, enabling it for another bus (e.g: SPI_RTIO when the LSM6DS is also on I3C with no I3C_RTIO) will cause a build-time failure.

Fixes #82992 

CI failure seen on: https://github.com/zephyrproject-rtos/zephyr/actions/runs/13402926473?pr=85963
closed 2025-02-21 by fabiobaltieri