zephyrproject-rtos/zephyr #83010: C++ global constructors won't work on ek_ra8m1 dev board. Failure to flash target device with application using them.

piotrrak, 2024-12-16:
**Describe the bug**

C++ global constructors won't work for Renesas ek_ra8m1 traget.

**To Reproduce**
1. `west build -b ek_ra8m1 samples/cpp/hello_world`
2. `west flash`
3. Observe hang or error while flashing.

As an alternative:

Create any c++ app with global constructors
For example to break working `samples/cpp/cpp_synchronization` it is sufficient to add:

`#include <vector>`

and global

`std::vector<char> _g;`

While ensuring `_g` won't be compiled/optimized out. 

**Expected behavior**

C++ global constructors like (`std::cout` object) do work, one is able to flash application using this language feature.

**Impact**
Showstopper for using C++ on this target.

**Logs and console output**

```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: reset after flashing requested
-- runners.jlink: JLink version: 8.10l
-- runners.jlink: Flashing file: /home/prak/DEV/ra8-head/zephyr/build/zephyr/zephyr.hex
FATAL ERROR: command exited with status 1: /home/prak/.local/JLink_Linux_V810l_x86_64/JLinkExe -nogui 1 -if swd -speed auto -device R7FA8M1AH -CommanderScript /tmp/tmpo0w3cezdjlink/runner.jlink -nogui 1
```
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK, llvm
 - SHA 85e83038613a2b292c3022b922425f3d54be4d56 9cebb3ff63393662394a3132c12331d2de249f1d

**Additional context**

Could not reproduced this case for:
`ek_ra6m5`, `ek_ra4m3`
Works just fine on those similar targets.
I may also try `ra8t1` sometime next weekend if that helps.

Issue was reproduced on two separate boards with same effect.
Device is not debugable while flashing over SWD using ozone.

RobinKastberg, 2024-12-19:
what is the size of the hex/bin files? When having sections with pbits accidentally put in RAM, objcopy will do some horrible things (We should detect this nicer btw)

piotrrak, 2024-12-20:
Didn't store logs but reproduced it just now on f786ecc075132297fc409bb1d89e3300de1a95db

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      126128 B      2016 KB      6.11%
             RAM:        8648 B       896 KB      0.94%
OPTION_SETTING_OFS:          20 B         24 B     83.33%
OPTION_SETTING_SAS:           4 B        204 B      1.96%
OPTION_SETTING_S:         208 B        256 B     81.25%
        IDT_LIST:          0 GB        32 KB      0.00%
```
Still doesn't work for me.

This issue really puzzles me since I could reproduce it also with clang/libc++.
No big differences from RA8T1 in linker script, which I had working.
Result ELF was mostly same - only minor difference in size of segments, alignments were same - result addresses were overaligned.

Slightly OT: 
I won't be able to test with RA8T1 anymore. I somehow broke it/lock myself out from flashing, while trying to  figure out what are differences between 2 of those zephyr builds that may cause it.
On RA8T1 all I get now is `Downloading file Debug/P.elf...Verification failed @ address 0x0300A200` with any binary, eg. minimal `e2_studio` with only `renesas/fsp`. 
Issue is unrelated to zephyr, but unfortunate since I lost last board with cm85 where I could compare working vs not working builds.
Just mentioning it since, I can't proceed myself using this method anymore (only for now I hope).

piotrrak, 2024-12-21:
@RobinKastberg Thank you, your question hex size kind of pointed me in right direction. I do know what `TZ` is and `tee`, but I am new to this on MCUs. 

I don't know if it should be considered user error or not, but issue seemed to be incorrect CODE boundary for TZ.
Both of my ra8m1 boards came with 32KB boundary and that is insufficient space for zephyr with `CONFIG_REQUIRES_FULL_LIBCPP`.

Using renesas flasher one is able to correctly flash both boards.

`rfp-cli -d RA -tool jlink -if swd -file ./build/zephyr/zephyr.hex -fo boundary 128,0 -p`

That will work, since this binary will fit required space.
It seems that mck_ra8t1 had bigger boundary set for code by default, which made it magically work.

Should `rfp-cli` be the flasher for renesas RA boards then?
Or maybe it is possible to do by jlink alone, after all rfp uses jlink interface to do that?

My mck_ra8t1 has broken now flash @0x0300_A270:0x0300_A273  - the reserved area which should be always 0xff byte gapfill according to spec, however my board only reads `0's` there. Once I've figured it out was easy to workaround placing new section there with value 0. And playing with `rfp-cli -rfo` helped me to solve this issue afterwards.

`eu-readelf -x .option_setting_s /home/prak/DEV/ra8-head/zephyr/build/zephyr/zephyr.elf`

`0x00000070 ffffffff ffffffff ffffffff ffffffff ................`
Is the correct value
`0x00000070 00000000 ffffffff ffffffff ffffffff ................`
Is what my particular RA8T1 needs. Fun.

piotrrak, 2024-12-21:
I should also mention that flashing with jlink still fails for me, just flasing with renesas own flasher does wrok
```
west flash -r jlink --no-erase
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: reset after flashing requested
-- runners.jlink: JLink version: 8.10l
-- runners.jlink: Flashing file: /home/prak/DEV/ra8-head/zephyr/build/zephyr/zephyr.hex
FATAL ERROR: command exited with status 1: /home/prak/.local/JLink_Linux_V810l_x86_64/JLinkExe -nogui 1 -if swd -speed auto -device R7FA8M1AH -CommanderScript /tmp/tmpge2unx8ojlink/runner.jlink -nogui 1
```

closed 2025-03-06 by github-actions[bot]