zephyrproject-rtos/zephyr #83033: net: http server: add support for additional methods

viktorcsanak, 2024-12-16:
**Is your feature request related to a problem? Please describe.**

The HTTP Server currently only supports `GET` and `POST` methods as seen in `http_server_http1.c`

``` 
switch (client->method) {
	case HTTP_HEAD:
		if (user_method & BIT(HTTP_HEAD)) {
			ret = SEND_RESPONSE(RESPONSE_TEMPLATE_DYNAMIC,
					    dynamic_detail->common.content_type);
			if (ret < 0) {
				return ret;
			}

			dynamic_detail->holder = NULL;

			return 0;
		}

	case HTTP_GET:
		/* For GET request, we do not pass any data to the app but let the app
		 * send data to the peer.
		 */
		if (user_method & BIT(HTTP_GET)) {
			return dynamic_get_req(dynamic_detail, client);
		}

		goto not_supported;

	case HTTP_POST:
		if (user_method & BIT(HTTP_POST)) {
			return dynamic_post_req(dynamic_detail, client);
		}

		goto not_supported;

not_supported:
	default:
		LOG_DBG("HTTP method %s (%d) not supported.",
			http_method_str(client->method),
			client->method);

		return -ENOTSUP;
	} 
```

And `http_server_http2.c`

```
switch (client->method) {
	case HTTP_GET:
		if (user_method & BIT(HTTP_GET)) {
			return dynamic_get_req_v2(dynamic_detail, client);
		}

		goto not_supported;

	case HTTP_POST:
		/* The data will come in DATA frames. Remember the detail ptr
		 * which needs to be known when passing data to application.
		 */
		if (user_method & BIT(HTTP_POST)) {
			client->current_detail =
				(struct http_resource_detail *)dynamic_detail;
			break;
		}

		goto not_supported;

not_supported:
	default:
		LOG_DBG("HTTP method %s (%d) not supported.",
			http_method_str(client->method),
			client->method);

		return -ENOTSUP;
	}
```

**Describe the solution you'd like**
The HTTP server should support additional methods, at least `PUT` and `DELETE` to allow proper API implementations.

rlubos, 2024-12-16:
Looking briefly into this it seems we could re-use the existing dynamic method handlers for PUT/DELETE, I'll try to wrap this up and send a patch.

rlubos, 2024-12-16:
@viktorcsanak Please give https://github.com/zephyrproject-rtos/zephyr/pull/83060 a try.

closed 2024-12-18 by kartben

viktorcsanak, 2024-12-19:
@rlubos Sorry, I could only get back to answering now.
I could use the newly added methods after applying said changes, thanks a lot.

PR #83060 (2024-12-18): net: http_server: Support PUT/PATCH/DELETE for dynamic resources 
* Extend the list of methods supported by dynamic resources with
PUT/PATCH/DELETE. The methods does not require any extra handling
from the server perspective, so the existing method handlers could be
reused.

* Add additional test cases which verify that the HTTP server code can
handle PUT/PATCH/DELETE request types.

Resolves #83033