zephyrproject-rtos/zephyr #83037: cannot initiate nvs in NRF52840 dongle

Saranya-karan, 2024-12-16:
iam new to nrf52840 DONGLE.i want to initiate nvs and read/write in it.
but it kept on failed.my main.c
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/storage/flash_map.h>
#include <zephyr/fs/nvs.h>
#include <cJSON.h>
#include <string.h>
#include <zephyr/drivers/flash.h>

static struct nvs_fs fs;

#define NVS_PARTITION          storage_partition
#define NVS_PARTITION_DEVICE   FIXED_PARTITION_DEVICE(NVS_PARTITION)
#define NVS_PARTITION_OFFSET   FIXED_PARTITION_OFFSET(NVS_PARTITION)

#define JSON_DATA_ID 10
#define SLEEP_TIME   2000  // 2 seconds

void store_json_data(struct nvs_fs *fs) {
    cJSON *root = cJSON_CreateObject();

    // Add actual numeric values
    cJSON_AddStringToObject(root, "name", "Bluetooth");

	char tx_power_str[16]; 
    sprintf(tx_power_str, "%d", -40);  // Example TX power value as string
    cJSON_AddStringToObject(root, "tx_power", tx_power_str);

    const char *json_string = cJSON_PrintUnformatted(root);
    printk("JSON data stored: %s\n", json_string);

    // Write to NVS
    int rc = nvs_write(fs, JSON_DATA_ID, json_string, strlen(json_string) + 1);
    if (rc < 0) {
        printk("Error: Failed to write JSON data to NVS, rc=%d\n", rc);
    } else {
        printk("JSON data successfully written to NVS\n");
    }

    cJSON_Delete(root);
}

#include <stdlib.h>  // For strtol

void read_json_data(struct nvs_fs *fs) {
    char json_buffer[256]; 
    
    int rc = nvs_read(fs, JSON_DATA_ID, json_buffer, sizeof(json_buffer));
    
    if (rc > 0) {
        json_buffer[rc] = '\0';  // Null-terminate the string
        printk("JSON data read: %s\n", json_buffer);

        cJSON *root = cJSON_Parse(json_buffer);
        if (root != NULL) {
            const cJSON *name = cJSON_GetObjectItem(root, "name");
            const cJSON *tx_power_str = cJSON_GetObjectItem(root, "tx_power");

            if (name && tx_power_str) {
                printk("Parsed JSON: name=%s, tx_power=%s\n",
                       name->valuestring,  
                       tx_power_str->valuestring 
                       );

                // Convert tx_power and battery_level to integers using strtol
                char *endptr;
                long tx_power = strtol(tx_power_str->valuestring, &endptr, 10);  // Convert to integer
              

                printk("Converted values: tx_power=%ld\n", tx_power);
            } else {
                printk("Error: JSON parsing error - missing fields\n");
            }

            cJSON_Delete(root);
        } else {
            printk("Error: Failed to parse JSON data\n");
        }
    } else if (rc == 0) {
        printk("Warning: No JSON data found in NVS\n");
    } else {
        printk("Error: Failed to read JSON data, rc=%d\n", rc);
    }
}

int main(void) {
    int rc;
    struct flash_pages_info info;

    // Initialize NVS file system
    fs.flash_device = NVS_PARTITION_DEVICE;
    if (!device_is_ready(fs.flash_device)) {
        printk("Error: Flash device not ready\n");
        return -1;
    }

    fs.offset = NVS_PARTITION_OFFSET;
    rc = flash_get_page_info_by_offs(fs.flash_device, fs.offset, &info);
    if (rc) {
        printk("Error: Failed to get page info, rc=%d\n", rc);
        return -1;
    }

    fs.sector_size = info.size;
    fs.sector_count = 3U;

    rc = nvs_mount(&fs);
    if (rc) {
        printk("Error: Failed to mount NVS, rc=%d\n", rc);
        return -1;
    }

    // Store and read JSON data
    store_json_data(&fs);
    read_json_data(&fs);

    while (1) {
        k_msleep(SLEEP_TIME);
    }

    return 0;
}
prj.conf

CONFIG_FLASH=y
CONFIG_LOG=y
CONFIG_UART_CONSOLE=y
CONFIG_CONSOLE=y
CONFIG_PM=y
CONFIG_PM_LOG_LEVEL_DBG=y

CONFIG_NVS=y

CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_NVS_LOG_LEVEL_DBG=y
CONFIG_REBOOT=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y

CONFIG_FLASH_MAP=y
CONFIG_USB_DEVICE_STACK=y

CONFIG_CJSON_LIB=y

Laczen, 2024-12-16:
Hi @Saranya-karan, this will probably be more a discussion than a bug. In the issue you don't mention what is failing in main. Is it the mounting, or something else ?

Saranya-karan, 2024-12-16:
> Hi @Saranya-karan, this will probably be more a discussion than a bug. In the issue you don't mention what is failing in main. Is it the mounting, or something else ? What error code is returned?

i really dont know.in build there is no errors.after uploading hex file through programmer tool,i cannot open the uart port in putty(so cannot see the error log).

i need the guidance..if anyother way to see the errors ...also why my nvs not getting initiated

kartben, 2024-12-17:
@Saranya-karan would you be able to bisect this to figure out when your code stopped working? Thanks!

Laczen, 2024-12-17:
> @Saranya-karan would you be able to bisect this to figure out when your code stopped working? Thanks!

I don't think it ever worked, the author is referring to a different device. The nrf52480 dongle is a bit of an annoying device as most users will want to use the usb to debug. This takes some time to get up and as the authors code is direct returning on error there might never be anything visible.

otzy, 2025-02-28:
The problem is the address of the storage partition. It does not take into account that all addresses on the dongle are 0x1000 shifted because of MBR.
I had the same problem and solved it by fixing zephyr/boards/nordic/nrf52840dongle/fstab-stock.dtsi
```		
               slot1_partition: partition@76000 {
			label = "image-1";
			reg = <0x00076000 0x00066000>;
		};
		storage_partition: partition@db000 {
			label = "storage";
			reg = <0x000db000 0x00004000>;
		};

```
namely the size of slot1_partition was reduced and the the start of storage changed to 0xdb000

de-nordic, 2025-02-28:
Referring to the solution by @otzy: I do not think that the default partitioning in the fstab-stock.dtsi is correct in the tree.
The dongle is by default loaded with the bootloader, which is even mentioned in the dtsi, so the default partition layout should account fr all partitions that are already used, and have them defined, because by default a Zephyr app will be built to be placed after the NRF bootloader.
This is a problem I have probably mentioned more than once, in general, that some configuration use start offset provided via Kconfig and some by `zephyr,code-partition` to position application, which is problematic, specifically if app uses other partitions in DTS, because linker, for app using Kconfig only, will not be aware of that and may allow app to cross into other partitions, as it only sees flash size as boundary.

otzy, 2025-03-03:
Hello, I have another problem. After I change the optimization level, it does not work again.
I changed from debug to optimize for speed, it stopped working. I thought it is somehow related to this and switched bag to the Optimize for debugging. It is still not working.
The addresses in the compiled device tree are correct:
```
                                        slot1_partition: partition@76000 {
                                                label = "image-1";
                                                reg = <0x00076000 0x00065000>;
                                        };

                                        storage_partition: partition@db000 {
                                                label = "storage";
                                                reg = <0x000db000 0x00004000>;
                                        };
```
But the device works only if I set `CONFIG_SETTINGS=n`

Is there anything I forgot/missed to change?

de-nordic, 2025-03-04:
> 
> UPDATE: somehow sysbuild option activated and enabled Partition Manager, which is doing it's own magic

Yeah, When using sdk-nrf, not upstream Zephyr, the Partition Manager may be enabled by default for some of configurations, like building app with MCUboot child image.

MSeesko, 2025-03-13:
Hi, i had the same problem. I can confirm, building with `CONFIG_SETTINGS=y` and with sysbuild on nrf52840dongle wont work.
Building the same with sysbuild turned off works.

Since this probably has to do with partitions changed by the Partition Manager - what is missing here to do it the correct way?

mmatthe, 2025-03-25:
I'm currently facing the same problem - I can neither use Settings nor NVS directly, as it hangs during the call to `nvs_mount`. I'm using sysbuild and also partition manager. Is there anything I can do to get this to work? I have already tried to apply the changes to the storage partitions in the dtsi-file, but with no help.

mmatthe, 2025-03-27:
> I'm currently facing the same problem - I can neither use Settings nor NVS directly, as it hangs during the call to `nvs_mount`. I'm using sysbuild and also partition manager. Is there anything I [...]

@otzy @MSeesko would you be so kind and provide me your example project that ran on the dongle? I have now also tried to run above code example using VS code without sysbuild and with the adaptations to the partition table as mentioned above. However, I'm missing something as still the dongle does not allow `nvs_mount`. 

I have also verified that the offsets in the firmware are correct and that flash writes are attempted to the correct places. Still, when finally trying to write something to flash (which is happening within nvs_mount), the dongle hangs.

otzy, 2025-03-27:
Hi @mmatthe, sorry, I can not share the entire project.
The related changes I did for the NVS to work were only those I mentioned in  https://github.com/zephyrproject-rtos/zephyr/issues/83037#issuecomment-2690348578
Then I went a little further and reduced slot1 even more, like it is in https://github.com/zephyrproject-rtos/zephyr/issues/83037#issuecomment-2694171362, but this is not really necessary.

This is my prj.conf:
```ini
CONFIG_SETTINGS=y
CONFIG_BT_SETTINGS=y
CONFIG_SETTINGS_NVS=y
CONFIG_SETTINGS_RUNTIME=y
```

This is how I enable them in code:
```c
#include <zephyr/settings/settings.h>

static void enable_ble() {
	int err = 0;
	int settings_err = 0;
	if (IS_ENABLED(CONFIG_SETTINGS)) {
		settings_err = settings_subsys_init();
		if (!settings_err) {
			LOG_DBG("settings_load...\n");
			settings_err = settings_load();
			if (!settings_err) {
				LOG_DBG("settings_load done");
			} else {
				LOG_ERR("settings_load failed, err=%d", err);
			}
		}
	}

	err = bt_enable(NULL);
	if (err) {
		LOG_ERR("bt_enable failed (err %d)", err);
		return;
	}
	LOG_INF("Bluetooth initialized");

	if (IS_ENABLED(CONFIG_SETTINGS)) {
		// see bt_enable() description
		if (!settings_err) {
			settings_err = settings_load_subtree("bt");
			if (!settings_err) {
				LOG_DBG("settings_load BT done");
			} else {
				LOG_ERR("settings_load BT failed, err=%d", err);
			}
		}
	}
}
```

I don't explicitly call `nvs_mount`

otzy, 2025-03-27:
@mmatthe 

> I have also verified that the offsets in the firmware are correct and that flash writes are attempted to the correct places.
keep in mind that on the dongle all partition addresses are offset 0x1000, because there is MBR in the beginning of flash.
So if for example the partition starts at address 0xdb000 in the device tree, it will be compiled with address 0xdc000.

The default dongle device tree does not take this into account and the storage partition overlaps with bootloader.

mmatthe, 2025-03-27:
Thanks a lot, @otzy . With your help I could finally make it work. For reference, I have attached a project 
[nvs.zip](https://github.com/user-attachments/files/19488352/nvs.zip) which compiles and runs on the nrf52840 dongle, including when using sysbuild. To build this, open in VS code, add a build configuration for the nrf52840dongle using sysbuild, compile it and flash via e.g. the nRF Connect Desktop Programmer.

As many people have mentioned, the problem is if the flash area is trying to write into the boot loader. I was a bit confused at first, becuase the MBR is located at 0x0000 to 0x1000. However, there's the MCUBoot loader located at 0xe00000 in the memory, as pointed out [here](https://devzone.nordicsemi.com/f/nordic-q-a/69117/flash-read-write-on-nrf52840-dongle). Maybe it helps others to better understand what is going on. 

Moreover, when using non-sysbuild, and the `fstab-stock.dtsi` modifications as mentioned by `otzy` are applied, the `STORAGE_OFFSET` is at 0xdb000  (897024 decimal). Compiling and running with this offset makes the example work. 

Then, when setting the offset manually (and not from the compile time constant `FIXED_PARTITION_OFFSET(storage_partition)`,  it will also works when using sysbuild. In sysbuild, the partion manager siliently ignores that there is the mcu bootloader living at the high end of the flash space. This can be overcome, by artificially increasing the storage partition size to grow below the bootloader start via

```
CONFIG_PM_PARTITION_SIZE_NVS_STORAGE=0x25000
```

Moreover, I have found out that the entire program only works, if 

```
ENABLE_BT=y
```

is contained in the project configuration. Otherwise, the device still hangs upon calling `nvs_mount`. I don't know why this is and I leave this for others to figure out. 

As mentioned by otzy, I changed the `fstab-stock.dtsi` file to (I think this is not needed when using sysbuild and the partition manager)
```
&flash0 {
  10   │     partitions {
  11   │         compatible = "fixed-partitions";
  12   │         #address-cells = <1>;
  13   │         #size-cells = <1>;
  14   │ 
  15   │         /* MCUboot placed after Nordic MBR (this is set with the FLASH_LOAD_OFFSET
  16   │          * Kconfig value when BOARD_HAS_NRF5_BOOTLOADER is enabled), otherwise MCUboot
  17   │          * will be placed at 0x0. The size of this partition ensures that MCUBoot can
  18   │          * be built with CDC ACM support and w/o optimizations.
  19   │          */
  20   │         boot_partition: partition@0 {
  21   │             label = "mcuboot";
  22   │             reg = <0x00000000 0x00010000>;
  23   │         };
  24   │ 
  25   │         slot0_partition: partition@10000 {
  26   │             label = "image-0";
  27   │             reg = <0x00010000 0x00066000>;
  28   │         };
  29   │         slot1_partition: partition@76000 {
  30   │             label = "image-1";
  31 ~ │             reg = <0x00076000 0x00065000>;
  32   │         };
  33 ~ │         storage_partition: partition@db000 {
  34   │             label = "storage";
  35 ~ │             reg = <0x000db000 0x00004000>;
  36   │         };
  37   │ 
  38   │         /* Nordic nRF5 bootloader <0xe0000 0x1c000>
  39   │          *
  40   │          * In addition, the last and second last flash pages
  41   │          * are used by the nRF5 bootloader and MBR to store settings.
  42   │          */
  43   │     };
  44   │ };
```

I hope others having the same problem can also benefit from these findings. I don't know what are the further implications that the storage area and the boot loader area are overlapping, but at least for now and for a simple NVS examples, this is sufficient.

doug-holtsinger, 2026-01-19:
I ran into the same problem where the NVS partition was overlapping with the dongle bootloader at address 0xe0000, and the result was that I'd get an unexpected exception while executing nvs_mount().   I am using sysbuild and the partition manager, and so the dts is ignored.

I added a file [pm_static_nrf52840dongle_nrf52840.yml](https://github.com/doug-holtsinger/BalancingRobotZ/blob/main/app/pm_static_nrf52840dongle_nrf52840.yml) to the project root directory, and the content tells the partition manager not to use the bootloader space for the dongle board.

```
# Define the existing nRF52840 dongle bootloader area so the
# partition manager doesn't use it.
nrf5_bootloader:
  address: 0xe0000
  size: 0x20000 # Covers bootloader and settings up to 0x100000
```