zephyrproject-rtos/zephyr #83046: Flow control not working in hci_uart on nRF52832

kjs-eir, 2024-12-16:
When using the zephyr/samples/bluetooth/hci_uart sample project with the nRF52DK eval kit (with nRF52832) the UART flow control is not function. RX and TX lines responed correctly when commands are transmitted and received, but when sending longer commands, the RX response before the TX has completed transmission. I have verified that CTS and RTS stays low during transmission with my scope. The only thing i modified is the baudrate in the nrf52dk_nrf52832.overlay file, i modified it to 115200. 
I am transmitting and receiving commands via the COM port when connecting the nRF52DK to my laptop. 

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

I am running zephyr v. 3.6.99 and west v. 1.2.0. 

**Additional context**

nordicjm, 2024-12-16:
There is no flow control on the nrf52dk, you need to use a USB <-> UART chip that supports flow control like an FTDI, the onboard jlink does not

kjs-eir, 2024-12-16:
But there is the option for flow control, if communication only happens between the nrf5340 (interface mcu) and nrf52832 right? The schematics shows rts ans cts wires between the two, and there is a switch in the evaluation board for turn flow control on and off. My issue is persist when I disconnect the nrf52832 from the interface mcu and connect the controller to another mcu acting as a ble host via the uart. With this setup I am still unable to detect changes on the cts/rts. So there is something indicating an issue with how flow control is configured. Best regards, Karl-Johan

nordicjm, 2024-12-16:
The lines are connected, the PC will completely ignore them. Yes flow control should work when communicating with an actual UART device that does support it, but you need to have `hw-flow-control;` set in dts for the UART, have you done that?

kjs-eir, 2024-12-16:
Thanks for clarification. But yes I got `hw-flow-control` set in my overlay file in the hci-uart project.

alwa-nordic, 2024-12-20:
If you open a ticket on [Nordic DevZone](https://devzone.nordicsemi.com/) they will be able to help you debug and find a solution quickly.

kjs-eir, 2025-01-06:
Hi @nordicjm 

I have acquired a FTDI (FT232) and has this hooked-up to my nRF52832 on the nRF52DK. Flow control is working properly. But i am unable to make the controller to start advertising? 

I am transmitting the following sequence to the controller, and the response is verified on my scope:
"01030C00",  # HCI Reset
"01010C08FFFFFFFFFFFFFFFF",  # HCI Set Event Mask
"0106200F0064006400000000000000000000000700",  # HCI LE Set Advertising Parameters
"010820070201060654657374",  # HCI LE Set Advertising Data ("Test")
 "010A200101"  # HCI LE Enable Advertising
 
 Is it correctly understood that i should be able the make the controller advertise with the correct set of HCI commands? If yes, what am i missing?