zephyrproject-rtos/zephyr #83157: net: net_pkt.c: clone_pkt_attributes not cloning all attributes

clamattia, 2024-12-18:
The function `clone_pkt_attributes` does not clone all attributes.
Missing for example `is_lldp` and `is_ppp`.

Ideally, please put testing (fuzzy?) in place to ensure, that any cloned package is equal to the source.

fabiobaltieri, 2025-01-07:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

clamattia, 2025-01-08:
> Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

Adjusted, hope it helps.

carlescufi, 2025-01-08:
@jukkar could you please take a look?

closed 2025-01-13 by kartben

clamattia, 2025-01-14:
@kartben Thank you

PR #83732 (2025-01-13): net: pkt: Clone all needed attributes
The net_pkt_clone() did not cloned all needed fields. Added what was missing from the clone.

Fixes #83157
commit ec965079 (2025-01-13): net: pkt: Clone all needed attributes

The net_pkt_clone() did not cloned all needed fields.
Added what was missing from the clone.

Fixes #83157