zephyrproject-rtos/zephyr #83161: net: ethernet L2: Do not drop packets (for example `lldp`)

clamattia, 2024-12-18:
Ethernet Layer 2 handling drops all `lldp` packets, even if  `CONFIG_NET_LLDP` is not enabled.
This prevents a `DGRAM_SOCKET` to receive them.

Consider to `NET_CONTINUE` it instead?

Same with other protocols.

fabiobaltieri, 2025-01-07:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

clamattia, 2025-01-07:
> Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

Updated. Hope it helps.

carlescufi, 2025-01-08:
@clamattia do you plan on opening a PR to fix this?

clamattia, 2025-01-20:
This has been fixed as a side effect with https://github.com/zephyrproject-rtos/zephyr/pull/84037

If a user would like to receive `lldp` frames on a `DGRAM_SOCKET`. They are required to set `CONFIG_NET_ETHERNET_FORWARD_UNRECOGNISED_ETHERTYPE`. This is now handled consistently with all protocols.

@jukkar  maybe you could confirm?

jukkar, 2025-01-20:
If a user wants to receive LLDP packets in zephyr there are three options:
1. Utilize the lldp support in zephyr and enable `CONFIG_NET_LLDP` and register a callback `net_lldp_register_callback()` to receive the lldp data. This bypasses the socket layer.
2. Disable `CONFIG_NET_LLDP` and enable `CONFIG_NET_ETHERNET_FORWARD_UNRECOGNISED_ETHERTYPE` and create a packet/raw socket to receive LLDP frames in the application via the socket layer.
3. Disable `CONFIG_NET_LLDP` and register your own `ETH_NET_L3_REGISTER` handler and receive data that way. This is a similar way as in described in 1., the socket layer is bypassed.

From these, the 2. is the most generic one and the same socket code could be utilized in Linux and zephyr.