zephyrproject-rtos/zephyr #83181: v3.7-branch: manifest: update percepio

eriktamlin, 2024-12-18:
Describe the bug
Inefficient tracing of ISRs is fixed in the release of TraceRecorder in the percepio module.

To Reproduce
Enable ISR tracing.

Expected behavior
Not every ISR should be anonymously traced.

Impact
Very inefficient tracing of ISRs.
Logs and console output
Environment (please complete the following information):
Additional context

eriktamlin, 2024-12-18:
Will be used to backport #83180

kartben, 2024-12-18:
Please don't open PRs for the backport, zephyrbot will automatically open them when your main PR will be merged

kartben, 2024-12-18:
My bad, that's the issue you want to use to link to the upcoming backport. Sorry.
You don't necessarily need multiple issues though, it's fine to have all backports referencing the same issue

eriktamlin, 2024-12-19:
Ah, noted!
Thanks!

nashif, 2025-01-07:
merged

closed 2025-01-07 by nashif