zephyrproject-rtos/zephyr #83194: doc: Read&Decode Stream code example is broken

stubb0rnCoder, 2024-12-19:
**Describe the bug**
Using current main (2460e894e4a071a79f90dc3917386d8b9e254210) the example for streaming read&decode [here](https://docs.zephyrproject.org/latest/hardware/peripherals/sensor/read_and_decode.html#streaming) is not compiling.

**Macros**
The `ACCEL_IODEV_PTR` and `ACCEL_DEFINE_IODEV` macros are missing a `...` argument or otherwise the compiler will complain about them taking only one but getting two arguments from `LISTIFY()`.

**decoder**
The statement `decoder->get_frame_count(buf, {SENSOR_CHAN_ACCEL_XYZ, 0}, &frame_count);` throws an error `expected expression before '{' token` on the second argument.

**print statements** 
Some of the `printk` statements use `dev->name` where (as I understand it) a `sensor->name` is needed.

**data retrieval** 
The final print statement extracts/formats the acceleration data for the print statement which throws a pointer vs. member error:
```c
note: in expansion of macro 'PRIsensor_three_axis_data
_arg'
  107 |              PRIsensor_three_axis_data_arg(accel_data, 0));
...
zephyr/include/zephyr/drivers/sensor_data_types.h:72:51: error: '(struct sensor_three_axis_data *)&a
ccel_data' is a pointer; did you mean to use '->'?
   72 |         (data_).header.base_timestamp_ns + (data_).readings[(readings_offset_)].timestamp_delta,   \
```

**To Reproduce**
Insert the snippet into a minimal application.

I used the `xg24_dk2601b` board which needs adjustment of the board overlay to have two IMUs defined but I don't see the reported issues being board related. 

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain Zephyr SDK v0.17.0
 - Commit SHA 2460e894e4a071a79f90dc3917386d8b9e254210

kartben, 2025-01-07:
> [...] can we please add doc snippet samples in some manner that doesn't require the full work of making them samples in some manner [...]

at the risk of sound like I am "mirroring" what you just said, why not actually do the "full work" (btw if by "full" you're implying/feel there's some kind of overhead to it, please let me know) of turning this into a proper sample, if the point *is* to have this code actually be runnable / used as a .... sample?

teburd, 2025-01-07:
> at the risk of sound like I am "mirroring" what you just said, why not actually do the "full work" (btw if by "full" you're implying/feel there's some kind of overhead to it, please let me know) of [...]

It's not meant to be a generic stand alone sample with independent rst docs. It makes no sense to document a sample that is meant to be a snippet in other documentation to me.

teburd, 2025-03-11:
@kartben any thoughts? I don't think these code snippets warrant setting up a complete documented sample, would setting up a small docs/code_snippets type directory where these things can live and be built in CI to ensure they build work for you?