zephyrproject-rtos/zephyr #83212: CI build fails with `RPC failed; HTTP 503 curl 22`

rruuaanng, 2024-12-19:
**Describe the bug**
from: https://github.com/zephyrproject-rtos/zephyr/actions/runs/12384969191/job/34570424325?pr=83138
```
Run export BSIM_VERSION=$( west list bsim -f {revision} )
  export BSIM_VERSION=$( west list bsim -f {revision} )
  echo "Manifest points to bsim sha $BSIM_VERSION"
  cd /opt/bsim_west/bsim
  git fetch -n origin ${BSIM_VERSION}
  git -c advice.detachedHead=false checkout ${BSIM_VERSION}
  west update
  make everything -s -j 8
  shell: sh -e {0}
  env:
    CCACHE_DIR: /node-cache/ccache-zephyr
    CCACHE_REMOTE_STORAGE: redis://cache-*.keydb-cache.svc.cluster.local|shards=1,2,3
    CCACHE_REMOTE_ONLY: true
    LLVM_TOOLCHAIN_PATH: /usr/lib/llvm-16
    COMMIT_RANGE: 4272353e29364c931cdc213915aa6b626f14cfeb..b8ebc2ee076c3cd6043e30c61070aeeb64aa192b
    BASE_REF: main
    ZEPHYR_SDK_INSTALL_DIR: /opt/toolchains/zephyr-sdk-0.17.0
Run '/home/runner/k8s/index.js'
  shell: /home/runner/externals/node16/bin/node {0}
Manifest points to bsim sha 1f242f4ed7fc141fdfcfeca8d21c6d9e801179d7
error: RPC failed; HTTP 503 curl 22 The requested URL returned error: 503
fatal: error reading section header 'acknowledgments'
```

**To Reproduce**
None

**Expected behavior**
Pass

**Impact**
None

**Logs and console output**
None

**Environment (please complete the following information):**
None
**Additional context**

rruuaanng, 2024-12-19:
I'm not sure if this has anything to do with my changes, but it doesn't look like it :(

nordicjm, 2024-12-19:
This is not a bug, the IP is probably blocked for a period of time due to rate limiting, rerunning CI after a period of time should make it pass

closed 2024-12-19 by nordicjm