zephyrproject-rtos/zephyr #83244: Enabling APS6404L PSRAM on flexspi2 for TEENSY41 fails

bdkrae, 2024-12-19:
**Describe the bug**
TEENSY41 board has an optional PSRAM chip (APS6404L). To enable it an overlay file was added to `samples/drivers/memc`. Building the sample for this target fails:

```
In file included from /Users/bernhardkramer/zephyrlatest/zephyr/include/zephyr/sys/util_macro.h:34,
                 from /Users/bernhardkramer/zephyrlatest/zephyr/include/zephyr/sys/util.h:17,
                 from /Users/bernhardkramer/zephyrlatest/zephyr/include/zephyr/sys/atomic.h:18,
                 from /Users/bernhardkramer/zephyrlatest/zephyr/include/zephyr/kernel_includes.h:25,
                 from /Users/bernhardkramer/zephyrlatest/zephyr/include/zephyr/kernel.h:17,
                 from /Users/bernhardkramer/zephyrlatest/zephyr/drivers/memc/memc_mcux_flexspi_aps6404l.c:13:
/Users/bernhardkramer/zephyrlatest/zephyr/drivers/memc/memc_mcux_flexspi_aps6404l.c:225:18: error: 'flexspi_device_config_t' {aka 'struct _flexspi_device_config'} has no member named 'addressShift'
  225 |                 .addressShift = false,                                  \
      |                  ^~~~~~~~~~~~
```

Mentioned type `flexspi_device_config_t` is defined in NXP mcux-sdk. It seems that the majority of nxp soc´s doesn´t have the struct member `.addressShift`, except the FRDM_RW612 and RD_RW612_BGA.

If line 225 is removed from `memc_mcux_flexspi_aps6404l.c`, then code builds properly and works like expected. Could this be a solution? Both boards FRDM_RW612 and RD_RW612_BGA are built correctly with the proposed solution, but the RD_RW612_BGA seems to have a deviating default value (addressShift=true) according to nxp-sdk. This should be checked on hardware for these boards. 

**To Reproduce**
Steps to reproduce the behavior:
1. Insert overlay file `teensy41.overlay` with PSRAM config to `samples/drivers/memc/boards`:

```
/ {
	aliases {
		sram-ext = &aps6404l;
	};
};

&pinctrl {
	/* FLEXSPI2 for optional PSRAM/FLASH on TEENSY41*/
	pinmux_flexspi2: pinmux_flexspi2 {
		group0 {
			pinmux = <&iomuxc_gpio_emc_24_flexspi2_a_ss0_b>,
				<&iomuxc_gpio_emc_25_flexspi2_a_sclk>,
				<&iomuxc_gpio_emc_26_flexspi2_a_data0>,
				<&iomuxc_gpio_emc_27_flexspi2_a_data1>,
				<&iomuxc_gpio_emc_28_flexspi2_a_data2>,
				<&iomuxc_gpio_emc_29_flexspi2_a_data3>;
			drive-strength = "r0-6";
			slew-rate = "fast";
			nxp,speed = "100-mhz";
			input-enable;
		};
	};
};

&flexspi2 {
	status = "okay";
	pinctrl-0 = <&pinmux_flexspi2>;
	pinctrl-names = "default";
	reg = <0x402a4000 0x4000>, <0x70000000 DT_SIZE_M(8)>;
	/* APS6404L is 8MB, 64MBit pSRAM */
	aps6404l: aps6404l@0 {
		compatible = "nxp,imx-flexspi-aps6404l";
		size = <DT_SIZE_M(8 * 8)>;
		reg = <0>;
		spi-max-frequency = <DT_FREQ_M(109)>;
		status = "okay";
	};
};

```
2. `west build -p -b teensy41 samples/drivers/memc`
3. See error

**Expected behavior**
Should build and work properly.

**Environment:**

 - OS: MacOS
 - Toolchain: Zephyr SDK 0.16.8
 - d789f339efb78a136a10c74efdeeb963e085e104

**Note**
Link to Teensy4.1 PSRAM option:
https://www.pjrc.com/store/psram.html

dleach02, 2025-02-18:
@davidmissael can you look at this? The code in question was something you added and the author is noting that removing

.addressShift = false,

Addresses his problem but causes problems with other platforms

ekwus, 2025-05-02:
Hey

I'm also experiencing this issue when trying to add PSRAM to the Teensy41 (teensy4.1) board.

I'm able to get things working by doing the same as the OP.

Cheers

Dave

bdkrae, 2025-05-07:
@davidmissael could you please check if proposed solution works with FRDM_RW612 and RD_RW612_BGA?

DerekSnell, 2025-07-29:
> Is there a checklist for first timers on doing so?

Hi @ekwus ,
Thank you for contributing.  Please see https://docs.zephyrproject.org/latest/contribute/guidelines.html 
Best regards

ekwus, 2025-07-29:
I've confirmed this is still an issue and that the fix outlined above still fixes it. I've also reached out on Discord to see if anyone has the RW612 based boards that would be able to test if it doesn't break them.

I will put together a commit and subsequent pull request too.