zephyrproject-rtos/zephyr #83257: Support config option to unify TX and RX net pkts and bufs

MaochenWang1, 2024-12-20:
**Is your enhancement proposal related to a problem? Please describe.**
Currently net pkts and net bufs are divided into TX and RX, which will consume more footprint for memory constraint device.
Usually for Wi-Fi device, there are net buffers must be used for RX-reorder, but these buffers can't be used for TX.

**Describe the solution you'd like**
Default still use separate buf for TX and RX.
Support config option to use one kind of pkt/buf for both TX and RX.

jukkar, 2024-12-20:
We had at some point in the past, a shared net_buf pools for TX and RX, but then divided the usage because of the problems. One big issue with one net_buf pool is that if there is high amount of incoming packets it would prevent packet sending. It is very difficult to control the incoming data rate in that case without distrupting the ongoing network connections.

jukkar, 2025-04-23:
This is probably not going to be done because of the past experience with unified TX/RX buffers, so closing the req.

closed 2025-04-23 by jukkar