zephyrproject-rtos/zephyr #83260: Support one continuous net_buf for TCP TX case

MaochenWang1, 2024-12-20:
**Is your enhancement proposal related to a problem? Please describe.**
Currently in function tcp_out_ext, for TCP TX, it will use two separate net_buf, one contains ip and tcp header, another contains data payload.
It has two side effects:
(1) for running TCP throughput case, the num of net_buf should be twice of net_pkt, and consume more memory.
(2) non-continuous buffer will decrease the memcpy efficiency when copy to lower interface.

**Describe the solution you'd like**
Use single net_buf for TCP TX data when NET_BUF_DATA_SIZE is large enough.

jukkar, 2024-12-20:
If memory is an issue, it is actually better to set `CONFIG_NET_BUF_DATA_SIZE` to smaller value so that buffers are utilized better if the amount of data to be sent is small. Other option is to use `CONFIG_NET_BUF_VARIABLE_DATA_SIZE` which uses malloc to allocate just enough memory to the data to be sent.

MaochenWang1, 2024-12-30:
Hi @rlubos  why configuring the tcp_rx_window/tcp_tx_window as `(CONFIG_NET_BUF_TX_COUNT * CONFIG_NET_BUF_DATA_SIZE) / 3`, not `(CONFIG_NET_BUF_TX_COUNT * CONFIG_NET_BUF_DATA_SIZE) / 2` or other value?
https://github.com/zephyrproject-rtos/zephyr/blob/1b70bfa14e288d97b44d2446c8ef0b93226514e6/subsys/net/ip/tcp.c#L57

rlubos, 2025-01-02:
> Hi @rlubos why configuring the tcp_rx_window/tcp_tx_window as `(CONFIG_NET_BUF_TX_COUNT * CONFIG_NET_BUF_DATA_SIZE) / 3`, not `(CONFIG_NET_BUF_TX_COUNT * CONFIG_NET_BUF_DATA_SIZE) / 2` or other [...]

It's just an arbitrary value that seemed reasonable as a default. If the default window size doesn't match the application needs (for example because it's expected to have multiple parallel TCP connections open), on can adjust the windows size with `CONFIG_NET_TCP_MAX_SEND/RECV_WINDOW_SIZE`, or at runtime with socket options.