zephyrproject-rtos/zephyr #83277: drivers/ethernet/eth_stm32_hal.c: net_pkt_write fails, when packet has size `1518`

clamattia, 2024-12-20:
When the system receives a packet with maximum payload (size 1518 including header), the driver fails with `<err> eth_stm32_hal: eth_rx: Failed to append RX buffer to context buffer, rx_header->size=1518` (I added the length to the message).
This happens consistenly (we are using an ethernet test setup). 

To reproduce, send a packet with maximum size.

Expected: The system does not fail, even with packets of maximal payload length.

Impact: This makes the driver unusable for our project. We need to find a solution.

Let me know if additional information is needed or, if I may test a patch on our setup.

clamattia, 2024-12-20:
`Still some length to go 4`

clamattia, 2024-12-20:
Yes, allocation is faulty:
`Data allocation maximum size 1514 (requested 1518)`

Please also make it a warning in case we allocate less then requested and change message to `Data allocation size 1514 (requested 1518)`, because it is not the 'maximum' it is the actual allocation size.

clamattia, 2024-12-20:
Please fix accordingly:
```diff
-ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
-		    NULL, &eth0_data, &eth0_config,
-		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_MTU);
+ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
+		    NULL, &eth0_data, &eth0_config,
+		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_FRAME_SIZE_MAX);
```

marwaiehm-st, 2024-12-30:
> This would for example fix the problem. But maybe it is also somewhere else with `NET_ETH_MAX_HDR_SIZE`.
> 
> ```diff
> -ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
> -		    NULL, &eth0_data, &eth0_config,
> -		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_MTU);
> +ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
> +		    NULL, &eth0_data, &eth0_config,
> +		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_FRAME_SIZE_MAX);
> ```

Using ETH_STM32_HAL_FRAME_SIZE_MAX will fix the problem, but the maximum will be 1532 (1518 + NET_ETH_MAX_HDR_SIZE). Therefore, you allocate more than you need. The allocation size should be adjusted accordingly to the maximum frame size of 1518 bytes.

The actual Ethernet Frame Components

- Ethernet Header: 14 bytes
- Payload (MTU): 1500 bytes (standard)

To work with 1518, we have to proposition:

1. Enable the VLAN, an additional 4 bytes are added to the Ethernet frame for the VLAN tag 
```cmake
CONFIG_NET_VLAN=y
``` 
In this case the value of NET_ETH_MAX_HDR_SIZE is 18(instead of 14 without VLAN)
[image]

- Ethernet Header: 14 bytes
- VLAN Tag: 4 bytes
- Payload (MTU): 1500 bytes

2. Add to the file eth_stm32_hal_priv.h

```c
/* *
   * Define the maximum frame size including Ethernet payload, and FCS
   * Payload (ETH_STM32_HAL_MTU): 1500 bytes (standard)
   * Frame Check Sequence (FCS): 4 bytes
   * /
#define ETH_STM32_HAL_FCS (ETH_STM32_HAL_MTU + 4)
```
The Frame Check Sequence (FCS) is an additional 4 bytes used for error checking. It is typically appended to the end of the Ethernet frame by the network hardware and is not always included in the frame size calculations reported by software tools.

> ```diff
> -ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
> -		    NULL, &eth0_data, &eth0_config,
> -		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_MTU);
> +ETH_NET_DEVICE_DT_INST_DEFINE(0, eth_initialize,
> +		    NULL, &eth0_data, &eth0_config,
> +		    CONFIG_ETH_INIT_PRIORITY, &eth_api, ETH_STM32_HAL_FCS);
> ```

In this case 1518 it will be allocated

- Ethernet Header: 14 bytes
- Payload (MTU): 1500 bytes
- Frame Check Sequence (FCS): 4 bytes

clamattia, 2025-01-06:
> To work with 1518, we have to proposition:
> 
> 1. Enable the VLAN, an additional 4 bytes are added to the Ethernet frame for the VLAN tag

I later also realized, that this may be a feasible solution. Will test this.

A further question is, whether the failure handling is good in the case when too big (including vlan) frames arrive. Currently behavior is:
- Driver requests frames of appropriate size
- Allocation gets silently truncated to the perceived maximum valid (because vlan is not activated)
- Driver fails to write all data into the allocation, because it was silently truncated to less then the requested size

clamattia, 2025-01-08:
The driver may need to be able to receive frames with ieee802.1q vlan tags, even if `CONFIG_VLAN` is disabled.
See related issue https://github.com/zephyrproject-rtos/zephyr/issues/83662 .

clamattia, 2025-01-08:
I think https://github.com/zephyrproject-rtos/zephyr/issues/83662 would resolve this for me.

clamattia, 2025-01-23:
Indirectly resolved by: https://github.com/zephyrproject-rtos/zephyr/pull/83734

closed 2025-01-23 by clamattia