zephyrproject-rtos/zephyr #83279: settings: settings_load() doesn't fail like the documentation says it should

maximevince, 2024-12-20:
The Zephyr documention here:
https://docs.zephyrproject.org/latest/services/settings/index.html#loading-data-from-persisted-storage
says this about calling `settings_load()`:
> After all data is loaded, the h_commit handler is issued, signalling the application that the settings were successfully retrieved.

However, if my storage backend (e.g. filesystem / LittleFS in my case), doesn't contain a settings file at all, the call to `settings_load()` will happily call the `h_commit()` handler **and** return `0`, all the while no settings have been loaded "succesfully".

See the relevant code of `settings_load_subtree()` here:
https://github.com/zephyrproject-rtos/zephyr/blob/86c979147db1625d78b2d32431ab3982711e0165/subsys/settings/src/settings_store.c#L60

Am I missing something here, or should the `settings_load()` function bail out with an error code if any of the subtrees failed to load?
Is there another way I am supposed to check for successful load of my settings?

Laczen, 2024-12-20:
Hi @maximevince , this is a bit of a corner case.

1. If there are no settings, the system has not loaded anything but the reporting is correct (the system has succesfully loaded 0 settings),
2. Should `h_commit()` be called, I think so because even with no settings the `h_commit()` can be used to enable certain subsystems (using defaults that are in code),

Whether there is an empty settings file or no settings file at all the system behaves in the same way (I don't remember if a settings file is created when it tries to load). All systems start with no settings file or an empty settings file, this should be no problem for the settings subsystem. The system also supports multiple backends (although rarely used) and the failure of one backend should not necessarily mean that no other settings can be loaded.

If you want to make sure that all your backends have succesfully loaded you can create a specific (for each backend) `h_commit()` that returns an error when a specific item has not been loaded, this will be returned by `settings_load()`. The returned value is not ideal as it is a combination of returned values by `h_commit()` but it should at least help discover problems.

maximevince, 2024-12-23:
@Laczen thanks for the feedback. I just tried using `h_commit()`. I indeed get the callback when the settings have been loaded, and it's not called when they are not present on the backend. But do you agree that's a pretty clumsy way to just know if a setting was loaded or not?

What's the pattern you suggest for loading defaults if no settings are available?
I am now considering to:
1. Always load default
2. Try and initialize the settings subsystem, possibly triggering the load of my setting
3. Now have either the default or the stored setting available in my "working variable".