zephyrproject-rtos/zephyr #83306: samples: usb: webusb-next: failure to enumerate on lpcxpresso55s16 under MS Windows

henrikbrixandersen, 2024-12-22:
**Describe the bug**
When running `samples/subsys/usb/webusb-next/` on the `lpcxpresso55s16/lpc55s16` board, it fails to enumerate on MS Windows, since the MSOSv2 descriptor cannot be read.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b lpcxpresso55s16/lpc55s16 samples/subsys/usb/webusb-next/`
2. `west flash`
3. See error below when board is plugged into a PC running MS Windows

**Expected behavior**
USB enumeration completes.

**Impact**
Unable to use MSOSv2 descriptors.

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-2624-g59c94db6f052 ***
[00:00:00.104,000] <inf> main: USBD message: Bus reset
[00:00:05.711,000] <inf> main: USBD message: Bus reset
[00:00:05.855,000] <err> usbd_core: UDC error event
[00:00:05.855,000] <inf> main: USBD message: Controller error
[00:00:10.878,000] <err> usbd_core: UDC error event
[00:00:10.878,000] <inf> main: USBD message: Controller error
```

**Environment (please complete the following information):**

 - OS: MS Windows USB host
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA: 59c94db6f052c9d848b4cec2008152d32d8ee597

**Additional context**
When the MS Windows USB host attempts to read the MSOSv2 descriptor, the code eventually reaches https://github.com/zephyrproject-rtos/zephyr/blob/196fc5c36c3cce4a91ee0cfc575da18776a4e5c0/drivers/usb/udc/udc_mcux_ip3511.c#L115 where `udc_ep_is_busy(dev, cfg->addr)` returns true, causing `udc_mcux_ep_feed()` to return `-EBUSY`.

henrikbrixandersen, 2024-12-22:
The same sample enumerates when running on the `from_k64f` board. Likely an issue specific to the `udc_mcux_ip3511.c` driver.

jfischer-no, 2025-02-05:
There is a typo
https://github.com/zephyrproject-rtos/zephyr/blob/e783f69ebfae51bad199b26a00e76cbeecf714f6/drivers/usb/udc/udc_mcux_ip3511.c#L362-L363
But according to the reporter, this alone does not solve the problem.

PR #85567 (2025-02-25): fix usb driver zlt handle typo
Fix #83306

Zephyr USB device stack handles the zlt and the NXP usb controller drivers
process the control endpoint zlt defaultly, disable it in controller driver.
closed 2025-02-25 by kartben