zephyrproject-rtos/zephyr #83309: usb: device: next: increased RAM and flash footprint with "device_next" USB stack

henrikbrixandersen, 2024-12-22:
**Is your enhancement proposal related to a problem? Please describe.**
The `device_next` USB stack comes with a significant increase in both RAM and flash footprint compared to the "legacy" stack. The impact is especially hard on smaller USB MCUs with e.g. 16k RAM and 64k flash (smaller STM32 parts) where the increase can prevent existing firmware applications from switching to the new stack.

**Describe the solution you'd like**
While the design and implementation of of the `device_next` is much superior to the "legacy" stack, it would be nice if the RAM and flash footprint could be reduced to be more friendly towards more resource constrained MCUs.

**Describe alternatives you've considered**
None.

**Additional context**
As an example, the `samples/subsys/usb/hid-mouse/` sample compiled for the `frdm_k64f` board increases significantly in RAM and flash footprint when built for the `device_next` stack, but the problem is not limited to this sample/board.

```
$ west build -b frdm_k64f/mk64f12 samples/subsys/usb/hid-mouse
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       41016 B         1 MB      3.91%
             RAM:       12760 B       192 KB      6.49%
           SRAML:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

```
$ west build -b frdm_k64f/mk64f12 samples/subsys/usb/hid-mouse/ -- -DCONF_FILE=usbd_next_prj.conf -DEXTRA_DTC_OVERLAY_FILE=usbd_next.overlay
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       54792 B         1 MB      5.23%
             RAM:       15480 B       192 KB      7.87%
           SRAML:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

henrikbrixandersen, 2024-12-22:
These are the numbers when building the same sample with logging disabled:

```
$ west build -b frdm_k64f/mk64f12 samples/subsys/usb/hid-mouse -- -DCONFIG_LOG=n
Memory region         Used Size  Region Size  %age Used
           FLASH:       27288 B         1 MB      2.60%
             RAM:       10648 B       192 KB      5.42%
           SRAML:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/brix/Projects/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: frdm_k64f
```

```
$ west build -b frdm_k64f/mk64f12 samples/subsys/usb/hid-mouse/ -- -DCONF_FILE=usbd_next_prj.conf -DEXTRA_DTC_OVERLAY_FILE=usbd_next.overlay -DCONFIG_LOG=n
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       37016 B         1 MB      3.53%
             RAM:       13368 B       192 KB      6.80%
           SRAML:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

henrikbrixandersen, 2024-12-31:
Numbers for the `nrf54h20dk/nrf54h20/cpuapp`:

```
west build -b nrf54h20dk/nrf54h20/cpuapp samples/subsys/usb/hid-mouse/ -- -DCONFIG_LOG=n
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       44052 B       248 KB     17.35%
             RAM:       11232 B       256 KB      4.28%
   DMA_RAM3x_APP:           2 B         4 KB      0.05%
        IDT_LIST:          0 GB        32 KB      0.00%
```

```
west build -b nrf54h20dk/nrf54h20/cpuapp samples/subsys/usb/hid-mouse/ -- -DCONF_FILE=usbd_next_prj.conf -DEXTRA_DTC_OVERLAY_FILE=usbd_next.overlay -DCONFIG_LOG=
...
Memory region         Used Size  Region Size  %age Used
           FLASH:       60564 B       248 KB     23.85%
             RAM:       17992 B       256 KB      6.86%
   DMA_RAM3x_APP:           2 B         4 KB      0.05%
        IDT_LIST:          0 GB        32 KB      0.00%
```

dleach02, 2025-01-08:
some number crunching on the frdm_k64f.

First thing I did was to add to the usbd_next_prj.conf CONFIG_INPUT_MODE_SYNCHRONOUS so that it is in alignment with the default prj.confg. It caused some minor differences that initially confused me.

Key differences between new and legacy:

- Kernel .text is 762 bytes larger due to inclusion of mem_slab.c, mempool.c, and msg_q.c
- Drivers module .text/.rodata is +3260, .data/.bss is smaller. New model has udc_common.c and udc_kinetis.c while old model only has usb_dc_kinetis.c. The .text/.rodata increase mostly comes from the udc_common.c
- The libsubsys__usb__device has a bunch of files added to support usbd. The legacy only has core.c but some of what I think are comparable files in legacy are actually compiled into section libzephyr.a. But if I compare the two sections relevant USB files the new model .text/.rodata is +6738 and .data/.bss is +1665
- libapp.a module is larger in the new model due to the inclusion of sample_usbd_init.c, which the legacy doesn't have. This adds +831 to .text/.rodata and 354 to .data

Attached are the spreadsheets I generated for the two target builds for your pleasure.

[usb_legacy.xlsx](https://github.com/user-attachments/files/18342022/usb_legacy.xlsx)
[usb_new.xlsx](https://github.com/user-attachments/files/18342023/usb_new.xlsx)

erwango, 2025-06-20:
These numbers are a bit worrying. In ST we have several constrained platforms with USB support (64K flash, 20k RAM, some even lower): C0, F0, F1.. with such increase it's likely that users won't be able to use Zephyr anymore on those platforms for USB based projects.

carlescufi, 2025-06-24:
@henrikbrixandersen can you remove the H20 from the table? it is very misleading, because the legacy stack has not USB High-Speed support, and so you are comparing a full-speed stack to a high-speed one.

henrikbrixandersen, 2025-06-24:
> [@henrikbrixandersen](https://github.com/henrikbrixandersen) can you remove the H20 from the table? it is very misleading, because the legacy stack has not USB High-Speed support, and so you are comparing a full-speed stack to a high-speed one.

Are you saying that if the `device_stack` is reduced to USB full-speed support for this device (how would a downstream user do this?), there's little to no increase in footprint? If not, the problem is still very real for anybody having shipped a full-speed USB product based on the existing stack.

carlescufi, 2025-06-24:
> Are you saying that if the `device_stack` is reduced to USB full-speed support for this device (how would a downstream user do this?), there's little to no increase in footprint? If not, the problem [...]

I think so, though I'd like @jfischer-no to confirm. The selection is via: https://github.com/zephyrproject-rtos/zephyr/blob/97519002ad04e744008f4e2661675697624832b3/subsys/usb/device_next/Kconfig#L19

jfischer-no, 2025-06-24:
> Are you saying that if the `device_stack` is reduced to USB full-speed support for this device (how would a downstream user do this?), there's little to no increase in footprint? If not, the problem [...]

"this device" cannot be used with the legacy device stack, even without high-speed support. It is not supported in usb_dc_dw.c driver. usb_dc_dw.c only provides the bare minimum required to support the specific DWC2 IP configuration. nRF54H20 is only supported in the UDC driver, and there is absolutely no way that a thoroughly written driver to support DWC2 IP would **not** be much larger than the usb_dc_dw.c**.**